ev3dev/ev3dev #487: ev3dev settings menu in Brickman

laurensvalk, 2016-01-03:
A settings option in Brickman could be a place for:
- Changing the current user.
- Setting the password for the current user.
- Setting the time zone.
- Setting the locale.

There are probably other useful things to add here as well, but these are the steps that currently have to be done with the command line in ["getting started"] (http://www.ev3dev.org/docs/getting-started/).

WasabiFan, 2016-01-03:
#316 (and to some extent #483) describe tools that could have overlap with these options and other functionality in Brickman. We'll need to make sure that there is a clear distinction between the items that go there and the items that go in Brickman, and keep it documented well.

laurensvalk, 2016-01-03:
Other potentially useful settings:

- Update
- System info (currently this is in the "about" menu.)
- Battery info (currently has its own menu option).
- About (currently has its own menu option)

I think the latter three are not so frequently used (except maybe battery, but it's already displayed on the top right), and moving them under settings reduces the number main menu options, so that everything fits on the screen without scrolling.

I.e. combined with #486 the main menu would become: 

- File Browser
- Device Browser
- Apps
- Wireless and Networks
- Settings

WasabiFan, 2016-01-03:
> moving them under settings reduces the number main menu options, so that everything fits on the screen without scrolling. (...) I.e. combined with #486 the main menu would become: 

That would be great! I think it would also at least partially remedy the issues with such a large font size. And I suspect that we will be adding more menu options in the future anyway, so good to get this sorted out now.

laurensvalk, 2016-01-03:
>  We'll need to make sure that there is a clear distinction between the items that go there and the items that go in Brickman.

Agreed. It's probably fine (and most practical) to let people use the command line for most settings (using ev3dev-config tool #316), but I think it would be nice to have any essential "getting started" settings accessible from Brickman.

rhempel, 2016-01-03:
:+1: for leaving the only a minimal set of "getting started" settings available in Brickman. Maybe we can consider a web interface for setting more complex items as an intermediate between Brickman and the command line.

Just because we have an LCD and buttons does not mean it's a good idea to use them :-)

dlech, 2016-01-03:
Note: as I was writing this, I realized that it may come across as contrary or negative. My intention here is to just explain why things are the way they currently are so that we can decide what is really important and what is not so important. I'm not trying to say we shouldn't make any of the suggested changes.

---

> Changing the current user.

For now, there is only the `robot` user. Using other usernames is not supported.

> Setting the password for the current user.

You mostly need the password to log into a terminal, so why not use the terminal to change the password. It's way easier to type it in that way.

> Setting the time zone.

We don't have a clock in brickman, so why does the timezone matter there? It only matters for logs, which you access via a terminal, so if you are accessing the terminal anyway...

> Setting the locale.

Brickman is not (yet) localized, so this would have no noticeable effect. And actually, when you log into a remote terminal, it uses the locale of the terminal program on the host computer, so changing the locale probably doesn't do anything for most people anyway and we could probably leave it out of the "getting started'.

> Update

I assume you mean updating packages. Things don't always work like they are supposed to when updating packages, so I think it is better to do this in a terminal where you can copy and paste error messages if things aren't working and you need help.

> System info (currently this is in the "about" menu.)

I consider the brickman version from the about menu an important part of the "system info" so separating them would be duplicating information.

> Battery info (currently has its own menu option).

Actually, I would like to move the battery to the "Device Browser" menu. I plan on adding more battery drivers, so there could be more than one battery listed. For example, if you have a NxtMMX or HiTechnic sensor multiplexer that requires an extra power supply, these would show up on the list too.

> About (currently has its own menu option)

Probably don't really need an about item if there is a separate system info item.

dlech, 2016-01-03:
> but I think it would be nice to have any essential "getting started" settings accessible from Brickman.

I say that all the essential steps already are. All of the command line stuff in "getting started" is optional.

rhempel, 2016-01-03:
I think we are in agreement @dlech - I don't think we want to add anymore settings to Brickman - from my point of view it's used for getting the network set up the first time and for browsing around and running programs. 

Now that I think about it, you are correct about not supporting arbitrary users. The default `robot` user is enough - if you are one of the few users that will create their own userid on the brick, then symlink your scripts to the `robot` user area where they can be browsed.

laurensvalk, 2016-01-04:
Sounds all good to me. One user account "robot" will be fine. The user account settings were a response to [this] (https://github.com/ev3dev/ev3dev/issues/486#issuecomment-168519577), but I think I misunderstood what was meant there.

> All of the command line stuff in "getting started" is optional.

OK. If we get to the point where apps can be launched without a command line, I would suggest to take optional command line stuff away from the getting started page.

> I consider the brickman version from the about menu an important part of the "system info" so separating them would be duplicating information.

I meant that we might want to move (not duplicate) some of the less frequently used options to a sub menu (e.g. settings), in order to free up space on the main menu. 

However, if battery info is moving elsewhere and system info would be the only "setting" then we won't need need a settings sub menu.