ev3dev/ev3dev #488: With lego motors, ramp down does not work (ramp up works well)

pootle, 2016-01-03:
Ramp down setting seems to have no effect at all with lego motors from an EV3 brick running ev3dev.

Ramp up works well. Here's part of the python class:
```
from ev3dev.auto import *

class two_motors:
        """synches motors for lego vehicles with 2 driven wheels"""
        def __init__(self,motorpair=(OUTPUT_A, OUTPUT_B), rpm=25):
                self._motors = [Motor(address) for address in (motorpair)]
                for m in self._motors:
                        if m.connected:
                                m.reset()
                                m.speed_sp=m.count_per_rot * rpm / 60
                                m.speed_regulation_enabled="on"

        def isGood(self):
                return self._motors[0].connected and self._motors[1].connected

        def drive(self, time=15, rpm=25):
                if not self.isGood():
                        return

                for m in self._motors:
                        m.ramp_up_sp=8000
                        m.ramp_down_sp=8000
                        m.time_sp=time*1000
                        m.stop_command="hold"
                        m.speed_sp=m.count_per_rot * rpm / 60

                for m in self._motors:
                        m.run_timed()

        def stop(self):
                if self._motors[0].connected:
                        self._motors[0].stop()
                if self._motors[1].connected:
                        self._motors[1].stop()
```

rhempel, 2016-01-03:
The intent is to ramp up to 25% power (should take 2 of 8 seconds total ramp time) run for a while then ramp down from 25% to 0 over the last 2 seconds?

Note that ramp time is in msec (you have that) and represents the time to get to 100% power. So if you only got to 25% it will be 1/4 of the total ramp time.

pootle, 2016-01-08:
It's rpm, not duty cycle. I understand that the time spent ramping will be a proportion of ramp_up. I'm really just concerned that ramp down appears to be totally broken.

rhempel, 2016-01-08:
This is a point of discussion around `speed_regulation` - with regulation set to `off` the `duty_cycle` is varied from 0 to the `duty_cycle_sp" in a percentage proportional to the `ramp_up_sp". So if your `duty_cycle_sp" is 75% and your `ramp_up_sp` is 8000, then the ramping time to get to 75% is 6 seconds.

With regulation set to `on`, the same argument applies to the `speed_sp` - and the maximum speed of the motor is used as the 100% ramp result. For example, if the `medium_motor` has a max speed of 900 RPM, then setting `speed_sp` to 450 will result in a ramp time of 4 seconds if `ramp_up_sp` is 8000.

We are planning to overhaul the tacho motor drivers soon, I am pushing out a proposed operation spec later this evening for review - I'd appreciate your comments.

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel (and other `10-ev3dev` kernels on other platforms). Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

If there are still issues with ramping, please open a new issue.

closed 2016-04-11 by dlech