ev3dev/ev3dev #489: Exception AttributeError: "'LargeMotor' object has no attribute '_path'"

f-steinbauer, 2016-01-04:
Hi,

as per the short chat I had yesterday, I just want to leave a note how to solve the issue mentioned in the topic.

If you try to run the python sample with an older  pre-release as ev3dev-jessie-2015-12-30, which is the latest one you will encounter this "AttributeError: ..."

After the hint from @dlech and @rhempel to update to the mentioned version, the demo scripts run just fine.

And thanks again for the fast support.

Frank

dlech, 2016-01-20:
This is an issue for ev3dev-lang which has a separate issue tracker, so closing.

closed 2016-01-20 by dlech

fryfrog, 2016-08-01:
This is the exact error I'm seeing trying to get a BrickPi on a Pi 3 working with the latest nightly builds. What was the hint from @dlech? :/

dlech, 2016-08-01:
There is a separate [issue tracker for ev3dev-lang-python](https://github.com/rhempel/ev3dev-lang-python). You should report this bug there. Based on your comments on irc, the bug is that ev3dev-lang-python does not show a useful error message if `/sys/class/tacho-motor` does not exist (or possibly is just empty).