ev3dev/ev3dev #490: Help with the basics of ev3dev

tituszban, 2016-01-05:
I'm asking for help with the basics of using ev3dev, and I haven't found any good tutorials.
I now a bit of python, but I've never worked with anything like this (including ssh).
I've followed the steps on the [website](http://www.ev3dev.org/docs/getting-started/), and set up the ["Ethernet over USB"](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb) communcation, but I'm having problems with actually running anything on the brick. Do I download the .py to the brick? Should it be compiled first?

moriarty, 2016-01-05:
What operating system are you running on your computer? 

If it's Linux or OSX it will have the command line ```scp``` available- which can be used to copy a file to the brick. Once the .py file is on the brick, you can ```ssh``` into the brick, and type ```python filename.py``` to run it. 

If you're using windows, see link to old wiki docs below. 

The ev3 python library https://github.com/rhempel/ev3dev-lang-python repository has an active branch with more documentation in the readme, if you need some info on the python. But I think you're looking just for the basic file transfer.

This issue is similar to #228 where I pointed to the old wiki documentation https://github.com/ev3dev/ev3dev/wiki/Set-Up-An-nfs-FileShare

Other command line based alternatives are to use vim and write the python code on the ev3, which may be slow but has the benefit of not needing to constantly copy files or write the code on your computer and use git to sync it to the ev3 but this is only needed if you have a large project with a bunch of files, or are copying projects to many bricks.

WasabiFan, 2016-01-05:
Firstly, I want to apologize for the difficulties that you are facing in getting started. We have only recently (last few months) gotten to the "reasonable usability" milestone where we think that most users (including newcomers to Linux) can use ev3dev. As linked to by the automatic GitHub issue reference above, we have been having a discussion over the past week or two about figuring out how we can better-document and improve this process. Unfortunately, we haven't gotten much implemented yet. For now, the advice from @moriarty should be fine, and we will always be happy to answer any questions that you have. On our website, there's a "Get help" page that has an interface for getting into our IRC chat room -- you can use that for asking small questions or having a sustained discussion, and @dlech is almost always monitoring it.

I plan to continue the discussion mentioned above (in #477) to see if we can get it at least partially resolved soon; once we get some docs written on the process you are asking about, I will post back here to let you know. Feel free to follow/read through that discussion if you want more context.

tituszban, 2016-01-06:
Thanks for the help!
I've managed to do file transfer with FileZilla, as suggested by someone at [#477](https://github.com/ev3dev/ev3dev/issues/477), but now I have the problam that the robot can't run the program (the .py) (it says it's not an executable). So do I have to do something with the program?

WasabiFan, 2016-01-06:
See the second image of FileZilla inn@laurensvalk's post in that other thread? I don't know how he got there (maybe right-clicked on the file), but that is what you want. The problem is that, by default, files on Linux are prevented from running (I suppose it is for safety). You want to find the properties dialog shown in that image and check the three boxes next to "execute". This tells it that you believe it is safe and want to be able to run it.

tituszban, 2016-01-13:
Thanks for all the help!
Here is a quick summary for those who want to start with ev3dev and python.

Setup linux on ev3 following steps [here](http://www.ev3dev.org/docs/getting-started/)
Setup connection using [usb](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/)
Download and [install](http://ddemidov.github.io/ev3dev-lang-python/) python to the ev3 through SSH
Download and install [FileZilla](https://filezilla-project.org/)
Connect and Drag and drop your file to the ev3 using FileZilla
Right click on the file you uploaded, allow permission to execute the file. (acces rule 744)
Make sure that the python code you write has it's line endings set to UNIX.

Enjoy programming ev3s using python!

closed 2016-01-13 by tituszban

rhempel, 2016-01-13:
I have been using FileZilla for a week or so now for file transfer - it works really well.