ev3dev/ev3dev #492: Create python hello world tutorial

dlech, 2016-01-05:
This is an actionable item from #477.

We need to add a python hello world tutorial for users that are new to Linux and have never used SSH or a command line before. This tutorial should be linked from the getting started page. Here is an outline.

* Introduction
* Recommendation of text editors.
     * Cross platform
        * Visual Studio Code
        * Atom
    * Windows
        * Notepad++
    * OS X
        * TextWrangler
* Sample code that can be copied and pasted into text editor

        #!/usr/bin/env python3

        import time

        def main():
            print('hello world!')
            time.sleep(5)
            print('goodbye world!')
            time.sleep(3)

        if __name__ == '__main__':
           main()

* Explanation of shabang and executable bit in Linux
* How to copy the file to the EV3
    * Windows
        * Use WinSCP
        * Show how to set the executable bit
        * Show how to copy the file
    * Linux/OS X
        * Use command line
        * `chmod +x program.py`
        * `scp program.py robot@ev3dev.local:~`
* How to run the program on the EV3
    * Find the program in the File Browser (brickman)
    * Point out that `*` indicates that file is executable
    * Run the file

laurensvalk, 2016-01-05:
If not taken by someone else by January 28 (when I'm done with current semester of exams), I'd be happy to do this. Looking forward to get back to writing tutorials!

rhempel, 2016-01-05:
I'd be happy and grateful if you can help with this - maybe we ought to do a NoStarch Teach Kids Python on EV3 book :-)

pootle, 2016-01-30:
just a suggestion - make it say hello as well as print it - much more impact...

hemuda, 2016-02-08:
I am new to ev3dev & linux and I have not programmed in decade. I am trying to do this so I can have my son learn Python with EV3, rather than the GUI interface from Lego. 

I got the ev3dev installed, and ssh working with USB. How do I get Python bindings installed? I could not find documentation on that. We would need that as well, before We can try the "Hello World". Any pointers on the Python bindings installation would be helpful.

Thanks,
Hemuda

dlech, 2016-02-08:
ev3dev-lang-python is installed by default starting with the 2015-12-30 image. There is currently not a python3 version, so omit the `3` in `!#/usr/bin/env python3` that is shown in the hello world example above. Here is a python(2) example...

```python
!#/usr/bin/env python

print "hello world!"
```

More about ev3dev-lang-python at <https://github.com/rhempel/ev3dev-lang-python> and <http://python-ev3dev.readthedocs.org/en/latest/index.html>.

hemuda, 2016-02-09:
Thanks dlech. 

I will follow the python instructions, and give them a try. It would be good to mention on the Wiki/documentation that Python bindings are already installed in the 12/30 ev3dev release.

dlech, 2016-02-09:
If you can point us to the place where it says otherwise, we can fix it.

hemuda, 2016-02-09:
It does not say its not included, in any official page (may be I got it from unofficial pages or old documentation). But the new documentation also does not say its included.

It would be good to mention that in "Getting Started" page in Step 5: "Do something awesome" http://www.ev3dev.org/docs/getting-started/

Alternatively it would be good to clearly state that on the library page "http://www.ev3dev.org/docs/libraries/" that the Unified libraries are already part of the 12/30 release. 

Thanks.

hemuda, 2016-02-09:
Is "Using the EV3 hardware drivers" under "Step 5" of Getting started page, meant for developers adding the language bindings (or wanting to use system/kernel functions), or for programmers, who just want to use Ev3 with some programming language? It's not clear from the current documentation.

Thanks.

rhempel, 2016-02-09:
That's a big-pictre view of the drivers - we now have language specific bindings (just Python for now) auto-generated from a specification.

Here are the Python binding docs:

[http://python-ev3dev.readthedocs.org/en/latest/?badge=latest]
[http://python-ev3dev.readthedocs.org/en/latest/spec.html]

WasabiFan, 2016-02-09:
I think this is a very good point -- we should really have the main "getting started" track take users to information regarding language bindings instead of to the driver info. No need to put info on the drivers front-and-center when most people won't ever interact with them directly.

rhempel, 2016-02-09:
[Make it so!][http://comicalconcept.com/illustrations/make-it-tso]

dlech, 2016-02-09:
@rhempel

GitHub tip: use angle brackets around urls.

    <http://www.ev3dev.org>

Result: 

<http://www.ev3dev.org>

Unless you want to create a link.

     [link](http://www.ev3dev)

Result: 

[link](http://www.ev3dev.org)

hemuda, 2016-02-10:
One other confusion was "Step 4", where I had to connect EV3 to the network. The documentation talks about connecting to Internet; but never tells why I would like to connect to Internet. From a user background who knows Lego EV3 SW, I could not figure out why I would want EV3 to connect to Internet. 

We should explain that we need to connect ev3 to network, so we can send commands (via ssh) to it via network. And connection to internet helps to install packages that user may need.

Thanks,
Hemuda

WasabiFan, 2016-02-10:
@hemuda you might be interested in reading through #477. I have been hoping to get feedback from users like you who come into the getting started guide with no preconceptions, and this feedback is very helpful -- there have probably been many people who had the same complaints before but did not communicate the potential improvement ideas to us. As maintainers, we grow blind to things as basic as explaining the reasons that one might want to connect to a network. I think it would be best if you opened a new issue for each potential improvement that you find, so that when we get the time we can track what has been completed.

Unfortunately, I am preoccupied and will be for at least the next month, so this will be reliant on others until then. But again, thanks for the feedback!

hemuda, 2016-02-10:
@WasabiFan Sure, I will open new issues for these documentation concerns.

How do I open new issues tied to ev3dev project? Do we have separate bug areas for documentation vs the code base(per library)? I am completely new to github as well.

Thanks.

laurensvalk, 2016-02-10:
> Looking forward to get back to writing tutorials!

Instead of expanding on the current tutorials right away, I thought I'd take a step back and consider what might be useful for beginning users. Here's an [overview] (http://robotsquare.com/downloads/files/guide.pdf) of what I imagine might be an "ev3dev beginner's guide". I'm open to feedback and suggestions on this, and I probably need to put this out in editable format.

Since I haven't really worked on collaborative (immediately accessible) writing projects before, I don't know yet what would be a practical writing platform to work with. I have not yet looked at how the current documentation is written and published. 

>The documentation talks about connecting to Internet; but never tells why I would like to connect to Internet. From a user background who knows Lego EV3 SW, I could not figure out why I would want EV3 to connect to Internet. 

Coincidently, one of the proposed section above is titled "Why connect to the internet?" There are some similar background topics as well, like "what is a terminal?".

> As maintainers, we grow blind to things as basic as explaining the reasons that one might want to connect to a network.

I don't think it's that bad -- the project is still approaching its first release, and just recently became quite accessible for non-Linux users. We'll get there :)

dlech, 2016-02-10:
> How do I open new issues tied to ev3dev project?

Just click the "New Issue" button on <https://github.com/ev3dev/ev3dev/issues>.

> Do we have separate bug areas for documentation vs the code base(per library)?

Libraries (like the ev3dev-lang libraries) have their own github repositories, so if there is an issue specific to a programming language, open the issue there. For everything else, use the link above. We have tags to help keep things sorted out.

dlech, 2016-02-10:
@laurensvalk, I think the outline is great. The only thing that jumped out at me as missing is a section about Linux file permissions.

We could try Microsoft OneNote for collaboration. They offer free versions for Windows, OS X, iOS and Android and there is a decent web app too if you are on Linux. It doesn't keep track of changes quite as well as Microsoft Word, but it is close.

The documentation on our website is written mostly using the [kramdown](http://kramdown.gettalong.org/) flavor of markdown and is published using [GitHub Pages](https://pages.github.com/), which is built on [Jekyll](http://jekyllrb.com/).

rhempel, 2016-02-10:
Why use OneNote when we can just do this using plain text (markdown,restructured text, etc) for now?

dlech, 2016-02-10:
The main reason for me is that I like to use tables all the time and tables in plain text are a real pain. In fact, I am considering moving all of my "kernel hackers notebook" stuff to one note because I don't keep it up to date because trying to make it look good using markdown takes way too long.

rhempel, 2016-02-10:
Is it really that much of a challenge - I'm asking seriously. I've started to move away from rich text formats and rely more on plain text even at work.

laurensvalk, 2016-02-10:
> Why use OneNote when we can just do this using plain text (markdown,restructured text, etc) for now?

This may come down to preference, but when writing longer pieces I prefer not to be distracted by things such as visual layout and markdown code tags. More importantly, you can simply insert images, which are a core element of technical text.

I typed my book chapters in open office/MS Word. In such tools, paragraphs can be marked with paragraph styles. My publisher had a template for various types of "paragraphs", such as several heading levels, body text, bullet lists, and so on. These could then be imported into their layout programs. This process would only happen at the very last stage, where there would no longer be any changes in text. 

In our case, this type of text could be readily converted into markdown. It is perhaps less practical to have it inaccessible until completely finished, however. I could imagine we could make "early-access" versions available for those who don't mind reading across the occasional typo.

> Is it really that much of a challenge - I'm asking seriously. I've started to move away from rich text formats and rely more on plain text even at work.

The approach above is really close to plain text, in my opinion. It's essentially just text sections bordered by html/markdown tags, but you just don't see them. On the other hand, it makes the text easier to copy-edit, and tech-review.

dlech, 2016-02-10:
I echo Laurens' sentiments. Furthermore, I've spent hours refactoring large tables in plain text vs. minutes in a WYSIWYG program. It's not necessarily more challenging, but certainly a waste of time.

rhempel, 2016-02-10:
I've written a book or two myself - definitely it's much better if the publisher has a Word/OpenOffice template that has a few styles already created for you. That's how I use Word at work.

If we decide to do this, then I'm OK with Word/OpenOffice - there are certainly good reasons to do it this way.

WasabiFan, 2016-04-09:
I want to take a minute to understand what has been discussed in this issue and make sure that everything that was said is captured in a productive way.

I see a few different threads in this discussion:
- The specifics of a Python-based tutorial
- Ideas for an overall "getting started" guide for beginners
- Understanding the issue system and docs
- Formats that we want to use for the aforementioned beginner's guide

_side note: @dlech could we move the discussion on where issues should be posted and GitHub link formatting to a different issue or sum it up in one comment (i.e. copy the contents and delete the posts themselves)? It's an important discussion, and I think it should be in its own issue for future reference. Additionally, it's hard to see the thought processes here with those interjections._ 

So, focusing on the first two points there:

I think that @laurensvalk's ideas and outline for that guide are great. (I think we should add a section or two on explaining the overall workflow options -- edit files on-brick, copy them over, etc. That way, a reader understands that there are options and editors other than the one that we focus on.) The content that is outlined is, in my mind, exactly what a new user should be seeing.

With that outline available to us, I would like to shift attention to the goals of said content and the mechanics of authoring it. There has already been discussion on possible editing platforms, but I think this is premature: we have yet to outline the purpose of this guide explicitly. So, what is its purpose and what role does it play? I think that:
1. The resultant content should not be duplicated in the getting started guide or anywhere else in our docs. 
2. The material that we create should be easily accessible from the landing page of the website as well as relevant places in other pages.
3. There should be no obstacles between this "ultimate guide" and the rest of our tutorials, documentation, and help content.

Looking at this set of requirements, I find it hard to imagine that we could do what we need to do using OneNote or a similar graphical program (at least not for a final form). If we did it in a tool like that and kept it as such, our only portable option would be exporting the document as a PDF, which fails point 3 outright and makes points 1 and 2 difficult to attain. I like OneNote and have used OneNote Online for similar documentation-authoring collaborative projects, but I don't think that it is what we need in this case. If we want to write it in OneNote initially and then translate it into another format for publishing and maintenance, we can do that, but we should make sure that we have the "translation" bit figured out first.

The second option that I would consider is authoring it in a text-based format and hosting it on GitHub. The disadvantage here is that collaboration (especially in the early stages) is more difficult and would not feel natural. On the flip side, it enables us to publish it alongside the other content on our site. I think that this ability is the most important here: it checks all three of the above boxes and is the smallest disruption from what we already do.

As an added bonus, it may be possible to find a service that allows us to collaborate in a form that still is exportable to have compatibility with our site. This is something to look in to.

### My vision

I believe that we should aim to generate a final outline based off of what @laurensvalk has already created. Once we have the set of points that we want to cover, we should split them into discrete pages and categorize them as tutorials, pages, and other types of content that makes sense on our site. If the content doesn't make sense with the existing format, we can modify it as necessary. _As an overarching goal, aim for bite-size pieces that don't overwhelm but still do inform. We want it to form a cohesive piece as a whole._ We could take the existing "Getting Started" and mold it into our desired outline, splitting it into separate pages. Then we add content as necessary to fill the (expectedly large) gaps. We can start by marking everything as a "draft", and make it go live once we are happy with it.

The big thing that we lose by putting this on separate pages is the logical grouping and flow of a scrollable document. There are two things that I think we could do here (and they are not mutually exclusive):
- Add a common navigational element to all of our pages that has primary sections, subsections, and headings within pages. This could be an added bar of second-tier navigation below the primary nav or a TOC-style pinned box.
- Investigate technologies that would take the markdown that we write and combine it with headings and render it as a PDF. I'm sure this could be done as a fairly simple Python script. If we have that PDF, we can publish the "downloadable version" on the site for offline reading.
  - This, of course, would still require some investment in on-site nav, but that could be minimal.

One of the advantages of this format is that content can be published granularly as it becomes ready. A whole document is not held back because we haven't finished an intro section, for example. Whatever form this assumes, it will undoubtedly take a long time to create, so being able to publish it as it is ready gives us compounding benefits without delay.

[... cut ...]
So, those are my thoughts. Take the first section as "laying it all out on the table;" all of that is intended as an explanation of the current status and options without much opinionation. The rest is my personal view of the ideal scenario. I am happy (and intend) to have discussion on this to make sure that we come to a conclusion that everyone is happy with.

I have decided to post in this issue because I believe that, if we do decide to go with this "mega guide" idea, this issue should be renamed to focus on that as it is a superset and manifestation of the python tutorial idea. If I should delete and repost as a new issue, I can do that too.

dlech, 2017-03-10:
I think https://sites.google.com/site/ev3python/ covers this pretty well.

WasabiFan, 2017-03-10:
I disagree. While that site is great as a beginners' reference, it does not serve the purpose of a generic "getting started" guide for Python and does not recommend tools that we officially support or suggest. I think that our goal needs to be one which gives every user an easy starting point with optimal guidance and tooling which we can control.

dlech, 2017-03-10:
If someone wants to volunteer to actually do something here, we can certainly re-open the issue.

WasabiFan, 2017-03-10:
I intend to work on https://github.com/ev3dev/ev3dev/issues/788 whenever we come to consensus over what we would like to do for that project and I have time to do it; as I see it, this issue is dependent on that one, which is why I haven't bumped this in a while. If we have an agreed-upon goal for the tooling described in that issue by April 1st, I'll commit to whatever work is necessary for that editor plus a start on this tutorial.

dlech, 2017-08-17:
https://github.com/ev3dev/vscode-python-hello