ev3dev/ev3dev #494: BrickPi no tachomotor

antonvh, 2016-01-06:
I burned the latest rpi-ev3dev image, plugged in a motor, but no tacho-motor shows up in sys/class. Did I miss something?

dlech, 2016-01-06:
Please don't say "latest", list the exact name of the image file that you used. Also, are you using BrickPi or PiStorms? Did you edit config.txt? Now that there are two devices that run on RPi, neither are enabled by default.

Neither of these devices have auto-detection for motors, but there should be 4 motors in /sys/class/tacho-motor by default even if no motors are plugged in.

antonvh, 2016-01-06:
Tx! Hadn't seen that in the docs. It used to work out of the box... I'll add that to my tutorial. :)

closed 2016-01-06 by antonvh