ev3dev/ev3dev #495: Remove "first things to do" section from Getting Started?

WasabiFan, 2016-01-07:
As evidenced by #487, the "first steps" that we suggest for people to do have either become mostly obsolete since they were added or don't have any real effect on the UX. I'd say that they should be removed from that page; maybe they should have their own page, or maybe we just don't need it.

The two items in that section that I think are useful to some extent are the "change your password" section and the "update packages/kernel" section. Personally, I haven't ever changed the password on my EV3s and don't see a reason that one would, unless you are connecting it directly to the open internet. Updating packages is important in some cases, but I don't think it is a "first thing to do" -- it can take 30+ minutes to finish on the EV3 and we update packages anyway with new image releases. Past that, when we tell people to upgrade to a new kernel, we always tell them to `apt-get upgrade`, so packages will be updated when needed. As for the others, I see no real reason why an ev3dev user would want to do it at all -- it just doesn't have a noticeable effect to change the locale or date.

So, I think we should remove that section. If we want to, we can add a tutorial for updating/upgrading and a tutorial for changing the user password -- but as a first step just removing the extraneous information should be helpful.

dlech, 2016-01-07:
:shipit:

(that's GitHubeze for gitter done).

WasabiFan, 2016-01-07:
Fixed from PR. Forgot to add a fix marker thingy.

closed 2016-01-07 by WasabiFan