ev3dev/ev3dev #496: Permissions Error, Please Help!

melonhead2, 2016-01-08:
Image file used to flash SD: ev3dev-jessie-2015-12-30
Kernel version: 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015
Library: ev3dev-lang-python (from ddemidov)

I would like to start off by saying that I am extremely inexperienced with programming. This may be an incredibly simple problem, but I just don't know how to fix it. I have attached a screenshot with the error messages circled in red. Essentially, whenever I go to install the language, those two permission errors pop up. I've tried everything I know to try and fix it. Let me know what I should input in order to resolve those messages.

Thank you in advance for your suggestions and help, and for your patience with my inexperience.

[image]

dlech, 2016-01-08:
You have to run `apt-get` with root privileges. The easiest way to do that it to use [sudo](http://manpages.debian.org/cgi-bin/man.cgi?query=sudo). This tell whatever is after it to run as "root" (similar top "run as administrator" on Windows). So...

    sudo apt-get install package1 package2

[sandwitch](http://imgs.xkcd.com/comics/sandwich.png)

WasabiFan, 2016-01-08:
Also @melonhead2, `ddemidov/ev3dev-lang-python` has been deprecated in favor of [rhempel/ev3dev-lang-python](https://github.com/rhempel/ev3dev-lang-python). It is _mostly_ the same from an API perspective as the one that you are looking at, but the internals are completely different. You should use the one hosted by @rhempel because it is the one that both @ddemidov and @rhempel are working on together. The other is no longer maintained.

@ddemidov This is probably a call to action for you to create a new branch on your repo and set it as default, and just have a readme in that branch with a single bolded line that points to the new version. I think the default reaction when one sees a README with sample code or terminal commands is to read the preceeding sentence and then run the code listed, so we really just need to remove it altogether.

melonhead2, 2016-01-08:
Beautiful. That totally makes sense. Thanks so much for helping me out with that.

I am ultimately trying to create the programming for the Print3rbot and that was the version that was listed on the project page. I noticed that it had been deprecated, but wasn't sure if the new version would still allow the Print3rbot code to work properly.
[image]

ddemidov, 2016-01-08:
@WasabiFan, depreciation warning (in bold) is the first sentence in the
current readme.

dlech, 2016-01-08:
The printerbot has it's own github page (the "Source Code" link in the screenshot). You can ask more about it there.

ddemidov, 2016-01-08:
@melonhead2, @rhempel's repo probably will not work with printer bot code,
but neither will mine work with the current kernel. Best thing to do would
be to update printer bot to work with the new bindings. You can either ask
@cavenel to do that, or, even better, make a pull request to his repo.

Cheers,
Denis

WasabiFan, 2016-01-08:
> @WasabiFan, depreciation warning (in bold) is the first sentence in the
current readme.

What, you actually _read_ the first sentence in a readme? :wink: We've seen a few issues posted over the last week or two that demonstrate that people automatically sniff out the code and don't read the warnings (this turns out not to be an example of that, but there have been multiple others). You might be able to just make it an `h1` heading to achieve the same effect as blanking the repo.

ddemidov, 2016-01-08:
Worse, I've had same user open new issue in that repo *after* I told him
the repo is no longer supported in the previous issue.

Cheers,
Denis

ddemidov, 2016-01-08:
Actually, blanking the master branch might be a good idea. I'll do that.

Cheers,
Denis

melonhead2, 2016-01-08:
Thanks a bunch for all the help fellas. I really appreciate it. I know hardly anything about coding, but I'm quickly learning!

WasabiFan, 2016-01-08:
> Thanks a bunch for all the help fellas. I really appreciate it. I know hardly anything about coding, but I'm quickly learning!

Side-note: We have been discussing ways that we can simplify the getting started process on ev3dev for new users in #477. You might find that discussion interesting, and we'd love to hear your thoughts!

P.S. Remember to close this issue once you determine that your issue is resolved.

closed 2016-01-08 by melonhead2