ev3dev/ev3dev #497: A Motor instance to a non-connected motor causes exception when instance is garbage collected

pootle, 2016-01-08:
Unless I am doing something silly (I understand I am trying to connect a non existant motor) then this results in an unavoidable rude message:
> from ev3dev.auto import *
> m=Motor(OUTPUT_B)
> m.connected
False
> m=None

Exception AttributeError: "'Motor' object has no attribute '_path'" in <bound method Motor.__del__ of <ev3dev.core.Motor object at 0xb69bd8d0>> ignored

WasabiFan, 2016-01-08:
This repo is for the ev3dev kernel and core systems. If you are having an issue with one of the Python libraries, please open an issue on the library-specific repo instead.

pootle, 2016-01-10:
Oops quite right, ddemidov has already got a fix over there, so I'll just close this

closed 2016-01-10 by pootle