ev3dev/ev3dev #498: polling infrared sensor, should there be callbacks?

pootle, 2016-01-10:
I've just started on some code to provide simple control from the IR remote on my ev3 using ev3dev with python (although this doesn't appear to be language specific)

I see I can arrange for callbacks when any button state changes, but I still have to poll by calling process() on the sensor instance.

This means I either have to integrate a poll into some other ongoing loop, or run up a separate process to poll so the callbacks are triggered.

Would it be realistic to have this functionality in the ev3dev sensor code rather than having the complication / overhead of this in the app? - then just setting up the callbacks would result in them being triggered (admittedly from a different execution thread - so care is needed) without further action.

Oh yes - and timestamps on them as well........

theZiz, 2016-01-10:
Hm, how should such a file based controling approach deal with feedbacks? I have no idea, how an application could register itself on the driver.

dlech, 2016-01-10:
This is essentially a duplicate of #104.

For data logging and event driven programs using sensors, I would like to implement [iio](https://wiki.analog.com/software/linux/docs/iio/iio) devices on top of the existing sensor drivers. This would give you a character device that you can poll/select(or blocking read on separate thread) for events and would have timestamps.

This is a pretty low priority for me right now though. (Anyone need a GSoC project?)

pepijndevos, 2016-01-14:
How hard is this to implement for an okay C programmer with no kernel experience? I keep looking for this feature every single time I look at ev3dev.

What is currently the fastest way to read out sensors? Just a tight loop using the C/C++ API?

rhempel, 2016-01-14:
There is a kernel call that you can use to indicate a change on a sys file attribute - but your language needs to support the userland side of it using `select` or some other mechanism.

The harder part is providing a mechanism that describes *when* you want the sensor to tell you something interesting has happened.

If your scripting language supports multi-threading or cooperative multitasking, then you can simply poll the sensor - most of them update a maximum of about 100 times a second which is pretty fast considering the kind of robots we can build with LEGO.

dlech, 2016-01-14:
> How hard is this to implement for an okay C programmer with no kernel experience?

I would expect it to take about two to four weeks working full time for this level of experience. It's not really hard, just lots to learn.

> What is currently the fastest way to read out sensors? Just a tight loop using the C/C++ API?

Yes, that is currently the fastest way. As @rhempel mentioned, you will most likely be reading faster than the values are updated.

piersh, 2016-01-16:
It would be great if you could somehow specify a minimum delta for value-change events, possibly with a configurable-length EMA applied before comparisons for noise reduction.

dlech, 2016-01-16:
What is EMA?

piersh, 2016-01-17:
EMA is 'exponential moving average'.  It's trivial to implement and gives better smoothing than SMA (mean of previous N).

dlech, 2017-10-02:
Notification for EV3/NXT touch sensors is implemented.

https://github.com/ev3dev/lego-linux-drivers/commit/1210c1b87f1d90c77d7d1842d9a35f001b0dd287
https://github.com/ev3dev/lego-linux-drivers/commit/ec1df92b8580dded40d8d1c3ad51c759efd5086e

for other sensors, the plan is to eventually get mainline drivers using the `iio` subsystem.