ev3dev/ev3dev #499: Units for attributes:

jabrena, 2016-01-10:
Hi,

In what units the battery show current & voltage?

``` bash
root@ev3dev:/home# cat /sys/class/power_supply/legoev3-battery/current_now
162666
root@ev3dev:/home# cat /sys/class/power_supply/legoev3-battery/voltage_now
8042133
```

I tried to find the info in the website & wiki but I didn't find it.

Juan Antonio

ddemidov, 2016-01-10:
[Here](https://github.com/ev3dev/ev3dev-lang/blob/develop/wrapper-specification.md#direct-attribute-mappings-5) it says microamps and microvolts. Not a canonical reference, but could be useful sometimes :).

dlech, 2016-01-10:
microvolts and microamps. It all comes from https://www.kernel.org/doc/Documentation/power/power_supply_class.txt

dlech, 2016-01-10:
Ha, @ddemidov was 3 seconds faster AND he got them in the right order.

jabrena, 2016-01-10:
Many thanks team. :)

I am writing the Javadoc for the java library and I don't want to invent the units. I will link the url.

closed 2016-01-10 by jabrena