ev3dev/ev3dev #500: cannot get robot commander on my iphone to pair with ev3dev

dlech, 2016-01-10:
From @dwalton76 on #329:

> 
> 
> Regarding bluetooth and robot commander, I jumped through few hoops but still cannot get robot commander on my iphone to pair with ev3dev. I did:
> 
>     # Added /etc/machine-info
>     PRETTY_HOSTNAME=EV3
>     
>     # Edited /etc/bluetooth/main.conf
>     Name=EV3
>     
>     
>     # Use bdaddr to change bluetooth MAC:
>     http://blog.petrilopia.net/linux/change-your-bluetooth-device-mac-address/
>     sudo apt-get install libbluetooth-dev
>     wget http://www.petrilopia.net/wordpress/wp-content/uploads/bdaddrtar.bz2
>     tar -xvf bdaddrtar.bz2
>     cd bdaddr
>     make
>     
>     # To change the address
>     robot@ev3dev[bdaddr]# ./bdaddr 
>     Manufacturer:   Texas Instruments Inc. (13)
>     Device address: D0:07:90:B6:30:8C
>     robot@ev3dev[bdaddr]# 
>     robot@ev3dev[bdaddr]# sudo ./bdaddr -i hci0  8C:30:B6:53:16:00
>     [sudo] password for robot: 
>     Manufacturer:   Texas Instruments Inc. (13)
>     Device address: D0:07:90:B6:30:8C
>     New BD address: 8C:30:B6:53:16:00
>     
>     Address changed - Reset device now
>     robot@ev3dev[bdaddr]# 
>     robot@ev3dev[bdaddr]# 
>     robot@ev3dev[bdaddr]# sudo service bluetooth restart
>     robot@ev3dev[bdaddr]# 
>     robot@ev3dev[bdaddr]# ./bdaddr 
>     Manufacturer:   Texas Instruments Inc. (13)
>     Device address: 00:16:53:B6:30:8C
>     robot@ev3dev[bdaddr]# 
> 
> bdaddr is a little buggy, the first time I did this I ran "sudo ./bdaddr -i hci0 00:16:53:B6:30:8C" and the "New BD address" looked correct but after restarting the bluetooth service the MAC was 8C:30:B6:53:16:00.
> 
> Robot Commander still doesn't see my ev3 though. Who knows they may be looking for some specific signature. Any ideas...other than hacking together a bluetooth sniffer to see how robot commander interacts with the normal EV3 software?

dlech, 2016-01-10:
1. #329 is not implemented yet, so even if you could get your iPhone connected paired, it wouldn't do anything.
2. We haven't implemented anything like the checkbox on the standard firmware that put the EV3 in iDevice mode. Apple devices apparently have their own Bluetooth thing going on.
3. I'm glad you figured out how to change the MAC address. I think the LEGO firmware has a hacked version of hciattach that lets you set the MAC address then. I think it might be useful to have ev3dev use the same MAC address as the standard firmware by default.

Just to keep things organized, let's keep this issue about 2. (pairing ev3dev with iDevices). We should open a new issue for further discussion of 3.

I don't have any iDevices, so someone that has one is going to have to fix this (or someone can send me their old iThing and I'll have a crack at it).

smcri, 2016-01-16:
when is #329 getting implemented?

dlech, 2016-01-16:
Whenever someone decides to spend the time to make it work.

dwalton76, 2016-01-22:
To clarify on this a bit, I could get my iphone to pair with ev3dev.  What I could not do was get the Robot Commander app on my iphone to pair with ev3dev.

JorgePe, 2016-06-17:
@dwalton76 I've never used Robot Commander but I already installed on my Android Phone.
Pure BT is much more complex than BT LE but maybe I'm lucky if you guide me. What do you want me to do?

dlech, 2016-11-18:
I have been looking at the ICs on the EV3 and discovered that there is a special chip (U29) required for communicating with iDevices. See <http://stackoverflow.com/questions/17458987/is-mfi-chip-required-for-communication-with-ios-devices-over-serial-port-profile>.

So, in a addition to getting lms2012 working, we also have to figure out how this chip works.

dlech, 2018-04-20:
I'm changing this to `wontfix` since Apple is very secretive about their MFi technology. As a workaround, you can use a Bluetooth dongle with BLE (low energy). Apple devices do not restrict BLE connections.

closed 2018-04-20 by dlech