ev3dev/ev3dev #501: CI not picking up empty links on ev3dev.github.io

dlech, 2016-01-13:
In ev3dev/ev3dev.github.io#97, @WasabiFan and I pointed out a few empty links. When we were getting CI going for the website, I distinctly remember having to fix and empty link like `[something]()`, but now CI does not seem to be catching it.

WasabiFan, 2016-01-13:
Can we get a definitive build log and markdown example? If it's still there in ev3dev/ev3dev.github.io#97, can you trigger a manual build so we can confirm?

dlech, 2016-01-13:
I justed added `[bad link]().` to the last paragraph of `support.md` and it fails for me locally with..

```
- ./support/index.html
  *  anchor has no href attribute (line 202)
htmlproof 2.6.1 | Error:  HTML-Proofer found 1 failure!
```

I'll create a PR to see if it fails on travis.

dlech, 2016-01-13:
Fails as expected. https://travis-ci.org/ev3dev/ev3dev.github.io/builds/102003468

dlech, 2016-01-13:
> If it's still there in ev3dev/ev3dev.github.io#97, can you trigger a manual build so we can confirm?

This is failing on an image link, but not the others. Maybe htmlproofer stops because of the other error before it gets to this sort of error.

WasabiFan, 2016-01-13:
Yeah... that's definitely possible. As long as it doesn't let errors through, I don't think it's something we need to worry about.

dlech, 2016-01-13:
Yup. Tested locally and that was it.

closed 2016-01-13 by dlech