ev3dev/ev3dev #502: dc-motor class ignored ramp_up_sp when calculating start time of ramp down

rhempel, 2016-01-13:
While implementing the changes to the tacho-motor class, I discovered a defect in the dc-motor class.

If we have the following attributes set:

- `ramp_up_sp` is 4000 
- `time_sp` is 8000
- `ramp_down_sp` is 4000

And we issue the `run-timed` command, the total run time should be 16 seconds. However the `dc-motor` class  gets a run time of only 12 seconds because of this code in `command_store()`:

```
			if (motor->command == DC_MOTOR_COMMAND_RUN_TIMED)
				schedule_delayed_work(&motor->run_timed_work,
					msecs_to_jiffies(motor->active_params.time_sp);
```
We forgot to add `ramp_up_sp` to `time_sp` in this calculation. I've fixed this in the current dc-motor driver, but I'd like to wait for a PR until I get the rest of the tacho driver cleaned up.

- `ramp_up_sp` is 4000

dlech, 2016-01-13:
12 seconds sounds correct to me. The timer for `run-timed` starts when you send the command. So, the motor ramps up for 4 seconds, then continues for 4 more seconds at which time the 8 second timer from `time_sp` is expired. This triggers a `stop` command internally, so the motor starts to ramp down, which takes another 4 seconds.

The ramp up time should not be added to the run time in `time_sp`.

dlech, 2016-01-13:
If you add the ramp time, then you can never run the motor for a duration of less than the full `ramp_up_sp`, which would be undesirable.

dlech, 2016-01-13:
Another way to look at it is

    echo 8000 > time_sp
    echo run-timed > command

should be the non-blocking equivalent of

    echo run-forever > command
    sleep 8
    echo stop > command

The motor should run for the same duration in both cases.

rhempel, 2016-01-13:
In that case, shouldn't we start the ramp down at the smaller positive result of:

- `time_sp` - `ramp_down_sp`
- intersection of the ramp up and ramp down lines at a particular duty cycle or speed

If I understand correctly, the `time_sp` is the TOTAL run time, so the current operation that runs for 12 seconds is also wrong...

dlech, 2016-01-13:
Nope , `time_sp` is the delay before sending the `stop` command, not the total run time.

rhempel, 2016-01-13:
I can see why that works now - it allows the stop command to be sent even while the motor is in the ramp up phase, and will cause the motor to begin ramping down....

OK, so no defect, just a mildly surprising operation. Closing defect.

rhempel, 2016-01-14:
I'm reopening this because there are some situations where it's hard to figure out how long the motor will actually run for. As an example:

- `ramp_up_sp` = 4000
- `time_sp` = 3000
- `ramp_down_sp` = 4000

The total run time here depends on the `duty_cycle_sp`:

- `duty_cycle_sp` = 25% -> total time is 4000 msec.
- `duty_cycle_sp` = 50% -> total time is 5000 msec.
- `duty_cycle_sp` = 75% -> total time is 6000 msec.
- `duty_cycle_sp` = 100% -> total time is 6000 msec.

This is non-obvious - you need to do some math (easy math) to get the total run time. I feel pretty strongly that `time_sp` should be to total time the motor is running, including ramp-up and ramp-down times.

Is there an example of a commercial system where the run time is defined the way it woks now, which does not include ramp down time?

dlech, 2016-01-14:
I did work on industrial systems, which I assume are the same as commercial systems. So, to go with that example, we need to talk about "call" vs "run". When we send a command, it can change the `call`. If the motor is actually running (powered), this is the `run`. Our `run` is read via the `state` attribute. (We also have a `ramping` flag to indicate ramping).

We don't actually have a `call` flag right now, but for the sake of example, I'm going to pretend that we do.

    echo 1000 > ramp_up_sp
    echo 1000 > ramp_down_sp
    echo 100 > duty_cycle_sp
    echo run-forever > command
    cat state
    ----> called running ramping
    sleep 1
    cat state
    ----> called running
    echo stop > command
    cat state
    ----> running ramping
    sleep 1
    cat state
    ----> 

The run time that you are used to in commercial systems in the `running` flag from our `state` attribute and this is what you should be looking at to know for certain that your motor is running rather than trying to guess from the commands that you send.

I still think that the time in `time_sp` should be the time that the motor is "called" to run and not the time that it is actually running.

The point of the `run-timed` command is that it is asynchronous, so why does it matter if you know exactly when it will stop?

dlech, 2016-01-14:
Another interesting outcome of using ramping with the speed_sp and run-timed the way I am advocating is that given equal ramp_up_sp and ramp_down_sp for any value, the number of rotations of the motor remains the same. The rotations in the area under the curve of speed over time.

[image]

So, if you have a wheeled robot and you are using time to make your robot travel an exact distance, having time_sp control the "call" rather than the "run" allows you adjust only the ramp without changing the distance that the robot travels.

rhempel, 2016-01-14:
There - that's the exact explanation that I needed to finally be convinced that the way you are advocating that `time-sp` is correct! I knew you had worked on industrial motor control so I figured that you'd have a fantastic nugget like this, I just had to dig for it. We'll want to put that in our API notes!

By the way, which drawing package did you use? :-)

dlech, 2016-01-14:
> By the way, which drawing package did you use?

I believe it is called Bond with the C-Thru Ruler Company plugin.

ddemidov, 2016-01-14:
>> By the way, which drawing package did you use?

> I believe it is called Bond with the C-Thru Ruler Company plugin.

Another option: http://matplotlib.org/xkcd/examples/showcase/xkcd.html :)

rhempel, 2016-01-14:
Aaand now I can change the TODO in the proposal to require Python scripts to draw the charts instead of LibreOffice templates :-) 

I never stop being amazed at this group!

rhempel, 2016-01-15:
There is still a minor issue around the ramp down not quite getting `duty_cycle_sp` to 0 ( think this is either an off by one or order of operations issue) but I also have a question about ramping down when there is a change in sign of duty cycle.

Should we use `ramp_down_sp` to lower absolute values of duty cycle, and then use `ramp_up_sp` towards higher values? For example:

- `ramp_up_sp` = 1000
- `ramp_down_sp` = 2000

If we are in `run-direct` command and we change duty cycle from 100% to -100%, should we ramp down for 1 second to 0 and then ramp up for 2 seconds to -100? Or just ramp down for 2 seconds to -100?

dlech, 2016-01-15:
> Should we use ramp_down_sp to lower absolute values of duty cycle, and then use ramp_up_sp towards higher values?

Yes, ramp up is away from 0, ramp down is towards 0.

> If we are in run-direct command and we change duty cycle from 100% to -100%, should we ramp down for 1 second to 0 and then ramp up for 2 seconds to -100? Or just ramp down for 2 seconds to -100?

I thought we decided that we `run-direct` would not include ramping since it is "direct".

dlech, 2016-01-15:
If you mean `run-forever`, then ramping from 100% to -100% should take ramp_down_sp milliseconds from ramp to 0 and ramp_up_sp milliseconds to ramp to -100% for a total of 3 seconds.

rhempel, 2016-01-15:
We did - but I'm making incremental changes. I guess we'll need to update the `dc-motor` driver to take away ramping in `run-direct` too.

So I'll need to introduce "two phase ramping". Right now we have the 2 second ramping implemented in a single ramp-down phase.

Time to go to sleep ...

rhempel, 2016-01-17:
Here's another problem. Let's say the motor is running at -50% duty cycle, and we send a `run-timed` command after changing the duty cycle to +50%.

Does the motor slow down to 0, then start the timed run? If yes, this makes the two phase ramping code a bit easier - otherwise it means a bit more juggling in the ramp work function and the command_store function.

dlech, 2016-01-17:
Nope, the timer starts as soon as you send the command, always. The implementation of the `run-timed` command should be that it just sends `run-forever` and `stop` commands to the underlying drivers. Underlying drivers should not be aware that a `run-timed` command even exists.

rhempel, 2016-01-17:
Sure, but we're implementing ramping in the tacho-motor driver, not the underlying driver, right? So in the example above, the tacho-motor driver would need to:

1. Ramp the motor down to 0% - which would take half the `ramp_down_sp` value
2. When the motor is down to 0%, start a ramp up but subtract the time I took ...

Oh, wait, I see my problem. I don't need to recalculate the delayed work for `run-timed` at all. When I cross the 0 point in the ramping operation, I recalculate to use the `ramp_up_sp`.

dlech, 2016-01-17:
> Sure, but we're implementing ramping in the tacho-motor driver, not the underlying driver, right?

Yes, it would certainly be better this way if we can make it work. Sounds like you have is mostly figured out now.

rhempel, 2016-01-17:
I'm implementing the `run-timed` and `run-forever` ramping (duty cycle or speed) in the `tacho-motor` driver, and removing ramping from run-direct.

The `run-*-pos` ramping may end up being in the specific motor driver though...

dlech, 2016-01-28:
I think we decided this is a non-issue. Feel free to reopen.

closed 2016-01-28 by dlech

rhempel, 2016-01-28:
I'll be closing a lot of this type of issue after I fix the tacho PR up so it can be merged.