ev3dev/ev3dev #503: lua

smcri, 2016-01-13:
lua seems to be missing from ev3dev officially supported languages.Why?

WasabiFan, 2016-01-13:
The lua library that we had was outdated and the maintainer for it was unable to keep working on it. @rhempel has been working on a new version at <https://github.com/rhempel/ev3dev-lang-lua>. He will need to comment on the status/usability of that library.

rhempel, 2016-01-13:
It's not useable yet - I'm focusing on the tacho-motor updates right now, then the event2sensor interface so that we can read buttions without requiring a language specific ioctl binding to read buttons.

dlech, 2017-10-24:
I would like to make Lua a first-class language in ev3dev-stretch. Any Lua fans out there still listening/interested?