ev3dev/ev3dev #504: aplay lags behind generated audio

pepijndevos, 2016-01-14:
I'm making a theremin using the IR sensor and the EV3 speaker. I wrote a simple program that polls the IR sensor and outputs sound to aplay. However, somewhere along the way it buffers up to 20 seconds of sound. So when I move my hand in front of the IR sensor it takes almost 20 seconds to change pitch.

I'm not sure where the bug is. It might be my code, the EV3 library, aplay, the sound driver. I don't know. But I added some print statements and as far as I can tell the sound I generate updates instantaneously.

I tried various combinations of buffering and flushing, but none produce smooth playback and reasonable update rate.

``` C
#include "ev3c.h"
#include <stdio.h>
#include <math.h>
#include <stdint.h>

int main() {
  const double R=8000; // sample rate (samples per second)
  double F=440; // frequency of middle-C (hertz)
  double Fp = F;
  double V=127; // a volume constant
  double t; // doudle counter, yeaaaa

  //char buf[800];
  //setvbuf(stdout, buf, _IOFBF, 800);

  int i;
  //Loading all sensors
  ev3_sensor_ptr sensors = ev3_load_sensors();
  ev3_sensor_ptr prox1 = sensors;
  ev3_mode_sensor(prox1,0);
  ev3_open_sensor(prox1);
  while(1)
  {
    ev3_update_sensor_val(prox1);
    Fp = F;
    F=(F*0.7)+(((double)prox1->val_data[0].s32+400)*0.3);
    t*=Fp/F;

    fprintf(stderr, "%d, %f\n", prox1->val_data[0].s32, F);
    
    for ( int i=0; i<800; i++ ) {
      t+=1;
      uint8_t temp = (sin(t*2*M_PI*F/R)+1)*V;
      putchar(temp);
    }
    fflush(stdout);
  }
  //Let's delete the list in the very end. It will also close the
  //sensors
  ev3_delete_sensors(sensors);
  return 0;
}
```
It was compiled and ran as follows

```
cc -std=gnu11 prox.c -o prox -I ev3c/ ev3c/lib/ev3c.a -lm
./prox | aplay --period-size=800 --buffer-size=8000
```

ddemidov, 2016-01-14:
Since your program produces more data than aplay is able to consume in real
time, would it be a good idea to put some sleep into your loop?

pepijndevos, 2016-01-14:
How much sleep though? One ms to few and it will slowly but surely fill the buffer. One ms to many and you'll get buffer undrruns.

It seems that somewhere in the audio stack there is a buffer that eventually fills up and blocks writes. It's just so huge that it fits over 10 seconds of audio. I was hoping someone could tell me which buffer exactly, and how to decrease it.

ddemidov, 2016-01-14:
I think you can put as much as is indistinguishable for human ear. It
should not be a problem if your program lags a bit, it will just lead to
coarser sampling rate.

pepijndevos, 2016-01-14:
Another interesting observation: If I remove the buffer parameters from aplay I get output like this

```
underrun!!! (at least 0.284 ms long)
9, 408.999999
9, 408.999999
9, 408.999999
9, 409.000000
9, 409.000000
underrun!!! (at least 0.271 ms long)
9, 409.000000
9, 409.000000
9, 409.000000
9, 409.000000
9, 409.000000
underrun!!! (at least 0.284 ms long)
9, 409.000000
9, 409.000000
9, 409.000000
9, 409.000000
9, 409.000000
```
But the audio still lags! So aplay can't even get enough data, but still manages to buffer endlessly. I also noticed that if I wave my hand in front of the sensor at program start, my waves get played back at less than half the speed.

I'll try to add some sleep. But I don't get what you're saying. I'm not setting the frequency, I'm generating the wave form. So if I'm to slow the audio will stutter.

pepijndevos, 2016-01-14:
Just as I thought, with `usleep(100000);` it stutters horribly. With `usleep(50000);` it slowly fills up the buffer anyway. With `usleep(80000);` it does a bit of both.

ddemidov, 2016-01-14:
I have to admit I have no experience in sound processing. I did try to create a theremin though: https://github.com/ddemidov/ev3dev-lang-python/blob/master-deprecated/demo/thereminvox.py. It did work without any problems. I think the main difference to your approach is that I used a call to `beep` utility (with explicit frequency parameter), and the call was synchronous.

dlech, 2016-01-14:
Why not use libasound instead of piping into aplay? Here's the docs: <http://www.alsa-project.org/alsa-doc/alsa-lib/pcm.html>. At the bottom of that page, there is mention of a sample sine wave generator. I bet you could use that and plug in your sensor generated data instead of the generated sine wave data.

pepijndevos, 2016-01-14:
I'm looking into that now. Trying to adapt http://www.alsa-project.org/alsa-doc/alsa-lib/_2test_2pcm_min_8c-example.html

But I'm afraid it'll just end up in the same buffer. Except now I maybe have more control and feedback mechanisms. It seems you can request the amount of frames available: http://www.alsa-project.org/alsa-doc/alsa-lib/group___p_c_m.html#ga577b4d51e08d94930a05bbe73291ed2a

So anyway, this huge buffer is probably a "feature", not a bug. So I'll close this and mess with libasound. Thanks for the input.

closed 2016-01-14 by pepijndevos

dlech, 2016-01-14:
> Except now I maybe have more control and feedback mechanisms

Yes, this is key. Also look at `snd_pcm_hw_params`. You don't have to use the full buffer size. You can specify smaller buffer and period sizes.

pepijndevos, 2016-01-14:
Works http://studl.es/2016/01/theremin/

dlech, 2016-01-14:
The :musical_note: might sound a bit better if you used a higher bit rate and 16-bit samples. The ideal bitrate for EV3 hardware (using ev3dev) is 22050.

pepijndevos, 2016-01-16:
I tried your suggestion, and it's better, but I get a ton of buffer undrruns. 16000 just barely works with a small glitch here and there. It uses like 80% CPU calculating all the sine waves.

I might be wrong, but it seems the EV3 either does not have an FPU, or GCC is not configured to use it

    robot@ev3dev:~$ gcc -c -Q -march=native --help=target | grep float-abi
      -mfloat-abi=                          soft

dlech, 2016-01-16:
It does not have a FPU. :cry: