ev3dev/ev3dev #505: using keyboard to type in ev3dev

smcri, 2016-01-14:
can you use keyboard to type in ev3dev?

rhempel, 2016-01-14:
The bigger problem is can you use the screen to see anything :-) Yes, you can use a USB keyboard to type characters, you just need to tell your program where to get the input from.

smcri, 2016-01-14:
i used a wireless keyboard.
it worked.
i think i pressed alt + left arrow and i managed to access the command line from the ev3 brick.

dlech, 2016-01-14:
Yes, you can use a USB or Bluetooth keyboard. You can use it to navigate around brickman using the up, down, left, right, enter and backspace buttons. You can also get a terminal using ALT+CTRL+F6. Other F keys could get you a terminal too, but F6 is *always guaranteed* to get a terminal. Use ALT+CTRL+F1 to return to the main user interface (brickman).

smcri, 2016-01-14:
can you code in python directly on the brick?

dlech, 2016-01-14:
Yes. We already discussed this in #466.

jabrena, 2016-01-14:
Hi, 

One alternative could be show the shell in a smartphone or a tablet to code.

dlech, 2016-01-28:
I believe the original question has been sufficiently answered.

closed 2016-01-28 by dlech