ev3dev/ev3dev #506: "fast" I2C

dlech, 2016-01-15:
The long awaited blog post from @gloomyandy: https://lejosnews.wordpress.com/2016/01/15/i2c-on-the-ev3/

**Secrets of the EV3 I2C revealed!**

---

Seriously though, this article has some great tips on how we can get 100kHz I2C working on ev3dev.

gloomyandy, 2016-01-15:
:-) Actually what I'd really like to do is get that PRU that is sitting
doing nothing and use that to do the i2c... So if anyone fancies giving
that a go let me know!

dlech, 2016-01-15:
You may be in luck. I'm working on the EVB on BeagleBone Black. BeagleBone does not let you use FIQ, so we can't reuse our existing code from the EV3. The AM355x processor on the BeagleBone has a couple PRUs though, very similar to the ones in the EV3. So, if we want I2C on the EVB, the PRU looks like the best bet. If it ever gets done, it should be easy to port to the EV3.

gloomyandy, 2016-01-15:
Very cool. I'll keep an eye out for it! If I do ever get around to doing
anything I'll be sure to let you know...

Andy