ev3dev/ev3dev #507: Navigation bar partly covers page heading.

ddemidov, 2016-01-16:
I first saw this on android chrome:

[image]

But, if you make the page thin enough, it also happens on desktop chrome:

[image]

Also, the search bar is not visible when the page is so thin, although there is enough place for it.

moriarty, 2016-01-16:
Changing margin in .page-heading and adding padding top (to stop the Edit on Github from touching the header) fixed this with Chromium "inspect element". But I don't know where to make that change in the ev3dev/ev3dev.github.io repository, either directly in the bootstrap.css or is it being generated from less?

From following the issues, I think @WasabiFan is the one to have the final say

These settings work:

    .page-header {
        padding-top: 1px;
        padding-bottom: 9.5px;
        margin: 60px 0 21px
        border-bottom: 1px solid transparent;
    }

These settings are the orginal:

    .page-header {
        padding-bottom: 9.5px;
        margin: 42px 0 21px
        border-bottom: 1px solid transparent;
    }

moriarty, 2016-01-16:
Also, I think this is a duplicate issue with #480

ddemidov, 2016-01-16:
It does seem a duplicate, closing.

closed 2016-01-16 by ddemidov