ev3dev/ev3dev #510: /sys/class/lego-port empty on BrickPi

m1ngy, 2016-01-17:
Hi there! First, thanks for adding BrickPi support! 

uname: 4.1.15-9-ev3dev-rpi2
flashed image: rpi2-ev3dev-jessie-2015-12-30.img
fully upgraded at date: 2016-01-17 

I am running this on a BrickPI+ v2.6, which i literally received yesterday. None of the brickpi kernel modules are getting loaded automatically and after manually modprobing brickpi, brickpi_battery, brickpi_i2c_sensor and i2c-dev /sys/class/lego-port remains empty. 

No sign in dmesg besides: Registered BrickPi line discipline. (28)

Any pointers to where to start to debug the problem are much appreciated.

dlech, 2016-01-17:
Did you edit `config.txt` to enable the BrickPi overlays?

m1ngy, 2016-01-17:
Of course I didn't *facepalm*. 

To recap what i did to make it work (for anyone who's as unfamiliar as I am):

1. Locate file /boot/flash/config.txt
2. Uncomment lines follow the comments starting with: dtoverlay=brickpi ... 

Thanks!

WasabiFan, 2016-01-17:
@dlech Is this something that could/should go into ev3dev-config? This seems like a pretty common question.

dlech, 2016-01-17:
Yes, that would be helpful.

m1ngy, 2016-01-17:
One more thing. Is there somewhere a "getting started guide" with a short example. 

Right now I am trying wild permutations of setting modes to either mode or set_device, but none of them triggered a lego sensor device node to appear. :) 

If someone would give me a quick intro I am more than happy to write one (in case none exists).

dlech, 2016-01-17:
What type of sensor are you trying to use? #374 might provide some hints.

dlech, 2016-01-17:
Also, please always open a new issue if you have a new question.

m1ngy, 2016-01-17:
Sure sorry for hijacking this issue.

It's the ev3 us sensor. I tried setting "set_device" to ev3-uart but always receive access denied if I'm not setting "mode" to the same value.

dlech, 2016-01-17:
`ev3-uart` is the mode, sot `echo ev3-uart > mode`. Then from [this table](http://www.ev3dev.org/docs/sensors/#supported-sensors), we see that the LEGO EV3 Ultrasonic sensor driver name is `lego-ev3-us`, so `echo lego-ev3-us > set_device` after you set the mode.

m1ngy, 2016-01-18:
Thanks a bunch!

closed 2016-01-18 by m1ngy