ev3dev/ev3dev #511: Steps to put c++ con & my c++ code into the EV3 MINDSTORMS.

ADVANCESSSS, 2016-01-17:
This is for my important AI project. I want to do this the easy way. I read the easy way is to "put the c con into the brick", I'm guessing that means c++ con for me & my c++ code. But um, how? I'm like a newb with this stuff...

dlech, 2016-01-17:
sudo apt-get update
    sudo apt-get install build-essential

But in addition to being easy, compiling on the EV3 itself is also the very slow way.

ADVANCESSSS, 2016-01-17:
My AI is a small program. Though big memory it'll get.

All I know is if it requires me to do lots of copy paste libs stuff I'm just out man, I'd rather buy the 10,000 robot to just upload my c++ all easy.

moriarty, 2016-01-19:
### **TL;DR**

@ADVANCESSSS, reading your questions make me think you'll appreciate a **TL;DR** section.

As @dlech mentioned, compiling on the EV3 itself is very slow. 

1. You should start by *reading* the tutorial. once you've *read* the tutorial, then maybe follow the tutorial. http://www.ev3dev.org/docs/tutorials/using-brickstrap-to-cross-compile/

2. The tutorial uses some ```scp``` to move the files to the brick. If you aren't familiar with the command line, or just looking for a GUI based tool, then see issue #477 and specifically this comment: https://github.com/ev3dev/ev3dev/issues/477#issuecomment-169096711

### The Long Version

> This is for my important AI project. I want to do this the easy way. I read the easy way is to "put the c con into the brick", I'm guessing that means c++ con for me & my c++ code. But um, how? I'm like a newb with this stuff...

I don't know how important this AI project really is. If you are, as you put it so kindly, a "newb", then it's likely not really an "important AI project"- it sounds more like homework. If it is homework, then just ask your instructor, they won't bite, and surprisingly they often answer the question much more clearly than the internet. 

I'm not familiar with "c con" or "c++ con". I thought that was the annual c++ conference. Let's just assume you mean: "how to compile a c or c++ program and put it onto the brick"

1. You should start by *reading* the tutorial. once you've *read* the tutorial, then maybe follow the tutorial. http://www.ev3dev.org/docs/tutorials/using-brickstrap-to-cross-compile/

2. The tutorial uses some ```scp``` to move the files to the brick. If you aren't familiar with the command line, or just looking for a GUI based tool, then see issue #477 and specifically this comment: https://github.com/ev3dev/ev3dev/issues/477#issuecomment-169096711

> My AI is a small program. Though big memory it'll get.

That first sentence would make most people in the AI community smile... I just wanted to warn you that the EV3 only has 64Mb of RAM.

To put this in prospective, I've graded final assignments which require 20+ GB of RAM, and they're not event "important AI projects", just relatively simple AI projects, just poorly programmed. I don't want to turn you away from developing on the EV3, or artificial intelligence, I just want to warn you. Students are often surprised when they design something poorly and it quickly uses more than the 8Gb of RAM on their laptops. 

> All I know is if it requires me to do lots of copy paste libs stuff I'm just out man, I'd rather buy the 10,000 robot to just upload my c++ all easy.

Let me know which robot you find which allows you to "upload [your] c++ all easy". I work with robots in the 30,000 - 350,000+ Euro range... I use basically the same methods to get my code onto the EV3 as I do on more expensive robots. Often using ```git``` and sometimes just ```scp```, the only difference with more expensive robots is they often have enough processing power to compile everything on the robot. You'll still need to write code. You'll still need to compile code. If you're doing "lots of copy paste libs stuff" then you're probably doing something wrong. Spending more money on a robot while you're still at the "newb" stage will likely just be a waste of *your* money. 

---

PS: @dlech, you're a saint for answering some of these issues.

theZiz, 2016-01-19:
Another thought: If the EV3 is not powerful enough for some useful AI magic, a remote control server (connected via wifi or bluetooth) may be a solution.
A simple control program, which wait for commands from outside and sends some sensor values back is much faster compiled on the ev3, too.
Some hybrid attempt with some prefiltering of sensor data before sending it to the remote AI could be useful, too.
Imho this is a very interesting topic. Unfortunately not easy, but that is not ev3's fault.

moriarty, 2016-01-19:
:+1: @theZiz good point

ADVANCESSSS, 2016-01-19:
But I was told I can add a 500 GB SD card to my brand new EV3.

It is the end AI I got it somehow. Yes it is simple.

DARwIn OP is a robot that is 12,000 that can upload my sketch to nad use the synax I believe to get ahold of the recordings and pass around the files like so, just like you can in EV3's blocks~ Without OpenCV/ROS.

dlech, 2016-01-19:
> But I was told I can add a 500 GB SD card to my brand new EV3.

The EV3 can only use SDHC type microSD cards, which means 32GB is the largest that will work. Any larger and you get data corruption.

> Lastly, does installing your dev to my "brand new EV3" ruin(change) it for good?

No, it doesn't "ruin" the EV3. It runs completely from a microSD card. If you remove the microSD card, your EV3 is exactly as it was before.

ADVANCESSSS, 2016-01-19:
Can I somehow add many 32Gb cards?...

dlech, 2016-01-19:
You could add additional storage via the USB port. But you need to know that the USB port is USB 1.1 only, so it will be rather slow no matter what type of storage you attach.

dlech, 2016-01-19:
It should be fast enough for logging data from the sensors, but if you need to constantly read and write, it could be a bottleneck.

rhempel, 2016-01-19:
Here's what I think we should do for this issue: @ADVANCESSSS - please read through the tutorials and get a simple Python script running that tuns a motor on when a touch sensor is pressed. That should take (I'm being very generous here) no more than an hour or two.

If you can't accomplish that, then you're going to be lost with your "simple" AI project.

ADVANCESSSS, 2016-01-19:
I don't have the EV3 yet...

Doesn't the EV3's sensors use small little bits like 0 to 100 for senses? It's not like 40MP, it's like 40 bits per sense, right? Though it saves a 1000 senses a sec....

Constantly all senses even ones in latest memory must be saved and check if match memory and constantly do actions.

rhempel, 2016-01-19:
Please get an EV3 and experiment with it to get a feel for what you can do - this is the wrong place to ask these questions.

closed 2016-01-19 by rhempel

theZiz, 2016-01-19:
I [once](https://github.com/ev3dev/ev3dev/issues/302#issuecomment-87833428) made some speed experiments. I don't know, whether this value is still valid, but I needed around ~40µs to read a sensor value. That would be 25000 sensor values per second (Ignoring, that the AI itself would need a bit CPU time, too ;) ). However there was an internal poll limit, in which the data will not change anyway. I forgot, whether it was around 100 or 1000 Hz.

ADVANCESSSS, 2016-01-19:
It was related to dev Idk why you closed this issue. I heard EV3's senses are actual numbers being 0 to 100 for all sensors. And I read it saves 1000 per sec.

ADVANCESSSS, 2016-01-19:
All I asked also was if I can set my AI to only save senses/etc. 5 times a sec or something.

dlech, 2016-01-19:
> I heard EV3's senses are actual numbers being 0 to 100 for all sensors. And I read it saves 1000 per sec.

This is for the official LEGO Education version firmware. ev3dev currently samples at 100Hz for most sensors but only stores the most recent value.

rhempel, 2016-01-19:
I closed the issue because you don't have an EV3 and if you want 25000 samples/sec but you're not sure if your application can tolerate 100 or 1000 then you don'd understand your application. And we already said you can simply save your samples in an array.

In another thread, I think you admitted that you want to use the EV3-G programming environment (the one with the blocks) [https://github.com/ev3dev/ev3dev/issues/513#issuecomment-172692348]

This group is not for answering EV3-G questions.

theZiz, 2016-01-19:
@rhempel: To be fair the 25000 samples/sec was from me and it was a totally theoritcal limit and I stated it this way + even mentioned the internal limit dlech later specified. ;)

@ADVANCESSSS: Did you actually read my answer? Yes, you can poll only 5 sensor values per second. The range of the values depends on the sensor.

ADVANCESSSS, 2016-01-19:
You know I don't understand what you said right? 100Hz? Rest meant? Lol.

Hey I din't say I want 25000 a sec....and if I did I asked if I could lower it, can I? If you know this I need to know this right now the lego community is very slow 1once a day reply.

ADVANCESSSS, 2016-01-19:
nonono for EV3 not dev can it pull only 5 senses per sec? I must know.

rhempel, 2016-01-19:
Please answer these two questions:

1. Do you actually have an EV3
2. Do you plan on using ev3dev or EV3-G (official LEGO programming environment)

ADVANCESSSS, 2016-01-19:
I know what your trying to do hey lol.............yes to all

ADVANCESSSS, 2016-01-19:
but only yous know the answer to those weird 2 questions we just mentioned, I must get the correct answer right away...

ADVANCESSSS, 2016-01-19:
Oh wait no I don't own one though.

rhempel, 2016-01-19:
Look, we are a very inclusive group, but now I think you are lying and that's just wasting our time.

You said here [https://github.com/ev3dev/ev3dev/issues/513#issuecomment-172692348] that you don't have an EV3 and here [https://github.com/ev3dev/ev3dev/issues/513#issuecomment-172692348] that you plan on using EV3-G.

Please do not comment further or open any new issues until you have an EV3 and have got a simple robot as described here [https://github.com/ev3dev/ev3dev/issues/513#issuecomment-172692348] working.

ADVANCESSSS, 2016-01-19:
No I don't have one I only meant I may soon get one if it's what I want and will use it's easy interface hopefully works, only said yes to all by accident.

There's only 1 question e brought up I'm confused about, for the EV3 without dev - can I lower the amount of times per sec it saves/etc. so it's like 20 a sec but still good enough?

rhempel, 2016-01-19:
We do not answer questions about the EV3-G software, please find a more appropriate group as described here [https://github.com/ev3dev/ev3dev/issues/513#issuecomment-172692600]

ADVANCESSSS, 2016-01-19:
Also why keep so many issues who the hell can go through them? You should only build a beaver dam that has the answers. Bad job. If you make people go through reading possibly pointless comments it's a mess buildup of text then. That is the wrong way to do things. You just "allow" multiples and then redirect somebody. Wow. Too big of a clump of text I say. I'd never go through all the issues. Never on any site (only did on important immortality forum [a few I checked not many]) And now I have a small compact The Everything of all important knowledge. Smaller things are linked from it, even how to learn c++ in 4 mins from my 4 days of learning&hierching how 40% of it works, not to mention EV3's bocks stabs that crypic code language in the back, fuc$@# pro-grammers are what smart yet the work grammerist, and you call my The Everything unreadble, you like only the way the=ings are and accept them, I broke all chains and am like ghost around your locked from taboo mind. Be cute, productive and nice, gov way, always friend cant ruin image, let's do things.