ev3dev/ev3dev #512: running a python program from the brickman file browser

pootle, 2016-01-18:
I see  here: http://www.ev3dev.org/news/2015/01/25/Package-Release/ that I can launch programs from the file browser, but I read elsewhere that accessing motors / sensors etc. with python requires root privilege, which the file browser does not allow.

What is the recommended way to run a python program from the file browser?

Or alternatively would it be possible to replace the openRoberta lab brick menu entry with my own program?

psychemedia, 2016-01-18:
If the python file is owned by user root and set to be executable, it should run from the file browser?

cavenel, 2016-01-18:
It would be interested to know where you read that accessing motors / sensors etc. with python requires root privilege. I never heard about that!
If you want to execute python files from the file browser, just think to add the shebang line as first line of your python code:

    
    #!/usr/bin/env python

and make the file executable (by chmod or filezila).

dlech, 2016-01-18:
This was also discussed recently in #466

pootle, 2016-01-30:
Just a note to wrap this up. and provide a clear trail for anyone else trying to do this - it is pretty straightforward:

1. stick **#!/usr/bin/env python** at the start of the python program
2. use the usual **if __name__ == "__main__"** statement to wrap the top level code
3. **chmod u+x myprog.py** to set execute permission on the file

Now you can browse to the file on the evo panel's file browser and lunch the program. Note that screen gets taken over for console output, but the text is _very small_, so not easy to read

I have a program running this way listening to the ir control and controlling motors in a 20ms loop in python quite happily.

closed 2016-01-30 by pootle

rhempel, 2016-01-30:
Pics or it didn't happen!

dlech, 2016-01-31:
tip: use backticks (``` ` ```) for code snippets.