ev3dev/ev3dev #513: Can the EV3 save sensory input in memory and then access any of the hundreds of them?

ADVANCESSSS, 2016-01-19:
Can the EV3 coontinually save sensory input into memory, and then get access to them and pick out one by a way like first letter or number or if is close to a number (like send the whole array saved up to this point to a check function~ Btw does the Array save them like a index of memory storage?

rhempel, 2016-01-19:
The EV3 language APIs allow you to read the sensor value by reading the `value*` attribute(s). It is up to your application to store the values in an array or whatever data structure you need.

dlech, 2016-01-19:
If we ever implement iio on top of our sensor drivers as mentioned in #498 then there could be a way to do this in the kernel driver.

ADVANCESSSS, 2016-01-19:
dlech er, what are you saying, that what I said won't work? Just wondering why you made a feeling or "it won't work unless" comment...

But it saves em all right? It doesn't just throw out each sense after it shows a binary does it?

dlech, 2016-01-19:
I'm saying that no one has written a driver that does this, so for now, you have to do what @rhempel says and read individual values and store them yourself.

ADVANCESSSS, 2016-01-19:
But I'm talking about using the programming-blocks >>> it gets sensory input and then saves them by " " by using a certain block, only blocks are used here no external mod hack ? ? ?

dlech, 2016-01-19:
What do you mean by programming blocks? The graphical programming language from LEGO?

ADVANCESSSS, 2016-01-19:
Yes, man...

ADVANCESSSS, 2016-01-19:
Maybe I didn't specify that above......

dlech, 2016-01-19:
ev3dev doesn't work with that. Better ask your question somewhere else like http://bricks.stackexchange.com/ or https://www.facebook.com/groups/legomindstorms/

closed 2016-01-19 by dlech

ADVANCESSSS, 2016-01-19:
Your comment doesn't make sense....

ADVANCESSSS, 2016-01-19:
dev....I'm not using dev though, just the blocks...

dlech, 2016-01-19:
This is an issue tracker for [ev3dev.org](http://www.ev3dev.org). We make an alternative operating system for the EV3. If you need help with the official software from LEGO, this is the wrong place to ask.

ADVANCESSSS, 2016-01-19:
Ok.