ev3dev/ev3dev #515: tacho-motor class - deprecate `speed_regulation` attribute

rhempel, 2016-01-21:
After working on the tacho motor driver proposal lately, it became clear that `run-direct` is the only command where we can practically use the `duty_cycle_sp` - this command is intended to allow the user to control ramping and other motor characteristics using the application code.

The `run-to-pos` commands are very difficult to implement with duty-cycle as the controlled parameter, and it is confusing to users that below 20% to 30% power the motor may not even turn when loaded.

Therefore, I am proposing that for all commands *except* `run-direct`, we use the `speed_sp` as the controlled parameter, which makes it easier for users to get the low speed control they expect when ramping to a position.

Same goes for timed ramping - the `speed_sp` makes sense as the controlled parameter, not `duty_cycle` because the speed regulaiton algorithm tries to maintain a set speed in the face of varying loads, battery voltages, and motor characteristics. The `duty_cycle` can't compensate for any of that.

I'm opening this issue for discussion so that we have a record of how we got here.

dlech, 2016-01-21:
I'm in favor of this. 

On PiStorms, this is the only way the motors will run. The options that are proposed to be removed don't work anyway. I'll have to check on BrickPi, but I think it pretty much only support run-direct (if there were three of me, I would modify the BrickPi firmware for better motor control).

The proposal also mirrors the behavior of the official EV3 software. Speed regulation is always used with the exception of the Unregulated Motor block in the advanced blocks. This is a direct parallel to all of our run modes using speed regulation except for `run-direct`. So, this behavior will be familiar to newcomers to ev3dev that are experienced with EV3.

ddemidov, 2016-01-21:
Does that mean there would be no alternative to the Unregulated Motor blocks in the official EV3 software? Would those have to be emulated on user level with `run-direct` command?

dlech, 2016-01-21:
Not exactly sure what you are asking, but Unregulated Motor block should be basically the same as run-direct.

ddemidov, 2016-01-21:
It was a long time since I touched the official software, but I thought the
unregulated blocks allowed at least run-timed command.

dlech, 2016-01-21:
Nope

[image]

ddemidov, 2016-01-21:
My bad :). Another question: are you planning to change the units of
speed_sp (or whatever that will be called) to percents relative to maximum
speed? I think that is the main difference to the official 'green'  blocks.

dlech, 2016-01-21:
Interesting idea. This could be useful. For example, the BrickPi uses 720 tacho counts per rotation for motors, so speed_sp on BrickPi is not the same as elsewhere for the same motor. But now that I'm thinking about it, you just make your library function...

    run_at_speed(percent):
        speed_sp = max_speed * percent / 100

I'm thinking we should keep the units in tacho counts since the position units are also tacho counts.

ddemidov, 2016-01-21:
Will max_speed be available as a sysfs attribute then?

dlech, 2016-01-21:
Yes, that is part of @rhempel's grand plan.

dlech, 2016-01-21:
max_speed in particular is being discussed on #356

jabrena, 2016-01-21:
Good Morning, 

yesterday I was reading the discussion a bit about the usage of parameters: 

* duty_cycle_sp
* speed_sp 

With my knowledge, the first parameter is used for Unregulated motors and the second one is used for motor with SpeedRegulation enabled.

@rhempel I didn't understand when you say that "run-direct" it the unique command to use with unregulated motors. In my examples, I use the command: run-forever without any problem.
https://github.com/jabrena/ev3dev-lang-java/blob/master/library/src/test/java/ev3dev/java/examples/motors/UnregulatedMotorDemo.java
https://github.com/jabrena/ev3dev-lang-java/blob/master/library/src/main/java/ev3dev/hardware/motor/BasicMotor.java

I followed the tuturial:
http://www.ev3dev.org/docs/tutorials/tacho-motors/

What is the problem?

Juan Antonio

rhempel, 2016-01-22:
Duty cycle is a very poor way to control a motor  -that's all that's available for simple DC motors. The old RCX was this way. The performance of the robots was very hard to predict when there were fluctuations in battery voltage, load, or motor characteristics. Some FLL teams ended up working out schemes for matching motors in pairs for better performance.

Using `speed_sp` eliminates that. For a motor that has a maximum unloaded speed of 900 counts per second at 9v, if you ask it to go at 720 counts per second (120 RPM) it will try very hard to do that no matter what the load, the battery voltage, or the motor characteristics.

The NXT motors (and now EV3) with their tacho feedback was one of the most significant improvements LEGO made to the MINDSTORMS system - let's take advantage of it.

jabrena, 2016-01-22:
So, the idea is to deprecate the usage of non-regulated features in a motor with encoders, isn't it?

If I developer uses a NXT/EV3 motor always should use regulared features. In my opinion, it is a correct approach.

But if I user connect a RCX Motor, the developer could use duty_cycle because in this case, it is the unique way to manage the motor.

In the java library side, this way cleans so much the code because it is not necessary to maintain 2 local variables.
https://github.com/jabrena/ev3dev-lang-java/blob/master/library/src/main/java/ev3dev/hardware/motor/BaseRegulatedMotor.java#L61-L62

What is the schedule to do this change in EV3Dev in order to update the library with time?

Cheers

Juan Antonio

rhempel, 2016-01-22:
The RCX Motor is connected as a DC motor - and you are correct, that one only has `duty_cycle` and ramping uses the duty cycle.

I hope to get more done this weekend - ramp to position may take a bit longer

dlech, 2016-01-22:
> But if I user connect a RCX Motor, the developer could use duty_cycle because in this case, it is the unique way to manage the motor.

You can put the output port into `dc-motor` mode instead of leaving it in `auto` mode exactly for the purpose of using RCX motors. In that case, it will use the `dc-motor` class (which uses duty cycle) instead of the `tacho-motor` class (which will not work anyway because the RCX motor does not have position feedback).

rhempel, 2016-01-22:
It's almost as if we thought of everything! This is the third rewrite of the tacho code - I plan to fix more than I break.

pootle, 2016-01-30:
Not sure I properly understand the potential collateral damage (!) of this, I would just like to note that I believe that there should still be a way to use duty cycle on the fly (i.e. using run_direct) while being able to read back position and speed attributes.

rhempel, 2016-01-30:
Yes, `run-direct` lets you control the duty cycle and read `speed` `position` attributes. Otherwise it would just be a `dc-motor` :-)
The drivers are done, I'm writing a test suite this weekend.

cavenel, 2016-01-30:
Having run-forever command without regulation was what made my print3rbot possible... :( Adding regulation is giving very unprecise drawing as both motors are regulating themselves independently.
For example, if one motor is a bit harder than the other, it will start slower and then accelerate to compensate, which will make a non straight line as the other motor will not do the same.
Without regulation, it was very easy to fix the speed of one motor proportionally to the speed of the other.
I hope I missed something here and it will still be possible to run forever without regulation...?

rhempel, 2016-01-30:
Yes, just use `run-direct` and vary the duty cycle yourself - and monitor `speed` and `position`

cavenel, 2016-01-30:
Great! Last time I checked, run-direct had strange behavior when changing duty-cycle while running. (the motor was stopped before restarting)
If it was a bug on my side then I will try again with new drivers!

rhempel, 2016-01-30:
We fixed that - the idea is that `run-direct` is for user-controlled motor operation

cavenel, 2016-01-30:
OK so run-direct is the old run-forever without the possibility of
regulation at all. Then I'm 100% happy! Thank you for the huge work here.

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel (and other `10-ev3dev` kernels on other platforms). Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech