ev3dev/ev3dev #516: [Documention] Getting Started / ssh example with default password r00tme?

jabrena, 2016-01-23:
Good morning,

Sometimes, I use the tutorial to get the default password to use into the terminal to get access in a brick.
Some weeks ago, I noticed that the example to use ssh has removed to the section about Getting started:

http://www.ev3dev.org/docs/getting-started/

Where is the classic example with the instruction to use ssh before section 5?

Juan Antonio

dlech, 2016-01-23:
It is at the end of each "Connecting to the Internet" tutorial (and possibly other places)

> Once you are connected, type in your username (`robot`) and your password (`maker` if you haven’t changed it yet) and then you should be logged in.

jabrena, 2016-01-23:
Hi @dlech you have reason, I used previous password: r00tme
http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

Oki

Cheers

closed 2016-01-23 by jabrena