ev3dev/ev3dev #517: Device class `ops` tables should be fully populated

rhempel, 2016-01-23:
In working on the `tacho-motor` drivers, I noticed inconsistency in some of the attribute getters/setters that are implemented using the `ops` tables that are implemented as pointers to functions in the lower level driver. In some cases, this results in code that looks like this:

```
static ssize_t max_speed_show(struct device *dev, struct device_attribute *attr,
			  char *buf)
{
	struct tacho_motor_device *tm = to_tacho_motor(dev);
	int speed;

	if (!tm->ops->get_max_speed)
		return -EOPNOTSUPP;

	speed = tm->ops->get_max_speed(tm->context);

	return sprintf(buf, "%d\n", speed);
}
```
This is clumsy because it forces a check to see if the pointer is non-NULL before calling - but on the other hand it allows the function to return the value directly. We infer that the operation is not supported because the function pointer is NULL.

I am proposing that we force all low level drivers to fully populate the `ops` tables for device classes, and that if the operaiton is not supported, the call returns -EOPNOTSUPP explicitly.

Of course, this means that we need to pass a pointer to the item we are getting, instead of returning the value directly, but I think it's a resaonable compromise. The point is that I'd like one definitive way of doing things instead of two.

dlech, 2016-01-23:
I consider this to be to separate issues.

Using the pattern

    int get_value(struct thing *context, unsigned *value);

    unsigned value;
    err = get_value(context, &value);
    if (err < 0)
        return err;

Instead of

    int get_value(struct thing *context);

    unsigned value;
    int ret;
    ret = get_value(context);
    if (ret < 0)
        return ret;
    value = ret;

Is a good thing and we should do that more often.

---

However, I disagree that the null check is clumsy. This is actually a common pattern in other kernel code. And it is actually how I find out if a function is optional or not, I look for the null check. If we don't do it that way, we have to be super careful to properly document which functions are optional and what return value they should return if they are "not supported" and keep that documentation up to date.

I think leaving the null checks makes for much less error prone and easier to maintain code. If we have boilerplate code everywhere and we decide we want to change the "not supported" return value, then we have to change it in 3 or more places and risk missing one.

dlech, 2016-01-23:
I actually spent a bit of time to implement the null checks the way it is now because it wasn't always like that and it was a mess.

rhempel, 2016-01-23:
My motivation here was to argue for doing it one way or the other consistently. I agree that the NULL check is no big deal for the actual getter/setter code - it's when we need to communicate with the driver outside the getter/setter (for example during ramping) that the NULL check seems like an extra step.

I'm OK to keep the NULL check as is then, as long as we're consistent. I think that as long as we're clear that the NULL check implies that the operation is optional then it's all good.

dlech, 2016-01-28:
Closing this as won't fix. We should be consistent in using null checks for optional functions.

closed 2016-01-28 by dlech