ev3dev/ev3dev #519: Netgear G54N150 Wireless USB Micro Adapter doesn't work with EV3 Brick

renders55, 2016-01-24:
enable wifi gives following error:
Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist

```
Linux ev3dev 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015 armv5tejl GNU/Linux
Bus 001 Device 002: ID 0846:9043 NetGear, Inc. 
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[    0.219811] usbcore: registered new interface driver usbfs
[    0.220427] usbcore: registered new interface driver hub
[    0.221106] usbcore: registered new device driver usb
[    0.924813] usb usb1: New USB device found, idVendor=1d6b, idProduct=0001
[    0.930540] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.936538] usb usb1: Product: DA8xx OHCI
[    0.939289] usb usb1: Manufacturer: Linux 3.16.7-ckt21-9-ev3dev-ev3 ohci_hcd
[    0.945218] usb usb1: SerialNumber: ohci.0
[    0.960172] usbcore: registered new interface driver usb-storage
[    0.989897] musb-hdrc: ConfigData=0x06 (UTMI-8, dyn FIFOs, SoftConn)
[    0.989956] musb-hdrc: MHDRC RTL version 1.800 
[    0.989982] musb-hdrc: setup fifo_mode 2
[    0.990020] musb-hdrc: 7/9 max ep, 2624/4096 memory
[    1.065484] usbcore: registered new interface driver usbhid
[    1.071226] usbhid: USB HID core driver
[    1.076935] usbcore: registered new interface driver r8712u
[    1.330972] usb 1-1: new full-speed USB device number 2 using ohci
[    1.563166] usb 1-1: New USB device found, idVendor=0846, idProduct=9043
[    1.568723] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.583425] usb 1-1: Product: WNA1000Mv2
[    1.586763] usb 1-1: Manufacturer: Realtek
[    1.590373] usb 1-1: SerialNumber: 00e04c000001
[   25.921788] usb0: HOST MAC 32:16:53:4a:40:51
[   25.922312] usb0: MAC 22:16:53:4a:40:51
[   25.989996] usb1: HOST MAC 12:16:53:4a:40:51
[   25.990724] usb1: MAC 02:16:53:4a:40:51
[   31.530435] device usb0 entered promiscuous mode
[   31.620232] device usb1 entered promiscuous mode
[   32.029207] tether: port 1(usb0) entered forwarding state
[   32.029556] tether: port 1(usb0) entered forwarding state
[  615.503065] tether: port 1(usb0) entered disabled state
[  617.927401] tether: port 1(usb0) entered forwarding state
[  617.927708] tether: port 1(usb0) entered forwarding state
[  756.062769] tether: port 1(usb0) entered disabled state
[  758.198301] tether: port 1(usb0) entered forwarding state
[  758.198621] tether: port 1(usb0) entered forwarding state
[ 1861.359891] tether: port 1(usb0) entered disabled state
[ 2454.300450] tether: port 1(usb0) entered forwarding state
[ 2454.300882] tether: port 1(usb0) entered forwarding state
[ 2862.054627] tether: port 1(usb0) entered disabled state
[ 2877.979884] tether: port 2(usb1) entered forwarding state
[ 2877.980145] tether: port 2(usb1) entered forwarding state
[ 2972.956766] tether: port 2(usb1) entered disabled state
[ 2973.405432] tether: port 1(usb0) entered forwarding state
[ 2973.405741] tether: port 1(usb0) entered forwarding state
[ 2977.270552] tether: port 1(usb0) entered disabled state
[ 2991.662069] tether: port 2(usb1) entered forwarding state
[ 2991.662342] tether: port 2(usb1) entered forwarding state
[ 5413.183303] tether: port 2(usb1) entered disabled state
[ 5413.628130] tether: port 1(usb0) entered forwarding state
[ 5413.628503] tether: port 1(usb0) entered forwarding state
[ 5432.199892] tether: port 1(usb0) entered disabled state
[ 5447.410097] tether: port 2(usb1) entered forwarding state
[ 5447.410354] tether: port 2(usb1) entered forwarding state
```
why isn't there a way to just add the driver
apt-get update && apt-get install firmware-iwlwifi

What's the point using Debian if you can't make changes?

dlech, 2016-01-24:
> why isn't there a way to just add the driver
> apt-get update && apt-get install firmware-iwlwifi
> 
> What's the point using Debian if you can't make changes?

There is no reason why this should not be able to install packages via apt unless your EV3 is not connected to the Internet. We have tutorials on how to connect your EV3 to the Internet via USB and Bluetooth.

Try connecting you your EV3 via USB or Bluetooth and follow our [new tutorial on setting up Wi-Fi via the command line](http://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/).

dlech, 2016-01-24:
> Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist

Does this error message appear on the EV3 itself (in the brickman user interface)?

dlech, 2016-01-24:
Error looks similar to #180. Perhaps this is a command line error. What is the command that you used?

renders55, 2016-01-25:
I did RTFM and searched through all the archives. Seems the wifi module
isn't loaded.

robot@ev3dev:~$ sudo connmanctl
[sudo] password for robot:
Error getting VPN connections: The name net.connman.vpn was not provided by
any .service fileconnmanctl> enable wifi
Error wifi: Method "SetProperty" with signature "sv" on interface
"net.connman.Technology" doesn't exist

connmanctl>

robot@ev3dev:~$ lsmod
Module                  Size  Used by
bridge                 81679  0
stp                     1295  1 bridge
llc                     3229  2 stp,bridge
bnep                    7705  2
usb_f_ecm               5563  2
usb_f_rndis            11952  2
u_ether                11005  2 usb_f_ecm,usb_f_rndis
hci_uart               12973  0
bluetooth             249589  6 bnep,hci_uart
rfkill                 10691  3 bluetooth
snd_legoev3             6520  0
suart_emu              23605  0
snd_pcm                65934  1 snd_legoev3
snd_timer              17534  1 snd_pcm
snd                    36187  3 snd_timer,snd_pcm,snd_legoev3
soundcore                676  1 snd
legoev3_ports          15320  0
ev3_motor               1811  1 legoev3_ports
tacho_motor_class       9870  1 ev3_motor
dc_motor_class          6319  1 tacho_motor_class
lego_bus                2808  2 legoev3_ports,ev3_motor
lego_port_class         2729  1 legoev3_ports
legoev3_battery         2190  0
legoev3_analog          7464  2 legoev3_battery,legoev3_ports
legoev3_bluetooth       2132  0
fuse                   76876  1
uinput                  6531  0
libcomposite           34697  16 usb_f_ecm,usb_f_rndis
configfs               22637  4 usb_f_ecm,usb_f_rndis,libcomposite
robot@ev3dev:~$

dlech, 2016-01-25:
It looks like this vendor and device id is not in the kernel driver. See <http://stackoverflow.com/questions/28705042/netgear-n150-wireless-usb-stick-not-working-with-raspbian-wheezy-on-my-pi> for a possible workaround.

renders55, 2016-01-30:
Finally I got it to work. Attached you find the entire story. A bit lengthy
but it might help folks to set this up.

Feel free to shorten as you prefer.

dlech, 2016-01-30:
Attachments don't work on GitHub issues - not via email anyway, so we didn't get the full message.

dlech, 2016-02-03:
Would you mind posting your attachment not as an attachment so we can know what you did?

renders55, 2016-02-03:
1. Install Debian on Virtualbox VM (Linux debian 3.16.0-4-amd64 #1 SMP
Debian 3.16.7-ckt20-1+deb8u3)
2. download driver package
rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911.tar.gz
3. cd to
~/RTL8188C_8192C_USB_linux_v4.0.2_9000.20130911/driver/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911$
4. edit Makefile and uncomment first line (#EXTRA_CFLAGS +=
$(USER_EXTRA_CFLAGS))
5. edit include/autoconf.h and uncomment:
  //#define CONFIG_DEBUG_RTL819X
  //#define CONFIG_PROC_DEBUG     1
  //#define DBG_CONFIG_ERROR_DETECT
  Credit for 4. and 5. goes to: JanRavn (ravn.co.uk)
6. Run make and observe driver building
7. cp 8192cu.ko to EV3 brick and copy the module into /lib/modules:
  root@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911# cp
8192cu.ko /lib/modules/3.16.7-ckt21-9-ev3dev-ev3/drivers/net/wireless/
root@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911# ls
-l /lib/modules/3.16.7-ckt21-9-ev3dev-ev3/drivers/net/wireless/
total 17848
-rw-r--r-- 1 root root   866936 Dec 15  2015 8188eu.ko
-rw-r--r-- 1 root root 17406032 Jan  1 00:25 8192cu.ko

t@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911#
modprobe -v 8192cu
install /sbin/modprobe --ignore-install 8192cu; echo "0846 9043" >
/sys/bus/usb/driver/rtl8192cu/new_id
### Credit for above line goes to: Mirko Sertic
insmod
/lib/modules/3.16.7-ckt21-9-ev3dev-ev3/kernel/drivers/net/wireless/rtl8192cu/8192cu.ko

sh: 1: cannot create /sys/bus/usb/driver/rtl8192cu/new_id: Directory
nonexistent
modprobe: ERROR: ../libkmod/libkmod-module.c:960 command_do() Error running
install command for 8192cu
modprobe: ERROR: could not insert '8192cu': Unknown symbol in module, or
unknown parameter (see dmesg)
root@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911# dmesg
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.
Total pages: 16256
[    0.000000] Kernel command line: mem=64M console=ttyS1,115200n8
root=/dev/mmcblk0p2 rw rootwait splash
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 59272K/65536K available (3930K kernel code, 217K
rwdata, 1172K rodata, 165K init, 135K bss, 6264K reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0503f14   (5104 kB)
      .init : 0xc0504000 - 0xc052d674   ( 166 kB)
      .data : 0xc052e000 - 0xc0564700   ( 218 kB)
       .bss : 0xc056470c - 0xc05865f0   ( 136 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every
178956969942ns
[    0.000449] Console: colour dummy device 80x30
[    0.000530] Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
[    0.070113] pid_max: default: 32768 minimum: 301
[    0.070626] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070687] Mountpoint-cache hash table entries: 1024 (order: 0, 4096
bytes)
[    0.073063] CPU: Testing write buffer coherency: ok
[... 655 lines cut ...]
Hidden SSID name? XXXXX
Passphrase? xxxxxxxxxxxxx
Error /net/connman/service/wifi_a42b8cf03639_hidden_managed_psk:
Input/output error
connmanctl> connect wifi_a42b8cf03639_hidden_managed_psk
Agent RequestInput wifi_a42b8cf03639_hidden_managed_psk
  Name = [ Type=string, Requirement=mandatory, Alternates=[ SSID ] ]
  SSID = [ Type=ssid, Requirement=alternate ]
  Passphrase = [ Type=psk, Requirement=mandatory ]
Hidden SSID name? XXXXXX
Passphrase? xxxxxxxxxxxxxxx
Error /net/connman/service/wifi_a42b8cf03639_hidden_managed_psk:
Input/output error
connmanctl> connect wifi_a42b8cf03639_hidden_managed_psk
Agent RequestInput wifi_a42b8cf03639_hidden_managed_psk
  Name = [ Type=string, Requirement=mandatory, Alternates=[ SSID ] ]
  SSID = [ Type=ssid, Requirement=alternate ]
  Passphrase = [ Type=psk, Requirement=mandatory ]
Hidden SSID name? XXXXXX
Passphrase? xxxxxxxxxxxxxx
Connected wifi_a42b8cf03639_hidden_managed_psk
connmanctl>

I had some issues connecting to my Airport Extreme at the 5GHz band. I
ended up connecting to the 2.4GHz and that worked.

renders55, 2016-02-03:
1. Install Debian on Virtualbox VM (Linux debian 3.16.0-4-amd64 #1 SMP
Debian 3.16.7-ckt20-1+deb8u3)
2. download driver package
rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911.tar.gz
3. cd to
~/RTL8188C_8192C_USB_linux_v4.0.2_9000.20130911/driver/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911$
4. edit Makefile and uncomment first line (#EXTRA_CFLAGS +=
$(USER_EXTRA_CFLAGS))
5. edit include/autoconf.h and uncomment:
  //#define CONFIG_DEBUG_RTL819X
  //#define CONFIG_PROC_DEBUG     1
  //#define DBG_CONFIG_ERROR_DETECT
  Credit for 4. and 5. goes to: JanRavn (ravn.co.uk)
6. Run make and observe driver building
7. cp 8192cu.ko to EV3 brick and copy the module into /lib/modules:
  root@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911# cp
8192cu.ko /lib/modules/3.16.7-ckt21-9-ev3dev-ev3/drivers/net/wireless/
root@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911# ls
-l /lib/modules/3.16.7-ckt21-9-ev3dev-ev3/drivers/net/wireless/
total 17848
-rw-r--r-- 1 root root   866936 Dec 15  2015 8188eu.ko
-rw-r--r-- 1 root root 17406032 Jan  1 00:25 8192cu.ko

t@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911#
modprobe -v 8192cu
install /sbin/modprobe --ignore-install 8192cu; echo "0846 9043" >
/sys/bus/usb/driver/rtl8192cu/new_id
### Credit for above line goes to: Mirko Sertic
insmod
/lib/modules/3.16.7-ckt21-9-ev3dev-ev3/kernel/drivers/net/wireless/rtl8192cu/8192cu.ko

sh: 1: cannot create /sys/bus/usb/driver/rtl8192cu/new_id: Directory
nonexistent
modprobe: ERROR: ../libkmod/libkmod-module.c:960 command_do() Error running
install command for 8192cu
modprobe: ERROR: could not insert '8192cu': Unknown symbol in module, or
unknown parameter (see dmesg)
root@ev3dev:/home/robot/rtl8188C_8192C_usb_linux_v4.0.2_9000.20130911# dmesg
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.
Total pages: 16256
[    0.000000] Kernel command line: mem=64M console=ttyS1,115200n8
root=/dev/mmcblk0p2 rw rootwait splash
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 59272K/65536K available (3930K kernel code, 217K
rwdata, 1172K rodata, 165K init, 135K bss, 6264K reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0503f14   (5104 kB)
      .init : 0xc0504000 - 0xc052d674   ( 166 kB)
      .data : 0xc052e000 - 0xc0564700   ( 218 kB)
       .bss : 0xc056470c - 0xc05865f0   ( 136 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every
178956969942ns
[    0.000449] Console: colour dummy device 80x30
[    0.000530] Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
[    0.070113] pid_max: default: 32768 minimum: 301
[    0.070626] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070687] Mountpoint-cache hash table entries: 1024 (order: 0, 4096
bytes)
[    0.073063] CPU: Testing write buffer coherency: ok
[... 654 lines cut ...]
  Passphrase = [ Type=psk, Requirement=mandatory ]
Hidden SSID name? XXXXX
Passphrase? xxxxxxxxxxxxx
Error /net/connman/service/wifi_a42b8cf03639_hidden_managed_psk:
Input/output error
connmanctl> connect wifi_a42b8cf03639_hidden_managed_psk
Agent RequestInput wifi_a42b8cf03639_hidden_managed_psk
  Name = [ Type=string, Requirement=mandatory, Alternates=[ SSID ] ]
  SSID = [ Type=ssid, Requirement=alternate ]
  Passphrase = [ Type=psk, Requirement=mandatory ]
Hidden SSID name? XXXXXX
Passphrase? xxxxxxxxxxxxxxx
Error /net/connman/service/wifi_a42b8cf03639_hidden_managed_psk:
Input/output error
connmanctl> connect wifi_a42b8cf03639_hidden_managed_psk
Agent RequestInput wifi_a42b8cf03639_hidden_managed_psk
  Name = [ Type=string, Requirement=mandatory, Alternates=[ SSID ] ]
  SSID = [ Type=ssid, Requirement=alternate ]
  Passphrase = [ Type=psk, Requirement=mandatory ]
Hidden SSID name? XXXXXX
Passphrase? xxxxxxxxxxxxxx
Connected wifi_a42b8cf03639_hidden_managed_psk
connmanctl>

The Netgear USB doesn't support 5GHz only 2.4.