ev3dev/ev3dev #520: zram swap size not big enough for dpkg-reconfigure locales

BertLindeman, 2016-01-24:
Using ev3 image `ev3dev-jessie-2015-12-30`

Running dpkg-reconfigure locales (during an upgrade) suffers:

```
Generating locales (this might take a while)...
  en_US.UTF-8...Killed
 done
Generation complete.
*** update-locale: Error: invalid locale settings:  LANG=en_US.UTF-8
```

I know en_US.UTF8 already is the defauls as installed :smile:

Looking at `/proc/swaps`  I see the zram swap is getting full.

As a circumvention I used a swapfile (just for the locales):
```
fallocate -l 512M /swapfile
chmod 600 /swapfile
mkswap /swapfile
swapon /swapfile
```
During the generate '/proc/swaps`:
```
Filename                                Type            Size    Used    Priority
/dev/zram0                              partition       65532   65532   16383
/swapfile                               file            524284  27932   -1
```

After the locales update, remove the swapfile:
```
swapoff /swapfile
rm /swapfile
```

In fact just reporting this for the circumvention.

dlech, 2016-01-24:
That is crazy that it take up so much RAM.

WasabiFan, 2016-01-24:
In my tests, this took upwards of a minute to load.

dlech, 2016-01-24:
so it ran for you without error?

WasabiFan, 2016-01-24:
Me? The menu loads after I wait 30-60 seconds, but I haven't tried actually changing the locale.

dlech, 2016-02-23:
I've increased the zram swap size so this might have a chance of working now.

BertLindeman, 2016-02-24:
Possibly related: I mounted some directories to tmpfs trying to protect my sd-card:
```
Filesystem      Size  Used Avail Use% Mounted on
tmpfs            31M     0   31M   0% /dev/shm
tmpfs            31M  9.0M   22M  30% /run
tmpfs           5.3M  4.1k  5.3M   1% /run/lock
tmpfs            31M     0   31M   0% /sys/fs/cgroup
tmpfs           105M     0  105M   0% /var/spool/cups
tmpfs           105M     0  105M   0% /var/cache/logwatch
tmpfs           105M     0  105M   0% /var/cache/apt/archives
tmpfs            21M  283k   21M   2% /var/log
```
This way I might loose some logging doing this.
I accept that.

Before update:
Current situation:
`'cat /proc/swaps` shows:
```
Filename                                Type            Size    Used    Priority
/dev/zram0                              partition       65532   20040   16383
```

update:
```
The following packages will be upgraded:
   zram-init (3.5ev3dev1 => 3.5ev3dev2)
```

`systemctl restart zram_swap.service`
`cat /proc/swaps`
```
Filename                                Type            Size    Used    Priority
/dev/zram0                              partition       98300   0       16383
```

Retry `time dpkg-reconfigure locales`
Seen:
```
Filename                                Type            Size    Used    Priority
/dev/zram0                              partition       98300   94124   16383
/dev/zram0                              partition       98300   94440   16383
/dev/zram0                              partition       98300   95096   16383
/dev/zram0                              partition       98300   98256   16383
```
And ready :smile: Pffffffff, close . . 

Terminal shows:
```
Generating locales (this might take a while)...
  en_US.UTF-8... done
Generation complete.

real    10m19.865s
user    1m30.050s
sys     2m27.020s
```
So: WORKS!
Thank you David

dlech, 2016-02-24:
Wow, what will you do with those extra 4 bytes?

If you are interested, you can setup another zram device for tmpfs. There is a service for this in the zram-init package. It is just turned off. <https://github.com/ev3dev/zram-init/blob/ev3dev-jessie/systemd/system/zram_tmp.service>. You will have to modify the kernel command line in `/etc/default/flash-kernel` to get more zram devices.

closed 2016-02-24 by dlech