ev3dev/ev3dev #522: stretch

dlech, 2016-01-28:
Doing some planning ahead here. It has recently been announced that [the package freeze for Debian Stretch will be in Q4 2016](https://nthykier.wordpress.com/2016/01/01/debian-please-plan-for-stretch/). I'm thinking that (if I'm not burned out by then) we will quit messing with ev3dev-jessie and call it ev3dev 1.0 at that point.

Any new features and breaking changes will be done in ev3dev-stretch and only bug fixes will be made in ev3dev-jessie. This way if anyone wants a long-term stable platform to build on, they will have it (finally).

dlech, 2016-09-10:
https://lists.debian.org/debian-devel-announce/2016/07/msg00002.html

dlech, 2016-11-17:
So, what do people think? Does anybody want ev3dev-stretch? (I have no doubt that we will get requests *after* stretch becomes stable.)

dwalton76, 2016-11-17:
I am not that familiar with stretch so was looking through the release notes:
https://www.debian.org/releases/stretch/mipsel/release-notes.en.txt

I didn't spot anything new that I am dieing to have but I only skimmed and maybe I missed something.

dlech, 2016-11-17:
Basically, all of the software packages will be 2 years newer. The release notes you linked look like they haven't updated version numbers (it shows wheezy>jessie version numbers despite having changed the titles to jessie>stretch). So, python 3.5 instead of 3.4, newer BlueZ for better bluetooth support and so on.

Also, something else to take into consideration... Debian is dropping armel support after stretch, so there will be no possibility of ev3dev-buster based on official Debian. So, ev3dev-stretch is really the last chance to have the latest and greatest debian software on the EV3.

dlech, 2016-11-17:
Stretch also means a 4.9 kernel and whatever bug fixes and new hardware support comes with that.

dwalton76, 2016-11-17:
If this is the last possible major debian upgrade that can be done on ev3dev agreed 100% you will get requests for it.

JorgePe, 2016-11-22:
Well since you are asking... I do want stretch :)
I've been messing with Bluetooth LE and having python 3.5 and newer kernel and BlueZ would be great.

Anty0, 2017-02-09:
I want stretch too :)
I have lot's of problems with outdated software on jessie. Lot's of libraries for Node.js requires newer version of node.
Also Python 3.5 and Java 8 and newer kernel...
So update to stretch would be great.

JorgePe, 2017-02-09:
Almost forgot this topic. Last week a found another (rather silly) reason for stretch: LIRC on jessie and current Ubunty is to old (0.90 or so), had to compile from source (0.94d) to get support for ftdix devices.

I found myself back to LIRC because I can make my own USB PF-IR emiiter with just one FTDI adapter and one IR LED. No big deal unless you need several emitters... and since LIRC can address several emitters by their serial numbers we can potentially have 127 independent emitters (probably impossible with the EV3 but perhaps quite possible with a Raspberry Pi 3).

dlech, 2017-02-09:
> Lot's of libraries for Node.js requires newer version of node.

This won't happen on EV3 (unless you know of a compatible javascript engine) - see #448

dlech, 2017-02-09:
Also, I've started a project to track stretch: https://github.com/ev3dev/ev3dev/projects/7

If there is anything that needs special attention that you don't see listed, open a new issue and I will add it.

dlech, 2017-03-25:
The very first stretch images are available for testing: https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-03-25/

closed 2017-04-30 by dlech