ev3dev/ev3dev #526: ros

f-e-l-i-x, 2016-01-29:
Has anyone managed to get ROS running on ev3dev according to the following instructions:

https://github.com/srmanikandasriram/ev3-ros

It seems to be working according to the Indian research project report:
http://www.ee.iitm.ac.in/~ee11b127/files/EV3/EV3_ROS_Manikandasriram.pdf

However, I cannot get it to work and the instructions on github are so vague that I do not know where I took a wrong turn. Anyone interested in getting this to work? Could be awesome!

moriarty, 2016-01-29:
I have instructions to install ros_comm onto the ev3-dev:

https://github.com/moriarty/ros-ev3

moriarty, 2016-01-29:
Sorry I didn't elaborate, I wanted to browse the repository by @srmanikandasriram

I don't have any time to test out the method you mentioned at the moment. It's using ros_serial and it does provide code to get you up an running if it works, but you said you were unable to follow the instructions. There is also another method which uses yocto-linux (by @cyborg-x1) it is listed here: http://wiki.ros.org/Robots/EV3

The instructions I provide install the [ros_comm](http://wiki.ros.org/ros_comm) package in [brickstrap](https://github.com/ev3dev/brickstrap) and then creating a new ev3dev image to put onto the microSD card which will include the ros communication packages. Installing them directly on the EV3 itself doesn't work. 

My instructions do not include any code to start using it right away- never had the time. Once you've followed the instructions, and copied the new image to microSD card you will have the default ev3dev environment, with the addition of ros_comm. You just need to write a small program which reads the sensor values and publishes the data onto ros topics, and subscribes to one or more ros topics to receive and set motor commands. 

If you're new to ev3dev and to ros, I recommend my instructions because they leave you with a default environment and you can then just follow the standard publisher/subscriber tutorials in the ros wiki: http://wiki.ros.org/ROS/Tutorials

I've tested with Python and C++, if you use C++ you try building on the EV3 if you are patient but you can also build in the brickstrap environment and then copy the required directories over - for this reason I recommend python to start. 

All methods: ros_comm, ros_serial and yocto-linux require a ros_core running on the network. 

I tagged the authors of the other methods in this issue to see if they have any advice. 

Also, the file in which my instructions are is poorly named, but linked to from the main readme. Here is a direct link: https://github.com/moriarty/ros-ev3/blob/master/brickstrap-build-status.md

cyborg-x1, 2016-01-30:
Well as he, I never tried the instructions I use Yocto, I have a working image for an SD-Card, but I wanted to finish my stuff before I share the whole Image. I created ros nodes which use ROS Control to load controllers for Sensors and Motors by launchfile from your PC. If you are not familiar with ROS Control yet, it works like this: You have a node I called mine EV3 Manager. It creates Interfaces for Joints and Sensors. If you want to use them you have to load controllers from libraries at runtime. This is done with launchfiles and the parameter server. You load the parameters to the parameter server (master) and then you tell the Node by service to load the controller.
Example for Sensors: 
https://github.com/Hacks4ROS/h4r_ev3_ctrl/blob/develop/h4r_ev3_launch/launch/infrared.launch

The parameters uploaded to the server:
https://github.com/Hacks4ROS/h4r_ev3_ctrl/blob/develop/h4r_ev3_launch/config/infrared.yaml
(Loads a controller for the IR Sensor in remote mode and publishes each remote channel as joy topic)

 For the joints standard ROS Controllers  work - like for example the diffdrive controller - I have a video of it linked in my README, with me driving by joystick https://github.com/Hacks4ROS/h4r_ev3_ctrl. 

---

This controller uses two motor interfaces with information about diameter and distance of the wheels to control it by a geometry_msgs::Twist topic (cmd_vel)

Well I also will also want to share my build environment later - so you can build your own controllers - actually I do not really know how I can do that efficiently - without uploading tons of stuff - multiple gigabytes (compiled packages) ).  You need that to build the controllers on your PC or Server so you do not have to use the poor performing EV3 Brick to do so.  

But Yocto compiles all the packages for the whole system - that's why it is so huge on your PCs harddisk when it is done. (>50 GB)

I maybe it can be done by just sharing the image with dev packages installed, but I have not done that so far - there was an issue with installing them natively as far as I remember.

I plan to publish a image on a Github repo in the recent future ... probably next month.

Later I plan to write a daemon (Linux) which starts this node when it finds the ros master and disables it after it disappears - already tried that concept - seems to work. And in the end you just do all your fancy stuff on the PC and just load the controllers with the launchfile started on your PC - using the brick more like a remote. Well and on the EV3 there is just setting the wireless and ROS_MASTER_URI ...

I do not know how much of your stuff you could move to the EV3 directly - but I would recommend to do the "other" stuff on the PC - if you do not exchange the Brick by a raspberrypi 1 or 2. Python is as far as I have seen a little to heavy - ROS Master works but takes ages (minutes) to start and crashes soon. Well the system runs out of memory (60MB). I also start my node directly without roslaunch or rosrun (cause it also uses python). I save the memory for loading the controllers (btw. also involving python).

So in conclusion my "method" is still "work in progress", but already works fine - except for a ugly bug freezing the system sometimes (mostly when plugging the Color sensor).

jabrena, 2016-02-01:
HI @cyborg-x1,

what ROS release are you using for the project? Jade?

cyborg-x1, 2016-02-02:
The version used in the Yocto meta-ros layer is indigo. Should be still compatible with jade afaik.

https://github.com/bmwcarit/meta-ros/releases

cyborg-x1, 2016-02-22:
Everybody interested, see here for getting a first alpha version of my Linux for EV3 http://hacks4ros.github.io/h4r_ev3_ctrl/

jabrena, 2016-02-22:
Nice work. In some weeks, I will follow your guide. 

Did you use ros_bridge?
http://wiki.ros.org/rosbridge_suite

Juan Antonio

cyborg-x1, 2016-02-22:
No not really, I used Yocto - this is a custom made Linux image - just the Kernel with Drivers is from EV3.
Most of the known ROS libraries are there, ... there is also the PCL installed but  well, I guess, it is a bit useless on that tiny system - also roscore - but forget about it, too less RAM. I also start my so called ev3_manager node directly, without rosrun to reduce the memory consumption by python helpers.

moriarty, 2016-04-11:
I recently had a some questions asked in the issues for my instructions (https://github.com/moriarty/ros-ev3)

I needed to update the instructions to include the ```ev3dev-lang-cpp``` package, and the ros ```common_msgs``` stack. I added a patchfile for the ev3dev-lang-cpp so that the library is installed where it can be found by catkin. I hadn't been testing the c++ as I usually use python so I don't know how long that was broken.

I took a little bit of time this afternoon and made a very short demo package with a c++ ros node.
- It just runs one listens to a topic for "start" and "stop" [std_msgs/String] and runs the motor.
- It also publishes it's internal state to another topic. 

But the main thing that this repository is demonstrating is the general instructions & CMakeList.txt for someone to use to create their own project.

https://github.com/moriarty/ev3dev_ros_demo

bidaian, 2017-02-28:
I'm running ROS on ev3dev. After trying brick and docker I simply added stretch to the sources and installed the package ros-robot. No compiling. It's going to be even easier once ev3-dev releases the stretch images. It's mostly python so it doesn't pull many dependencies. The only tricky part is getting the environment variables right. So far, I can list nodes, and send messages on topics (like moving the turtle from the ev3). I didn't test many things but I think everything should be working.

I'm using a bluetooth connection to ev3dev and run roscore in the host.

moriarty, 2017-02-28:
I have ROS running on my EV3, I added this project page here Years ago:

http://www.ev3dev.org/projects/2014/11/17/ROS-on-EV3/

Occasionally I receive pull requests or questions to update the documentation, I haven't changed anything major in a while... it's been stable for ages. I haven't updated it for Kinetic or Lunar. But usually the changes required are obvious.

There is another project which is linked to from this page: http://wiki.ros.org/Robots/EV3

A few months ago one of the ROS contributors opened a pull request with a suggestion, so I should really update the docs...

https://github.com/moriarty/ros-ev3/issues/14

jabrena, 2017-02-28:
Last week, installed kinetic on Raspi for Brickpi+ &   PiStorms. I hope creating a rosjava node this week.

https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/192
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/194

closed 2017-07-25 by dlech