ev3dev/ev3dev #527: Question: Does changing EV3 brick have any downsides?

laurensvalk, 2016-01-31:
I have one SD card with ev3dev, and two EV3 bricks. So sometimes, I just put the ev3dev SD card in the other brick.

Does this have any implications on the preconfigured Bluetooth/USB configurations? For example, does my laptop see each brick as a different one due to different hardware addresses, or do all bricks look the same?

(I seem to be having some connection issues after making the change, and I wonder if this might be the cause.)

rhempel, 2016-01-31:
Your Bluetooth may be affected - as I understand it the BT MAC address is used during binding. I'm not sure if the processor itself uses a different hardware address for the USB connection - that part should be fine.

dlech, 2016-01-31:
The Bluetooth and USB MAC addresses will be different. So, you if you are using Bluetooth, The 2 EV3s will be seen as 2 different devices. Also, if you are using Internet Connection Sharing (on any OS), the two EV3s will show up as separate network connections. You will probably have a different IP address assigned as well if you are using DHCP.

It is not surprising that you might have trouble connecting since the hostname stays the MAC address or IP address has changed.

If you really want to, it is possible to override this behavior and assign fixed MAC addresses.

laurensvalk, 2016-01-31:
Thanks for the information. That should explain the issues I ran into.

I'll stick to using just one EV3 brick.

closed 2016-01-31 by laurensvalk