ev3dev/ev3dev #528: xkcd Style Graphs for Motor Ops

rhempel, 2016-01-31:
Some time ago, I wanted to create xkcd style graphs - and thanks to matplotlib and Python it is possible. I have a little helper routine that runs a motor command and logs `time`, `speed`, `position` and `duty_cycle` in a tuple at specified intervals, then I write the settings and raw data to a file when done.

That file is then imported into the plot generator, and the ouptut looks like this:

[image]

I'm going to create diagrams like this that show how the motor commands work, and the charts will also be useful for tuning motor control parameters and verifying operation.

dlech, 2016-02-01:
I'm not a fan of the hand drawn look - not for posting on our website anyway. It makes it hard to tell what is styling and what is actual variations in the data. I assume there are more "professional" styles to choose from.

rhempel, 2016-02-01:
Yes, there are more scientific looks - I just have to get rid of the xkcd() statement :-) I'll provide a few other styled samples - I think there's a nice default dark style that looks like a scope.

That being said, I found a defect in the driver where the speed is non-zero while holding - still need to figure that one out...

rhempel, 2016-02-01:
How about this:

[image]

rhempel, 2016-02-01:
Looks like there's a ded time of almost exactly 100 msec at the start of the ramp - the first ramp callback leaves the speed setpoint at 0, speed is not changed until the second callback. 

These graphs are pretty useful :-)

ddemidov, 2016-02-01:
@rhempel, if you are planning to put the plots on the site, it could be a better idea to use a javascript based plotting library like [Bokeh](http://bokeh.pydata.org/). It is easy to use from python ([example](http://nbviewer.jupyter.org/gist/ddemidov/c508d44b94e49970daa4)), it produces interactive plots (with zooming, panning and other tools), and it is easy to [dump to html](http://bokeh.pydata.org/en/latest/docs/user_guide/embed.html).

rhempel, 2016-02-01:
At first I was against yet another tool to learn, but I have to say that I like that Bokeh example. Is it possible to put arbitrary notatoins to points on the plot, ie "Here is where the `time_sp` expires and ramp down begins?

ddemidov, 2016-02-01:
Something like this: http://nbviewer.jupyter.org/gist/ddemidov/6a41b15ca59b134ed4b9?

rhempel, 2016-02-02:
Yes, that looks almost exactly like the right tool - so next question is where to put the motor test scripts - probably in the Python binding under tests - but can we specify which files are intended to be used by Travis and which ones are for running on the actual target?

ddemidov, 2016-02-02:
Travis exports `TRAVIS=true` environment variable, so we could use that to decide what tests to run. But what exactly the motor tests should do? As I understand, you just dump some data to produce some plots. How would you automate analysis of the data? We could just place the logging and plotting scripts into `plots` dir.

dlech, 2016-10-11:
OK to close this issue?

closed 2017-02-08 by dlech