ev3dev/ev3dev #529: Can't see lego-sensors or motors on BrickPi

jfcloutier, 2016-02-01:
I can't seem to see my sensors or motors on the BrickPi under ev3dev.

- lego-sensor class is missing from /sys/class even when Ev3 sensors are plugged into S1, S2 etc.

- tacho-motor class contain motor0...motor3, each with driver_name = lego-nxt-motor even though no motor is connected

My configuration is:

Raspberry Pi2
BrickPi rev.2.8
running rpi2-ev3dev-jessie-2015-12-30

config.txt:

####################################
# Dexter Industries BrickPi config #
####################################

# uncomment both if you are using BrickPi
dtoverlay=brickpi
init_uart_clock=32000000

# uncomment only if you are using BrickPi+
dtparam=brickpi_battery=okay

# uncomment to disable BrickPi LEDs on boot.
#dtparam=brickpi_led1_trigger=none,brickpi_led2_trigger=none

# uncomment if you have a Model B Rev 1 board that uses gpio 21 instead of
# gpio 27
#dtparam=brickpi_led1_gpio=21

# uncomment if you have a mindsensors.com RTC sensor connected to input port 5
#dtoverlay=i2c-rtc,ds1307

 ---> (everything else is unchanged)

From dmesg:

robot@ev3dev:/sys/class$ dmesg | grep lego
[    4.997226] lego-port port0: Registered 'in1' on 'ttyAMA0'.
[    4.999941] lego-port port1: Registered 'in2' on 'ttyAMA0'.
[    5.002762] lego-port port2: Registered 'ttyAMA0:outA' on 'ttyAMA0'.
[    5.005566] lego-port port2: Added new device 'ttyAMA0:outA:lego-nxt-motor'
[    5.011504] lego-port port3: Registered 'ttyAMA0:outB' on 'ttyAMA0'.
[    5.014313] lego-port port3: Added new device 'ttyAMA0:outB:lego-nxt-motor'
[    5.019719] lego-port port4: Registered 'in3' on 'ttyAMA0'.
[    5.022521] lego-port port5: Registered 'in4' on 'ttyAMA0'.
[    5.025338] lego-port port6: Registered 'ttyAMA0:outC' on 'ttyAMA0'.
[    5.028204] lego-port port6: Added new device 'ttyAMA0:outC:lego-nxt-motor'
[    5.030766] lego-port port7: Registered 'ttyAMA0:outD' on 'ttyAMA0'.
[    5.033280] lego-port port7: Added new device 'ttyAMA0:outD:lego-nxt-motor'
[    5.119559] tacho-motor motor0: Registered 'lego-nxt-motor' on 'ttyAMA0:outA'.
[    5.122483] tacho-motor motor1: Registered 'lego-nxt-motor' on 'ttyAMA0:outB'.
[    5.125247] tacho-motor motor2: Registered 'lego-nxt-motor' on 'ttyAMA0:outC'.
[    5.128830] tacho-motor motor3: Registered 'lego-nxt-motor' on 'ttyAMA0:outD'.

Your help is greatly appreciated.

dlech, 2016-02-01:
All is as it should be. BrickPi cannot automatically detect sensors or motors. Since the BrickPi was designed for use with NXT motors, we setup the those by default. However, there are too many kinds of sensors to guess which ones you will have, so none are loaded.

Use the `mode` and `set_device` attribute of `lego-port` to get sensors working. See <http://www.ev3dev.org/docs/ports/brickpi-in-port/>.

jfcloutier, 2016-02-01:
Got it. It works. Thanks for the near instantaneous response!

closed 2016-02-01 by dlech

jabrena, 2016-03-23:
Hi @dlech, 

I think that the configuration is Ok:

``` # For more options and information see 
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

###################################
# mindsensors.com PiStorms config #
###################################

# uncomment if you are using PiStorms
#dtoverlay=pistorms

# uncomment to invert the display
#dtparam=rotate=270

####################################
# Dexter Industries BrickPi config #
####################################

# uncomment both if you are using BrickPi
dtoverlay=brickpi
init_uart_clock=32000000

# uncomment only if you are using BrickPi+
dtparam=brickpi_battery=okay

# uncomment to disable BrickPi LEDs on boot.
#dtparam=brickpi_led1_trigger=none,brickpi_led2_trigger=none

# uncomment if you have a Model B Rev 1 board that uses gpio 21 instead of
# gpio 27
#dtparam=brickpi_led1_gpio=21

# uncomment if you have a mindsensors.com RTC sensor connected to input port 5
#dtoverlay=i2c-rtc,ds1307

################################
# Standard Raspberri Pi config #
################################

# uncomment if you want to use the camera module
#start_x=1
#gpu_mem=128
#disable_camera_led=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2
#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800
```

dlech, 2016-03-23:
What is `lsmod`? Are there any relevant error messages in `sudo journalctl -b`? Sensors and motors are not automatically detected - see comments above.

dlech, 2016-03-23:
Also, there are a few BrickPi bug fixes that have not been officially release yet. You can find an RPi2 kernel package [here](https://github.com/ev3dev/ev3dev/issues/374#issuecomment-181975748) that has some fixes that are not yet available via `apt-get`.

jabrena, 2016-03-23:
Hi mate, 

**lsmod output:**

``` bash
robot@ev3dev:/sys/class/lego-port$ lsmod
Module                  Size  Used by
8021q                  19200  0 
garp                    5471  1 8021q
stp                     1479  1 garp
llc                     3600  2 stp,garp
arc4                    1753  2 
brickpi                13386  1 
lego_bus                3376  1 brickpi
lego_port_class         3333  1 brickpi
ath9k_htc              55734  0 
ath9k_common           19520  1 ath9k_htc
ath9k_hw              423968  2 ath9k_common,ath9k_htc
ath                    18816  3 ath9k_common,ath9k_htc,ath9k_hw
mac80211              518846  1 ath9k_htc
cfg80211              399468  4 ath,ath9k_common,mac80211,ath9k_htc
rfkill                 16799  3 cfg80211
brickpi_battery         1918  0 
bcm2835_gpiomem         3023  0 
i2c_bcm2708             5006  0 
uio_pdrv_genirq         2986  0 
uio                     8228  1 uio_pdrv_genirq
fuse                   82598  1 
ipv6                  343236  34 
``` 

A possible clue with: sudo journalctl -b

``` bash
Jan 01 00:00:14 ev3dev kernel: tty ttyAMA0: Failed to init while setting sensors. (-110)
```

jabrena, 2016-03-23:
Oki, burning this image now:

https://github.com/ev3dev/ev3dev/issues/531#issuecomment-180933735
https://github.com/ev3dev/ev3dev/files/120554/linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.zip

Many thanks mate, I go with the test :D

dlech, 2016-03-23:
`-110` means that it timed out, so something is preventing serial communications from working correctly. Would you happen to be using RPi3 instead of RPi2? Also, make sure you completely remove power from everything and try again, not just reboot.

jabrena, 2016-03-23:
Hi @dlech, I am using RPI2 with BrickPi. 
Give me some minutes. I will burn another sdcard.

jabrena, 2016-03-23:
``` bash
robot@ev3dev:~$ uname -r
4.1.15-10-rc2-ev3dev-rpi2
```

``` lsmod
robot@ev3dev:/sys/class/lego-sensor$ lsmod
Module                  Size  Used by
8021q                  19600  0 
garp                    5535  1 8021q
stp                     1479  1 garp
llc                     3652  2 stp,garp
arc4                    1753  2 
brickpi                13390  1 
ev3_uart_sensor         6024  1 brickpi
nxt_i2c_sensor         30531  1 ev3_uart_sensor
lego_bus                3376  3 ev3_uart_sensor,nxt_i2c_sensor,brickpi
lego_port_class         3393  2 nxt_i2c_sensor,brickpi
servo_motor_class       6380  1 nxt_i2c_sensor
lego_sensor_class       7671  2 ev3_uart_sensor,nxt_i2c_sensor
ath9k_htc              55566  0 
ath9k_common           19624  1 ath9k_htc
ath9k_hw              424332  2 ath9k_common,ath9k_htc
ath                    18920  3 ath9k_common,ath9k_htc,ath9k_hw
mac80211              549504  1 ath9k_htc
cfg80211              426623  4 ath,ath9k_common,mac80211,ath9k_htc
rfkill                 17335  3 cfg80211
brickpi_battery         1918  0 
i2c_bcm2708             5066  0 
bcm2835_gpiomem         3023  0 
uio_pdrv_genirq         2966  0 
uio                     8228  1 uio_pdrv_genirq
fuse                   83229  1 
ipv6                  344580  34 
```

Now, in /sys/class I see more pahts, but I could see anything in /lego/port

``` bash
robot@ev3dev:/sys/class$ ls
backlight        gpio              iscsi_endpoint   mem           rtc          thermal  vtconsole
bcm2708_vcio     graphics          iscsi_host       misc          scsi_device  tty      watchdog
bcm2835-gpiomem  hidraw            iscsi_iface      mmc_host      scsi_disk    uio
bdi              i2c-adapter       iscsi_session    net           scsi_host    vc
block            i2c-dev           iscsi_transport  power_supply  servo-motor  vc-cma
bsg              ieee80211         leds             pwm           sound        vc-mem
devcoredump      input             lego-port        raw           spi_master   vc-sm
dma              iscsi_connection  lego-sensor      rfkill        spidev       vchiq
robot@ev3dev:/sys/class$ ls ./lego-port/
robot@ev3dev:/sys/class$ ls ./lego-sensor/
```

I have attached to brickpi  a EV3 middle motor, EV3 large motor & NXT large motor, but I couldn't detect them.

in journalctl -b

``` bash
robot@ev3dev:/sys/class$ journalctl -b
-- Logs begin at Sun 2016-03-13 18:15:44 UTC, end at Wed 2016-03-23 17:20:18 UTC. --
Mar 23 17:20:18 ev3dev systemd[378]: Starting Paths.
Mar 23 17:20:18 ev3dev systemd[378]: Reached target Paths.
Mar 23 17:20:18 ev3dev systemd[378]: Starting Timers.
Mar 23 17:20:18 ev3dev systemd[378]: Reached target Timers.
Mar 23 17:20:18 ev3dev systemd[378]: Starting Sockets.
Mar 23 17:20:18 ev3dev systemd[378]: Reached target Sockets.
Mar 23 17:20:18 ev3dev systemd[378]: Starting Basic System.
Mar 23 17:20:18 ev3dev systemd[378]: Reached target Basic System.
Mar 23 17:20:18 ev3dev systemd[378]: Starting Default.
Mar 23 17:20:18 ev3dev systemd[378]: Reached target Default.
Mar 23 17:20:18 ev3dev systemd[378]: Startup finished in 57ms.
```

dlech, 2016-03-23:
Is the same error ` tty ttyAMA0: Failed to init while setting sensors. (-110)` still present in journalctl (or `dmesg`)? If there is nothing in `/sys/class/lego-port/`, then there is still something wrong during initialization of the driver.

jabrena, 2016-03-23:
No, the full log is the showed un the previos comment.
If you like, i will repeat the test from zero with a raspberry pi 1 and 2

dlech, 2016-03-23:
There is no way that is the full log. :wink: `journalctl` by default uses `less` so you can scroll up and down. Have a look at the output of `dmesg` instead.

jabrena, 2016-03-24:
**Test iteration 1 with RPI1 and Brickpi V2.8:**

I burned a sdcard with rpi-ev3dev-jessie-2015-12-30.img but I didn't update the system with any upgrade.

``` bash
robot@ev3dev:~$ uname -r
4.1.15-9-ev3dev-rpi
```

``` bash
robot@ev3dev:~$ lsmod
Module                  Size  Used by
8021q                  20796  0 
garp                    7024  1 8021q
stp                     2106  1 garp
llc                     5670  2 stp,garp
ev3_motor               2752  0 
tacho_motor_class      14245  1 ev3_motor
dc_motor_class          9159  1 tacho_motor_class
brickpi                15322  1 
lego_bus                4796  2 brickpi,ev3_motor
lego_port_class         4527  1 brickpi
brickpi_battery         2491  0 
8192cu                567860  0 
bcm2835_gpiomem         3703  0 
i2c_bcm2708             6004  0 
uio_pdrv_genirq         3550  0 
uio                    10002  1 uio_pdrv_genirq
fuse                   85816  1 
ipv6                  354313  28 
``` 

Using dmesg, I could see that raspi register ports from brickpi:

**dmesg:**

``` bash
[    9.382729] Registered BrickPi line discipline. (28)
[    9.395317] lego-port port0: Registered 'in1' on 'ttyAMA0'.
[    9.402480] lego-port port1: Registered 'in2' on 'ttyAMA0'.
[    9.417407] lego-port port2: Registered 'ttyAMA0:outA' on 'ttyAMA0'.
[    9.434907] lego-port port2: Added new device 'ttyAMA0:outA:lego-nxt-motor'
[    9.451017] lego-port port3: Registered 'ttyAMA0:outB' on 'ttyAMA0'.
[    9.453168] lego-port port3: Added new device 'ttyAMA0:outB:lego-nxt-motor'
[    9.474798] lego-port port4: Registered 'in3' on 'ttyAMA0'.
[    9.498394] lego-port port5: Registered 'in4' on 'ttyAMA0'.
[    9.501363] lego-port port6: Registered 'ttyAMA0:outC' on 'ttyAMA0'.
[    9.512434] lego-port port6: Added new device 'ttyAMA0:outC:lego-nxt-motor'
[    9.519286] lego-port port7: Registered 'ttyAMA0:outD' on 'ttyAMA0'.
[    9.521419] lego-port port7: Added new device 'ttyAMA0:outD:lego-nxt-motor'
[   11.395588] tacho-motor motor0: Registered 'lego-nxt-motor' on 'ttyAMA0:outA'.
[   11.436786] tacho-motor motor1: Registered 'lego-nxt-motor' on 'ttyAMA0:outB'.
[   11.459189] tacho-motor motor2: Registered 'lego-nxt-motor' on 'ttyAMA0:outC'.
[   11.481713] tacho-motor motor3: Registered 'lego-nxt-motor' on 'ttyAMA0:outD'.
```

My questions for raspberry pi:

* Using raspberry pi2, I see more modules loaded in the system (ev3_uart_sensor,nxt_i2c_sensor). Is it necessary to update this version?
* At the moment, I have connected: 1x ev3 medium motor, 1x ev3 large motor, 1x nxt motor but reading dmesg, all motors are considered as nxt motor.
* Why the system didn't detect any lego sensor?

I continue the same test with raspberry pi2

Juan Antonio

jabrena, 2016-03-24:
**Test iteration 1 with Raspberry Pi 2 and Brickpi V2.8:**

I burned a sdcard with rpi2-ev3dev-jessie-2015-12-30.img but I didn't update the system with any upgrade.

**uname -r**

``` bash
robot@ev3dev:/sys/class/lego-port$ uname -r
4.1.15-9-ev3dev-rpi2
``` 
**lsmod**

``` bash
robot@ev3dev:/sys/class/lego-port$ lsmod
Module                  Size  Used by
8021q                  19200  0 
garp                    5471  1 8021q
stp                     1479  1 garp
llc                     3600  2 stp,garp
brickpi                13386  1 
lego_bus                3376  1 brickpi
lego_port_class         3333  1 brickpi
brickpi_battery         1918  0 
i2c_bcm2708             5006  0 
bcm2835_gpiomem         3023  0 
ath9k_htc              55734  0 
ath9k_common           19520  1 ath9k_htc
ath9k_hw              423968  2 ath9k_common,ath9k_htc
ath                    18816  3 ath9k_common,ath9k_htc,ath9k_hw
mac80211              518846  1 ath9k_htc
cfg80211              399468  4 ath,ath9k_common,mac80211,ath9k_htc
rfkill                 16799  2 cfg80211
uio_pdrv_genirq         2986  0 
uio                     8228  1 uio_pdrv_genirq
fuse                   82598  1 
ipv6                  343236  34 
``` 

**dmesg**

Using dmesg, I couldn't see that raspi register ports from brickpi:

``` bash
    4.920094] Registered BrickPi line discipline. (28)
[    5.425653] ath: phy0: Invalid EEPROM Magic. Endianness mismatch.
[    5.427493] ath: phy0: Unable to initialize hardware; initialization status: -22
[    5.431055] ath: phy0: Unable to initialize hardware; initialization status: -22
[    5.435069] ath9k_htc: Failed to initialize the device
[    5.447238] usb 1-1.5: ath9k_htc: USB layer deinitialized
[    5.569346] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    5.572270] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    5.865776] cfg80211: Calling CRDA to update world regulatory domain
[    7.165841] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    7.168490] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    7.196711] 8021q: 802.1Q VLAN Support v1.8
[    9.015792] cfg80211: Calling CRDA to update world regulatory domain
[   12.165757] cfg80211: Calling CRDA to update world regulatory domain
[   14.915764] tty ttyAMA0: Failed to init while setting sensors. (-110)
[   15.315767] cfg80211: Calling CRDA to update world regulatory domain
[   18.465781] cfg80211: Calling CRDA to update world regulatory domain
[   19.398015] EXT4-fs (mmcblk0p2): resizing filesystem from 448496 to 958200 blocks
[   19.444736] EXT4-fs (mmcblk0p2): resized filesystem to 958200
[   21.616116] cfg80211: Calling CRDA to update world regulatory domain
[   24.765769] cfg80211: Calling CRDA to update world regulatory domain
[   27.915764] cfg80211: Calling CRDA to update world regulatory domain
[   31.065765] cfg80211: Calling CRDA to update world regulatory domain
[   34.215776] cfg80211: Calling CRDA to update world regulatory domain
[   37.365766] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
``` 

**ls /sys/class**

``` bash
robot@ev3dev:/sys/class$ ls
backlight        dma          input             leds          pwm          sound       vc-cma
bcm2708_vcio     gpio         iscsi_connection  lego-port     raw          spi_master  vc-mem
bcm2835-gpiomem  graphics     iscsi_endpoint    mem           rfkill       spidev      vc-sm
bdi              hidraw       iscsi_host        misc          rtc          thermal     vchiq
block            i2c-adapter  iscsi_iface       mmc_host      scsi_device  tty         vtconsole
bsg              i2c-dev      iscsi_session     net           scsi_disk    uio         watchdog
devcoredump      ieee80211    iscsi_transport   power_supply  scsi_host    vc
robot@ev3dev:/sys/class$ ls ./lego-port/
robot@ev3dev:/sys/class$ 
``` 

The distro doesn't detect anything.

I am going to not update the system but add the debian package: 
https://github.com/ev3dev/ev3dev/files/120554/linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.zip

jabrena, 2016-03-24:
**Test iteration 2 with Raspberry Pi 2 and Brickpi V2.8:**

I burned a sdcard with rpi2-ev3dev-jessie-2015-12-30.img and later I added the debian package: https://github.com/ev3dev/ev3dev/files/120554/linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.zip

``` bash
sudo dpkg -i linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb 
```

**uname -r**

``` bash
robot@ev3dev:/sys/class$ uname -r
4.1.15-10-rc2-ev3dev-rpi2
``` 

**lsmod**

``` bash
robot@ev3dev:/sys/class$ lsmod
Module                  Size  Used by
8021q                  19600  0 
garp                    5535  1 8021q
stp                     1479  1 garp
llc                     3652  2 stp,garp
arc4                    1753  2 
brickpi                13390  1 
ev3_uart_sensor         6024  1 brickpi
nxt_i2c_sensor         30531  1 ev3_uart_sensor
lego_bus                3376  3 ev3_uart_sensor,nxt_i2c_sensor,brickpi
lego_port_class         3393  2 nxt_i2c_sensor,brickpi
servo_motor_class       6380  1 nxt_i2c_sensor
lego_sensor_class       7671  2 ev3_uart_sensor,nxt_i2c_sensor
brickpi_battery         1918  0 
bcm2835_gpiomem         3023  0 
i2c_bcm2708             5066  0 
ath9k_htc              55566  0 
ath9k_common           19624  1 ath9k_htc
ath9k_hw              424332  2 ath9k_common,ath9k_htc
ath                    18920  3 ath9k_common,ath9k_htc,ath9k_hw
mac80211              549504  1 ath9k_htc
cfg80211              426623  4 ath,ath9k_common,mac80211,ath9k_htc
rfkill                 17335  2 cfg80211
uio_pdrv_genirq         2966  0 
uio                     8228  1 uio_pdrv_genirq
fuse                   83229  1 
ipv6                  344580  34 
```

**dmesg**

Using dmesg, I couldn't see that raspi register ports from brickpi:

``` bash
[    4.911135] Registered BrickPi line discipline. (28)
[    5.860776] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    5.871354] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.136332] ath9k_htc 1-1.5:1.0: ath9k_htc: FW Version: 1.3
[    6.146414] ath9k_htc 1-1.5:1.0: FW RMW support: Off
[    6.156200] ath: EEPROM regdomain: 0x60
[    6.159080] ath: EEPROM indicates we should expect a direct regpair map
[    6.159098] ath: Country alpha2 being used: 00
[    6.159109] ath: Regpair used: 0x60
[    6.196363] cfg80211: Calling CRDA to update world regulatory domain
[    6.294685] ieee80211 phy0: Atheros AR9271 Rev:1
[    7.473140] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    7.486461] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    7.520648] 8021q: 802.1Q VLAN Support v1.8
[    9.346343] cfg80211: Calling CRDA to update world regulatory domain
[   12.506297] cfg80211: Calling CRDA to update world regulatory domain
[   14.926295] tty ttyAMA0: Failed to init while setting sensors. (-110)
[   15.666309] cfg80211: Calling CRDA to update world regulatory domain
[   18.826309] cfg80211: Calling CRDA to update world regulatory domain
[   21.986320] cfg80211: Calling CRDA to update world regulatory domain
[   25.146317] cfg80211: Calling CRDA to update world regulatory domain
[   28.306302] cfg80211: Calling CRDA to update world regulatory domain
[   31.466306] cfg80211: Calling CRDA to update world regulatory domain
[   34.626331] cfg80211: Calling CRDA to update world regulatory domain
[   37.786318] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
```

Questions:

* Is it necessary to upgrade the system and repeat the tests?

jabrena, 2016-03-24:
**Test iteration 3 with Raspberry Pi 2 and Brickpi V2.8:**

I updated the system:

``` bash
he following packages will be upgraded:
  apt bind9-host ca-certificates dpkg ev3devkit-data flash-kernel isc-dhcp-client isc-dhcp-common
  libapt-pkg4.12 libbind9-90 libc-bin libc-dev-bin libc6 libc6-dev libdns-export100 libdns100
  libev3devkit-0.3-0 libfreetype6 libgssapi-krb5-2 libgudev-1.0-0 libicu52 libirs-export91
  libisc-export95 libisc95 libisccc90 libisccfg-export90 libisccfg90 libk5crypto3 libkrb5-3
  libkrb5support0 libldap-2.4-2 libldb1 liblwres90 libpam-modules libpam-modules-bin libpam-runtime
  libpam-systemd libpam0g libpcre3 libpng12-0 libsasl2-2 libsasl2-modules-db libsmbclient libssl-dev
  libssl1.0.0 libsystemd0 libudev1 libwbclient0 libxml2 linux-libc-dev locales login
  multiarch-support ntp openrobertalab openssh-client openssh-server openssh-sftp-server openssl
  passwd perl perl-base perl-modules python-ldb python-samba rpcbind samba samba-common
  samba-common-bin samba-dsdb-modules samba-libs screen ssh sudo systemd systemd-sysv tzdata udev
  wpasupplicant
79 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

**uname -r**

``` bash
robot@ev3dev:/sys/class$ uname -r
4.1.15-10-rc2-ev3dev-rpi2
``` 

**lsmod**

``` bash
robot@ev3dev:/sys/class$ lsmod
Module                  Size  Used by
8021q                  19600  0 
garp                    5535  1 8021q
stp                     1479  1 garp
llc                     3652  2 stp,garp
arc4                    1753  2 
brickpi                13390  1 
ev3_uart_sensor         6024  1 brickpi
nxt_i2c_sensor         30531  1 ev3_uart_sensor
lego_bus                3376  3 ev3_uart_sensor,nxt_i2c_sensor,brickpi
lego_port_class         3393  2 nxt_i2c_sensor,brickpi
servo_motor_class       6380  1 nxt_i2c_sensor
lego_sensor_class       7671  2 ev3_uart_sensor,nxt_i2c_sensor
ath9k_htc              55566  0 
ath9k_common           19624  1 ath9k_htc
ath9k_hw              424332  2 ath9k_common,ath9k_htc
ath                    18920  3 ath9k_common,ath9k_htc,ath9k_hw
mac80211              549504  1 ath9k_htc
cfg80211              426623  4 ath,ath9k_common,mac80211,ath9k_htc
rfkill                 17335  2 cfg80211
brickpi_battery         1918  0 
bcm2835_gpiomem         3023  0 
i2c_bcm2708             5066  0 
uio_pdrv_genirq         2966  0 
uio                     8228  1 uio_pdrv_genirq
fuse                   83229  1 
ipv6                  344580  36 
```

**dmesg**

Using dmesg, I couldn't see that raspi register ports from brickpi:

``` bash
[    5.665435] Registered BrickPi line discipline. (28)
[    6.611816] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.622905] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.721786] ath9k_htc 1-1.5:1.0: ath9k_htc: FW Version: 1.3
[    6.734451] ath9k_htc 1-1.5:1.0: FW RMW support: Off
[    6.734463] ath: EEPROM regdomain: 0x60
[    6.734467] ath: EEPROM indicates we should expect a direct regpair map
[    6.734478] ath: Country alpha2 being used: 00
[    6.734482] ath: Regpair used: 0x60
[    6.746193] cfg80211: Calling CRDA to update world regulatory domain
[    6.774363] ieee80211 phy0: Atheros AR9271 Rev:1
[    8.208351] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    8.221977] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    8.255160] 8021q: 802.1Q VLAN Support v1.8
[    9.896381] cfg80211: Calling CRDA to update world regulatory domain
[   13.056340] cfg80211: Calling CRDA to update world regulatory domain
[   15.676331] tty ttyAMA0: Failed to init while setting sensors. (-110)
[   16.216351] cfg80211: Calling CRDA to update world regulatory domain
[   19.376335] cfg80211: Calling CRDA to update world regulatory domain
[   22.536349] cfg80211: Calling CRDA to update world regulatory domain
[   25.696346] cfg80211: Calling CRDA to update world regulatory domain
[   28.856351] cfg80211: Calling CRDA to update world regulatory domain
[   32.016345] cfg80211: Calling CRDA to update world regulatory domain
[   35.176350] cfg80211: Calling CRDA to update world regulatory domain
[   38.336336] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
```

**ls /sys/class**

``` bash
robot@ev3dev:/sys/class$ ls
backlight        gpio              iscsi_endpoint   mem           rtc          thermal  vtconsole
bcm2708_vcio     graphics          iscsi_host       misc          scsi_device  tty      watchdog
bcm2835-gpiomem  hidraw            iscsi_iface      mmc_host      scsi_disk    uio
bdi              i2c-adapter       iscsi_session    net           scsi_host    vc
block            i2c-dev           iscsi_transport  power_supply  servo-motor  vc-cma
bsg              ieee80211         leds             pwm           sound        vc-mem
devcoredump      input             lego-port        raw           spi_master   vc-sm
dma              iscsi_connection  lego-sensor      rfkill        spidev       vchiq
robot@ev3dev:/sys/class$  ls ./lego-port/
robot@ev3dev:/sys/class$ ls ./lego-sensor/
```

jabrena, 2016-03-24:
Why I receive this timeout?

``` bash
[   15.676331] tty ttyAMA0: Failed to init while setting sensors. (-110)
```

dlech, 2016-03-24:
I don't know why you are getting a timeout. Does this BrickPi work with the official BrickPi software? The only other thing I know to do is put a logic analyzer on it to spy on the communications.

It is clear now that this is actually a different issue from the original closed issue here. You can open a new issue for "tty ttyAMA0: Failed to init while setting sensors. (-110)"

jabrena, 2016-03-24:
Oki Doki, I will create another issue :)

Meanwhile, I will use Raspberry Pi 1 to continue with the development. I think that Raspberry Pi 1 is stable for my both boards. I caught the idea to connect sensors and motors with this issue: #370 

Cheers

gaborlegradi, 2018-09-21:
Hello,
How can I use the _mode_ and _set_device_ attribute of _lego-port_ to get sensors/motors working? [I see the documentation](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/brickpi.html#input-ports), but I have no idea practically what to do. Am I to set device from the code itself? Would you send me an example code part for this, pls!
Thanks,
LeGo