ev3dev/ev3dev #530: Link to EV3Dev releases

jabrena, 2016-02-01:
Good morning team,

reading the tutorial to install a distro using Windows, I noticed that the document doesn't have a link in this paragraph:
http://www.ev3dev.org/docs/tutorials/writing-sd-card-image-win32diskimager/

``` bash
1. Open up the folder where you downloaded the image file from GitHub.
```

In my opinion, we should have a link in the word "Github" to:
https://github.com/ev3dev/ev3dev/releases

Juan Antonio

dlech, 2016-02-01:
Sounds good to me. You could even make the entire phrase "image file from GitHub" a link. Would you like to make a pull request for this?

jabrena, 2016-02-01:
Yes,  this week mate

jabrena, 2016-02-06:
I have made the pull request.
https://github.com/ev3dev/ev3dev.github.io/pull/104

closed 2016-02-06 by dlech