ev3dev/ev3dev #531: BrickPi disconnects on setting sensors

jfcloutier, 2016-02-01:
When I set sensors on the BrickPi, it disconnects after the second ev3-uart sensor.

Last login: Mon Feb  1 21:20:52 2016 from 192.168.1.128
robot@ev3dev:~$ cd /sys/class/lego-port
robot@ev3dev:/sys/class/lego-port$ cd port4
robot@ev3dev:/sys/class/lego-port/port4$ ls
address  driver_name  modes       status     uevent
device   mode         set_device  subsystem
robot@ev3dev:/sys/class/lego-port/port4$ cat address
in3
robot@ev3dev:/sys/class/lego-port/port4$ echo ev3-uart > mode
robot@ev3dev:/sys/class/lego-port/port4$ cat mode
ev3-uart
robot@ev3dev:/sys/class/lego-port/port4$ echo lego-ev3-ir > set_device
robot@ev3dev:/sys/class/lego-port/port4$ cd ../port5
robot@ev3dev:/sys/class/lego-port/port5$ cat address
in4
robot@ev3dev:/sys/class/lego-port/port5$ echo ev3-uart > mode
robot@ev3dev:/sys/class/lego-port/port5$ cat mode
ev3-uart
robot@ev3dev:/sys/class/lego-port/port5$ echo lego-ev3-us > set_device
Connection to 192.168.1.147 closed.

dlech, 2016-02-01:
Is it just killing the network or does the whole system reboot?

jfcloutier, 2016-02-02:
I can rlogin right away. So I guess the system is not rebooting.

dlech, 2016-02-04:
Here is the stacktrace (from `dmesg`):

```
[  507.705823] lego-port port5: Added new device 'in4:lego-ev3-color'
[  507.804752] lego-sensor sensor0: Registered 'lego-ev3-color' on 'in4'.
[  541.618741] lego-sensor sensor1: Registered 'lego-ev3-us' on 'in3'.
[  541.620980] Unable to handle kernel NULL pointer dereference at virtual address 000000d8
[  541.625319] pgd = da178000
[  541.627668] [000000d8] *pgd=18870831, *pte=00000000, *ppte=00000000
[  541.629923] Internal error: Oops: 17 [#1] ARM
[  541.632120] Modules linked in: ev3_uart_sensor nxt_i2c_sensor servo_motor_class lego_sensor_class 8021q garp st
p llc ev3_motor tacho_motor_class dc_motor_class brickpi lego_bus lego_port_class rtc_ds1307 i2c_bcm2708 bcm2835_g
piomem uio_pdrv_genirq uio fuse ipv6
[  541.641757] CPU: 0 PID: 507 Comm: bash Not tainted 4.1.15-9-ev3dev-rpi #1
[  541.644150] Hardware name: BCM2708
[  541.646522] task: da0bc420 ti: da07c000 task.ti: da07c000
[  541.648955] PC is at strncmp+0x18/0x84
[  541.651339] LR is at 0x6c
[  541.653686] pc : [<c02caf44>]    lr : [<0000006c>]    psr: 20000013
sp : da07dbd0  ip : da07dbe0  fp : da07dbdc
[  541.658482] r10: bf0becc8  r9 : 00000000  r8 : 0000000e
[  541.660929] r7 : bf125448  r6 : 00000000  r5 : 000000d8  r4 : da350848
[  541.663375] r3 : bf0be13c  r2 : 00000032  r1 : bf0bf124  r0 : 000000d8
[  541.665824] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  541.668415] Control: 00c5387d  Table: 1a178008  DAC: 00000015
[  541.671010] Process bash (pid: 507, stack limit = 0xda07c188)
[  541.673608] Stack: (0xda07dbd0 to 0xda07e000)
[  541.676260] dbc0:                                     da07dbfc da07dbe0 bf0be170 c02caf38
[  541.681628] dbe0: bf0be13c da3e4000 da3e4130 00000000 da07dc14 da07dc00 bf12404c bf0be148
[  541.687100] dc00: da3e4000 da2b7c00 da07dc34 da07dc18 bf1241c4 bf12400c bf1240e8 c08d02e4
[  541.692769] dc20: da2b7c00 00000000 da07dc44 da07dc38 bf0b8028 bf1240f4 da07dc6c da07dc48
[  541.698719] dc40: c0348854 bf0b800c 00000007 bf125448 da2b7c00 c034890c da3508c0 c08d02c0
[  541.704824] dc60: da07dc84 da07dc70 c034896c c03486b4 00000000 da2b7c00 da07dcac da07dc88
[  541.711053] dc80: c0346d0c c0348918 da10f05c da35b4b4 c05375d8 da2b7c00 da2b7c34 bf0b87c0
[  541.717495] dca0: da07dccc da07dcb0 c034869c c0346cb0 da10f000 da2b7c08 da2b7c00 bf0b87c0
[  541.724184] dcc0: da07dcec da07dcd0 c0347c9c c0348624 00000000 da2b7c08 da2b7c00 00000000
[  541.731191] dce0: da07dd2c da07dcf0 c0345d64 c0347c14 ff0a0004 ffffffff dab39250 da2b7c00
[  541.738476] dd00: da2b7c00 da2b7c00 bf0becc8 da07dd55 da350850 00000000 da2b7cd8 0000000c
[  541.746021] dd20: da07dd44 da07dd30 c0345f1c c03459c4 ffffffff da2b7c00 da07ddac da07dd48
[  541.753695] dd40: bf0b83ac c0345f04 da2b7cd8 da07dd58 c013b5a8 336e6974 67656c3a 76652d6f
[  541.761515] dd60: 73752d33 c0189b00 56b297be 11472765 da07dd94 dab38880 dab39300 dab39250
[  541.769527] dd80: da07de88 00000001 da350848 da350848 da07de25 d88848cc da07df78 d88848c0
[  541.777506] dda0: da07ddd4 da07ddb0 bf0be254 bf0b82c8 00000000 da07de88 da07ddd4 bf0be2dc
[  541.785522] ddc0: 00000002 0000000f da07de1c da07ddd8 bf0be304 bf0be210 dac04340 da07ded0
[  541.793506] dde0: da07de64 da07ddf0 c012cfa0 c012ccf0 00008060 da0f5aa0 c0318824 c02cfd2c
[  541.801519] de00: bf0be2b4 da350848 0000000c d88848cc da07de74 da07de20 bf0b32b8 bf0be2c0
[  541.809503] de20: 00000000 67656c22 76652d6f 73752d33 00000000 00000000 00000000 00000000
[  541.817524] de40: 00000000 00000000 00000000 00000000 00000000 c0000000 0059dc08 da2bcd00
[  541.825508] de60: 00000000 da2bcd00 da07de84 da07de78 c0343f68 bf0b3270 da07de9c da07de88
[  541.833529] de80: c018a550 c0343f4c c018a4fc 00000000 da07ded4 da07dea0 c0189d1c c018a508
[  541.841508] dea0: 00000000 00000000 ffffff9c 0059dc08 da07df78 da0f5aa0 da07df78 0000000c
[  541.849530] dec0: 0059dc08 00000000 da07df44 da07ded8 c0121868 c0189c60 e8fe0f4a 0000000a
[  541.857501] dee0: d884b010 00000000 00000000 ffffff9c dab39250 00000301 00000002 0000028c
[  541.865543] df00: 00000000 c01221a0 00000001 00000000 dab39250 c0120030 da0f5140 c0120044
[  541.873530] df20: 0059dc08 0059dc08 da07df78 da0f5aa0 0000000c 0000000c da07df74 da07df48
[  541.881559] df40: c01220b4 c0121840 da07df74 c013ca8c 00000000 00000000 da0f5aa0 da0f5aa0
[  541.889549] df60: 0000000c 0059dc08 da07dfa4 da07df78 c0122a04 c0122020 00000000 00000000
[  541.897570] df80: 0000000c 0059dc08 b6ee9b40 00000004 c000f708 da07c000 00000000 da07dfa8
[  541.905561] dfa0: c000f500 c01229c4 0000000c 0059dc08 00000001 0059dc08 0000000c 00000000
[  541.913560] dfc0: 0000000c 0059dc08 b6ee9b40 00000004 0000000c 0059dc08 0000000c 00000000
[  541.921548] dfe0: 00000000 beaa38bc b6e16fe8 b6e6e63c 60000010 00000001 00000000 00000000
[  541.929617] [<c02caf44>] (strncmp) from [<bf0be170>] (brickpi_in_port_set_ev3_uart_sensor_mode+0x34/0xc8 [brickpi])
[  541.937675] [<bf0be170>] (brickpi_in_port_set_ev3_uart_sensor_mode [brickpi]) from [<bf12404c>] (ev3_uart_sensor_set_mode+0x4c/0x98 [ev3_uart_sensor])
[  541.945469] [<bf12404c>] (ev3_uart_sensor_set_mode [ev3_uart_sensor]) from [<bf1241c4>] (ev3_uart_sensor_probe+0xdc/0x11c [ev3_uart_sensor])
[  541.953049] [<bf1241c4>] (ev3_uart_sensor_probe [ev3_uart_sensor]) from [<bf0b8028>] (lego_device_driver_probe+0x28/0x2c [lego_bus])
[  541.960302] [<bf0b8028>] (lego_device_driver_probe [lego_bus]) from [<c0348854>] (really_probe+0x1ac/0x264)
[... cut ...]
[  542.048467] [<c018a550>] (sysfs_kf_write) from [<c0189d1c>] (kernfs_fop_write+0xc8/0x1ac)
[  542.055397] [<c0189d1c>] (kernfs_fop_write) from [<c0121868>] (__vfs_write+0x34/0xe8)
[  542.062345] [<c0121868>] (__vfs_write) from [<c01220b4>] (vfs_write+0xa0/0x1b0)
[  542.065945] [<c01220b4>] (vfs_write) from [<c0122a04>] (SyS_write+0x4c/0xa0)
[  542.069459] [<c0122a04>] (SyS_write) from [<c000f500>] (ret_fast_syscall+0x0/0x54)
[  542.076103] Code: e24cb004 e3520000 0a000019 e5d1e000 (e5d03000) 
[  542.079874] ---[ end trace d2cfb8f680e32184 ]---
```

dlech, 2016-02-04:
Kernel oops, null string in `strcmp` in `brickpi_in_port_set_ev3_uart_sensor_mode`.

dlech, 2016-02-04:
Actually wasn't null string, but null `info->sensor`.

jfcloutier, 2016-02-04:
Cool! How do I get the fix? apt-get upgrade?

WasabiFan, 2016-02-04:
The "fix committed" tag means that @dlech made the change in the source code but has not yet built and released an image with it included, so it won't be available through apt-get. You can build a kernel image yourself and install that, or you can wait until another release is published -- the timing of that depends on David's schedule.

jfcloutier, 2016-02-04:
Thanks. For clarification, would this be a new image release or an update that can be installed via apt-get upgrade?

WasabiFan, 2016-02-04:
AFAIK, all planned changes coming up will be available as a kernel upgrade via apt-get. The only time when you must reflash with a new image is when something is changed that fundamentally modifies the system to the point that the changes can't be made as an upgrade; for example, this happened a month or two ago when it was discovered that all previous images had improper file permissions and were causing trouble.

So, you should be able to upgrade directly without reflashing for most (all?) updates unless something else is determined to be horribly wrong. Sorry for not being clear earlier!

jfcloutier, 2016-02-04:
Thank you! I appreciate your help and David's. What's the usual turn-around time (roughly, no commitment)? I speak at a conference in a few weeks and I will demo a robot. I am hoping to replace the (slow) EV3 brick by the (oh so much faster) BrickPi.

dlech, 2016-02-04:
I just depends. I am hoping to include @rhempel's motor changes in the next kernel as well, so it could be a few more weeks before an official release. In the meantime, I might be able to send you a snapshot for testing in a few days.

jfcloutier, 2016-02-04:
Thanks! Much appreciated.

dlech, 2016-02-07:
Do you use RPi 1 or 2?

dlech, 2016-02-07:
nevermind, it's in the first comment

dlech, 2016-02-07:
Here is a kernel for testing. Copy the `.deb` file to your RPi and run `sudo dpkg -i linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb` then `sudo reboot`.

Beware: port names have changed. Input ports are `ttyAMA0:S1` to `4` and output ports are `ttyAMA0:MA` to `D`.

[linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.zip](https://github.com/ev3dev/ev3dev/files/120554/linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.zip)

jfcloutier, 2016-02-07:
Thanks!

I installed the package and I can now register all my sensors and access all my motors. I am however running into a number of issues:

## Infrared sensor

While the IR-PROX mode works...

    robot@ev3dev:/sys/class/lego-sensor/sensor2$ echo IR-PROX > mode
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    76
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    13
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    2

The IR-SEEK mode does not. I had the beacon on channel 1 and the IR sensor could not see it no matter how close it was or how I positioned it relative to the sensor.

    robot@ev3dev:/sys/class/lego-sensor/sensor2$ echo IR-SEEK > mode
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1

## Tacho motors

I can not specify the speed and I can not read the current speed.

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat speed
    cat: speed: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 4 > speed_sp
    -bash: echo: write error: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat speed_sp
    0

I can specify the duty cycle but not read the current duty cycle.

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat duty_cycle
    cat: duty_cycle: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 100 > duty_cycle_sp
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat duty_cycle
    cat: duty_cycle: Operation not supported

Inverting polarity has no effect

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat polarity
    normal
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo inversed > polarity
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat polarity
    inversed
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command

The motor's effective polarity was no changed.

I appreciate you help, as always.

rhempel, 2016-02-07:
@jfcloutier - please close this issue (if you are satisfied that the original issue is fixed) and open up a new issue for this - we need to be able to track things separately

rhempel, 2016-02-07:
@jfcloutier - which conference are you speaking at?

jfcloutier, 2016-02-07:
Created a new issue. Sorry about that.

I am speaking at ElixirDaze on March 4th http://www.elixirdaze.com/

closed 2016-02-07 by jfcloutier

rhempel, 2016-02-07:
No problem @jfcloutier - everybody gets the "open a new issue" notice at one time or another - it really does help keep issues separate.

That sounds like a neat conference! So are you setting up a  language binding for elixir or erlang? Please email me offline via Facebook or LinkedIn - I'm easy to find...

jfcloutier, 2016-02-07:
The conference should be fun! I organize an Erlang & Elixir meetup in Portland, Maine. The last 3 meetups were about my Elixir-based robotics project, the next meetup will be as well. See http://www.meetup.com/Portland-Erlang-Elixir/ I also keep a blog about it at http://jfcloutier.github.io/robotex/ . I recently presented at Elixir Boston. The video and slides are in the latest post.

antonvh, 2016-02-09:
When I do `sudo dpkg -i linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb` I get - more or less as expected - this error:

```/usr/lib/linux-image-4.1.15-10-rc2-ev3dev-rpi2/bcm2708-rpi-b-plus.dtb not found
Kernel /boot/vmlinuz-4.1.15-10-rc2-ev3dev-rpi2 does not match any of the expected flavors (rpi rpi-rpfv), therefore not writing it to flash.
```
Is there B+ .deb file too? (My Pi2 currently has a navio+ autopilot attached to it. :D )

dlech, 2016-02-09:
Here you go. I've enabled `IIO_ST_GYRO_3AXIS` from #81 in this kernel too if you want to try it out.

[linux-image-4.1.15-10-rc3-ev3dev-rpi_1_armhf.zip](https://github.com/ev3dev/ev3dev/files/123614/linux-image-4.1.15-10-rc3-ev3dev-rpi_1_armhf.zip)

antonvh, 2016-02-09:
Brilliant! HT Gyro works! NXT Touch works! I like the new sensor port naming!

dlech, 2016-04-11:
This fix has been released in the `4.1.21-10-ev3dev-rpi` kernel. Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.