ev3dev/ev3dev #532: BrickPi doesn't finish to load with PiStorms & USB Battery

jabrena, 2016-02-02:
Good Morning,

Today, I started testing PiStorm with a old RaspberryPi 1.
In the screen I see BrickPi loading:
https://pbs.twimg.com/media/CaMxAn4WYAAq85u.jpg:large

but after 15m. the screen continues in the same status.
What is the problem?

I updated config.txt

I am using a usb battery connected to raspberry pi.

Juan Antonio

dlech, 2016-02-02:
You need to power the PiStorms via the power connection on the PiStorms itself. And you probably need more than the 5V supplied by USB.

jabrena, 2016-02-03:
Using external battery provided by @mindsensors, the doubt was fixed.
Demo: https://twitter.com/juanantoniobm/status/694973523157188608

closed 2016-02-03 by jabrena