ev3dev/ev3dev #534: Brickman on EV3 & BrickPi // Launch terminal

jabrena, 2016-02-03:
Hi,

using the screen, does exist a way to launch the terminal?

Juan Antonio

dlech, 2016-02-03:
Do you mean using the HDMI or composite out on Raspberry Pi or do you mean PiStorms and not BrickPi?

jabrena, 2016-02-03:
Hi David,

I thought on the Screen but with a UI, the terminal could be appear on HDMI/composite. For the Brickpi I don't know if exist some LCD Hat for it, but if exist it, it could be a nice feature.

What is your opinion?

Juan Antonio

jabrena, 2016-02-03:
This kind of Screen could be ok for Brickpi:
http://www.ebay.es/itm/Raspberry-Pi-RPI-HDMI-5-Touch-Screen-800-x-480-XPT2046-/272110568668?hash=item3f5b0dc4dc:g:uOMAAOSwd0BV2dcg
http://www.ebay.es/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=raspberry+pi+screen&_sacat=0

dlech, 2016-02-03:
In the 2015-12-30 release, there are some environment variables set to use `/dev/fb1` for brickman an other programs. To use the HDMI/composite, you will have to edit `/etc/environment` and `/etc/systemd/system.conf` to use `fb0` instead of `fb1`.

The 5" screen seems a bit big, but the [adafruit 2.8" touchscreen](https://www.adafruit.com/products/1601) would work well. GPIOs 18 and 21 are used for BrickPi Leds, so you can only use 2 of the buttons. But since PiStorms only has 1 button (the GO button) you might just use one anyway. You need to make sure the display is the kind with a 26-pin header and not a 40-pin header.

dlech, 2016-02-03:
Actually, after studying the picture a bit more, it looks like it would require some modification. There is a ribbon cable connector on the bottom that would have to be removed. Also, the 26-pin female header is probably not tall enough and would have to be replaced with an extended on like on the BrickPi.

dlech, 2016-02-03:
It looks like the sell (or at least used to) unassembled versions. And it does look like the extra tall header is included. <https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/assembly>

I might have to get me one of these. :grin:

jabrena, 2016-02-03:
Hehehe
:)

jabrena, 2016-02-06:
A hour ago, I started using the PiStorm with the keyboard and the experience was nice.

[image]

I think that in some moments, the Brick could be used to experiment with the /sys/class directly without using an external computer.

I read your comment:

``` bash
In the 2015-12-30 release, there are some environment variables set to use /dev/fb1 for brickman an other programs. To use the HDMI/composite, you will have to edit /etc/environment and /etc/systemd/system.conf to use fb0 instead of fb1.
```

But I didn't understand the process.

Using the file browser, I suppose that I can execute .sh located on /home/robot so, what is the shell script to open a terminal on the PiStorm screen?

I suppose that later, I could reboot the system and recover the original screen configuration.

jabrena, 2016-02-06:
I tested the keyboard in the EV3 Brick and the experience was nice too:

[image]

I think that a EV3Dev user could use the system with the screen to avoid using a ssh connection from a host. 

Cheers

dlech, 2016-02-06:
> Using the file browser, I suppose that I can execute .sh located on /home/robot so, what is the shell script to open a terminal on the PiStorm screen?

Same as any Linux computer. Press ALT+CTRL+F6 on the keyboard. See <https://github.com/ev3dev/ev3dev/issues/505#issuecomment-171721459>. Works on EV3 too. I only do this if SSH is not working because the screen is just too small.

> But I didn't understand the process.

This only applies to BrickPi, not PiStorms because BrickPi does not have a display. Research Linux framebuffer and `FRAMEBUFFER` environment variable and you will understand more. However, this may no longer be relevant in the future because of #525.

jabrena, 2016-02-06:
Hi, it is great!!!

[Image](https://pbs.twimg.com/media/Cajehk8WIAELpxQ.jpg)

I suppose that FRAMEBUFFER is used to increase the font-size. I will research about it tomorrow.

Do you think that it is possible to connect a screen to raspberry pi and use the same technique or enable some feature from brickman?

Brickman run on ALT-CTRL-F1 but my question is:

is it possible to show the session using the HDMI output?
In the case of EV3 Brick, it is possible to use some USB adapter?
http://www.amazon.com/StarTech-External-Video-Monitor-Adapter/dp/B005G306I2

Juan Antonio

jabrena, 2016-02-07:
Framebuffer: 
https://en.m.wikipedia.org/wiki/Linux_framebuffer

jabrena, 2016-02-07:
DRM: 
https://en.m.wikipedia.org/wiki/Direct_Rendering_Manager

dlech, 2016-02-07:
> is it possible to show the session using the HDMI output?

Yes.

> In the case of EV3 Brick, it is possible to use some USB adapter?

Unlikely since the USB port on the EV3 is 1.1 only. Unless it is a very small display like [this](https://www.youtube.com/watch?v=INLrSd0f03k).

jabrena, 2016-02-07:
Oki, Next week I will test raspberry pi 1 & 2 with some screen and LCD projector. 

Is it possible to add some desktop environment? 
It could a very interesting feature for students. 

Scenario:
One student connect a screen and a keyboard to code for his robot.

jabrena, 2016-02-07:
http://lifehacker.com/five-best-linux-desktop-environments-1648022755

jabrena, 2016-02-15:
Hi,

Yesterday, I connected a Screen with Raspberry Pi using HDMI wire but I didn't see anything:

[image]

How to show the console on a Screen using HDMI?
Is it possible to add XFCE in a easy way?
http://www.xfce.org/ 

Juan Antonio

dlech, 2016-02-15:
We are not going to officially support using HDMI or graphical desktops on ev3dev. However, if you would like to make it work, you need to set `fbcon=map:10` in `config.txt` on the FAT partition (`/boot/flash`). This overrides the value set by the [PiStorms device tree overly](https://github.com/ev3dev/rpi-kernel/blob/ev3dev-jessie/arch/arm/boot/dts/overlays/pistorms-overlay.dts#L15). It is not needed on BrickPi since BrickPi does not have a display.

Use apt to install a display manager and desktop environment and I am sure you can get it working. You will have to do some research to figure out how to control which display that it uses.

dlech, 2016-05-04:
> I might have to get me one of these. :grin:

I got one. http://lechnology.com/2016/05/adding-a-display-to-brickpi/

moriarty, 2016-05-05:
@dlech thanks for the awesome guide, I needed one of these displays for another project and it's since been sitting in my desk. Now I know how to make it fit :+1:

dlech, 2016-05-08:
Can we close this issue?

closed 2016-10-11 by dlech