ev3dev/ev3dev #535: Brickman / Options / File browser / navigate in the whole file system

jabrena, 2016-02-06:
Hi,

At the moment, if a user try to use the option "File browser", the path showed is /home/robot.
My question is about if it is possible to allow to user to navigate in the whole file system. Maybe the user could begin at /home/robot and later, if some need to explore the Debian FS, the could do it.

What is your opinion?

Cheers

Juan Antonio

dlech, 2016-02-06:
Brickman runs as root, so you would be able to get into lots of places where you shouldn't if we allowed browsing the entire file system. Also, when running programs via the file browser, we currently run as the `robot` user, so it makes sense to just limit it to `robot`'s home directory.

jabrena, 2016-02-07:
For me,  it is a nice answer. If some user need to list other path,  an alternative could be the usage of a terminal: alt + ctrl + f6

closed 2016-02-07 by jabrena