ev3dev/ev3dev #536: PS4 DualShock4 controller support

meehien, 2016-02-06:
Hey guys.

Would it be possible to include this [patch](https://launchpadlibrarian.net/223799259/hid-sony-ds4.patch) in the ev3dev-kernel. Full discussion is available [here](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1511855) and the patch has been included in kernel 4.5.

I have 3 versions of the DS4 controllers and I can confirm that at least one of my controllers does not work without the patch and works with the patched kernel.

Also, somewhat related, it would be nice to have the DS4 controller kernel support enabled by default assuming there are no space constraints. Currently I need to rebuild by kernel to enable that, and it is the only reason I need to do it.

Thank

dlech, 2016-02-06:
Sure we can turn it on and add the patch.

dlech, 2016-02-06:
https://github.com/torvalds/linux/commit/b71b5578a84d297954e4812ba0ca2d466e61cf42

dlech, 2016-02-07:
Which kernel option are you enabling? I looks like `HID_SONY` is already enabled.

dlech, 2016-02-07:
`HID_SONY` is enabled in the `v3.16.7-ckt21-9-ev3dev-ev3` kernel. Perhaps you are using an older kernel?

https://github.com/ev3dev/ev3-kernel/commit/581d5b6f9fd95048131b6d6db18be873314e9521

meehien, 2016-02-07:
yes i was using a custom config file to build my kernel and didn't notice `HID_SONY` was added.

I also have the `CONFIG_SONY_FF` enabled for DS4, which works really nice with the sdl (and PySDL2) library.

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel. Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

meehien, 2016-05-14:
the bug is back in kernel v4.4.9-11-ev3dev-ev3.
would it be possible to patch hid-sony.c again?

Thanks

dlech, 2016-05-14:
The patch is still there. https://github.com/ev3dev/ev3-kernel/commit/371f336f30f3601732494674bbb2b0f7aea2e21c

It could be that this patch is interfering: https://github.com/ev3dev/ev3-kernel/commit/19c8b8948be55c6130db52701cfacd69df2a97b1

you can try reverting it and see if it fixes the problem.

meehien, 2016-05-14:
You are right it's not that. It seems like neither is the ev3dev/ev3-kernel@19c8b89 patch.

It think something must have changed in the way bluetooth is pulled. My diagnostics program (a quick c program that pulls bluetooth joystick SDL events and prints them to the console) shows a constant stream of events. Same program on the 3.x kernel was only showing real events (i.e. actuall button presses).

I think this overwhelms the bluetooth driver and causes huge lag throughout (brickman, ssh connection, etc. ) and eventually disconnects the bluetooth device.

I will further investigate.

dlech, 2016-06-14:
Since this is not the same issue as before, I am closing. Open a new issue if you get more information on the new problem.

closed 2016-06-14 by dlech