ev3dev/ev3dev #537: Something broke with GitHub Pages upgrade

dlech, 2016-02-06:
GitHub recently [upgraded to Jekyll 3.0](https://github.com/blog/2100-github-pages-now-faster-and-simpler-with-jekyll-3-0). We are getting some [errors in CI now](https://travis-ci.org/ev3dev/ev3dev.github.io/builds/107443995) which I assume are a result of this.

WasabiFan, 2016-02-06:
And I was under the impression that it wouldn't have any impact on us :wink: Those errors seem peculiar... Can check it out in a couple hours.

WasabiFan, 2016-02-06:
Might be related to this: http://jekyllrb.com/docs/upgrading/2-to-3/#sitecollections-has-changed

WasabiFan, 2016-02-06:
OK, I (sort of) figured it out. I'm seeing this wherever we @mention someone:

```html
<a href="https://github.com/dlech" class="user-mention"><a href="https://github.com/dlech" class="user-mention">@dlech</a></a>
```

Not quire sure _why_ this is happening, but I imagine it will either be a simple fix on our end (maybe we have duplicated new functionality with an old plugin) or it is a bug in the mentions plugin.

dlech, 2016-02-15:
https://github.com/jekyll/jekyll-mentions/pull/25

WasabiFan, 2016-02-15:
Ah, that makes sense. Any quick fix options?

dlech, 2016-02-15:
we can turn off jekyll-mentions temporarily or just ignore the errors and wait for the fix to trickle down.

dlech, 2016-02-18:
upstream was faster than us. Fixed in jekyll-mentions 1.0.1.

closed 2016-02-18 by dlech

WasabiFan, 2016-02-18:
But has it been pushed to GH pages prod yet?

dlech, 2016-02-18:
yup, https://pages.github.com/versions/

WasabiFan, 2016-02-18:
Oh, cool! They're surprisingly fast :wink: You can trigger a Travis build if you'd like so that our repo status shows up green.

dlech, 2016-02-18:
There is actually a broken link,so no green.