ev3dev/ev3dev #538: EV3Dev-lang-java v0.2.0 is out

jabrena, 2016-02-07:
Good afternoon everyone,

yesterday, I released the second version of the _Java binding for EV3Dev_.
https://jabrena.github.io/ev3dev-lang-java

In this version, the way to use the objects follow leJOS syntax. This minor version has support for the following kits:

- 45544 LEGO MINDSTORMS Education EV3 Core Set
- 31313 LEGO MINDSTORMS EV3

When you uses the EV3 brick with the package manager, the available JDK is Java 7 but for Raspberry Pi 1 & 2, is possible to develop with Java 8. Using EV3Dev with Java, is able to install OpenCV for Java:

``` bash
sudo apt-get install libopencv2.4-java
```

Besides, it is possible to use Profiling tools as visualvm or jconsole. 
Further information here: https://github.com/jabrena/ev3dev-lang-java/issues/38

**Javadocs:**
https://jabrena.github.io/ev3dev-lang-java/docs/api/

**Examples:**
https://github.com/jabrena/ev3dev-lang-java/tree/master/library/src/test/java/ev3dev/examples

**Issues:**
https://github.com/jabrena/ev3dev-lang-java/issues

Cheers

Juan Antonio

dlech, 2016-02-07:
It would be more appropriate to make this announcement via news on the website. Just add a new page to <https://github.com/ev3dev/ev3dev.github.io/tree/master/news/_posts> and send a pull request.

jabrena, 2016-02-07:
Oki doki

closed 2016-02-25 by dlech