ev3dev/ev3dev #539: Some sensor and motor attributes can not be read or set or are inoperative

jfcloutier, 2016-02-07:
I installed the linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb package. I am however running into a number of issues:

## Infrared sensor

While the IR-PROX mode works...

    robot@ev3dev:/sys/class/lego-sensor/sensor2$ echo IR-PROX > mode
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    76
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    13
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    2

The IR-SEEK mode does not. I had the beacon on channel 1 and the IR sensor could not see it no matter how close it was or how I positioned it relative to the sensor.

    robot@ev3dev:/sys/class/lego-sensor/sensor2$ echo IR-SEEK > mode
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1

## Tacho motors

I can not specify the speed and I can not read the current speed.

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat speed
    cat: speed: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 4 > speed_sp
    -bash: echo: write error: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat speed_sp
    0

I can specify the duty cycle but not read the current duty cycle.

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat duty_cycle
    cat: duty_cycle: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 100 > duty_cycle_sp
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat duty_cycle
    cat: duty_cycle: Operation not supported

Inverting polarity has no effect

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat polarity
    normal
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo inversed > polarity
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat polarity
    inversed
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command

The motor's effective polarity was no changed.

I appreciate you help, as always.

dlech, 2016-02-07:
This is two separate issues. Please separate, on for the IR sensor and one for the motors.

dlech, 2016-02-07:
And actually, the motor issues could be split into multiple issues as well.

jfcloutier, 2016-02-07:
No problem. I reposted the above as separate issues #542, #543, #544 and #545

closed 2016-02-07 by jfcloutier

dlech, 2016-02-07:
Thanks. It really makes it a lot easier to keep track that way.