ev3dev/ev3dev #541: PiStorms // How to use a Mindsensors AbsoluteIMU

jabrena, 2016-02-07:
http://www.mindsensors.com/ev3-and-nxt/15-gyro-multisensitivity-accelerometer-and-compass-for-nxt-or-ev3

``` bash
robot@ev3dev:/sys/class/lego-port/port0$ cat address 
pistorms:BBS1
robot@ev3dev:/sys/class/lego-port/port0$ cat modes 
none nxt-analog nxt-color i2c-thru ev3-analog ev3-uart
```

dlech, 2016-02-07:
This is an NXT/I2C sensor, so you need to put the port in `i2c-thru` mode. 

Read the footnotes here: http://www.ev3dev.org/docs/ports/pistorms-in-port/

jabrena, 2016-02-07:
Any example using a i2c sensor with pistorms? 
I would like to learn the idea and later write a script to handle this example.

dlech, 2016-02-07:
It is really not any different than using I2C sensors on EV3 other than you have to put the port in `i2c-thru` mode first.

jabrena, 2016-02-07:
Yes, I can put the sensor in this mode: i2c-thru, but later my question is if exist some example in the community using i2c with this environment.

jabrena, 2016-02-09:
Hi @dlech if exist the class for this sensor:
http://www.ev3dev.org/docs/sensors/mindsensors.com-gyro-multisensitivity-accelerometer-and-compass/

Why using PiStorms, I can't detect it?

Juan Antonio

dlech, 2016-02-09:
Maybe because of #425?

jabrena, 2016-02-11:
If I have set the sensor in mode i2c-thru, what is the next step?

[image]

Does exist some way using /sys/class to send i2c commands to some lego-port?

Juan Antonio

dlech, 2016-02-11:
The reason it is called `i2c-thru` is because I2C messages are passed through to the Raspberry Pi's I2C pins. All 4 ports share the same bus. This is explained here:

> Read the footnotes here: http://www.ev3dev.org/docs/ports/pistorms-in-port/

So, you don't send I2C commands to a PiStorms port, you send commands to the Raspberry Pi I2C port. `/dev/i2c1` or something like that.

You can use the sensor using the python library from mindsensors.com. They get around the issue in #425 by sending read and write as 2 separate messages.

jabrena, 2016-02-11:
So, from the Library side, it is necessary to detect the platform to use i2c or not.

I will research how to send i2c messages with java to path: /dev/i2c1

dlech, 2016-02-11:
You have to put the port in `i2c-thru` mode, but after that, LEGO compatible I2C sensors should be automatically detected unless they suffer from #425.

evthree, 2016-12-01:
I have not been able to make this work. The sensor shows up in the pistorms screen, I am able to watch its values manually, but never via the C++ API. Can you please explain what is wrong with the following code:

`#include <stdio.h>
#include <unistd.h>
#include "../common/ev3dev.h"

using namespace ev3dev;

int main( int argc, char* argv[] )
{
  lego_port imu_port( "pistorms:BAS2" );
  imu_port.set_mode( "i2c-thru" );
  //imu_port.set_set_device( "ms-absolute-imu" );                                                                           

  i2c_sensor imu( "pistorms:BAS2:i2c17" );
  //i2c_sensor imu( "/dev/i2c-1" );                                                                                         
  //imu_port.set_set_device( "ms-absolute-imu" );                                                                           

  printf( "%s\n", (imu.connected()?"connected" : "not connected") ); //prints "not connected"

  while( true )
  {
    int imuval = imu.value(); //always fails
    printf( "imu: %d       \r", imuval );
    usleep( 10000 );
  }

  return 0;
}
`

Thank you!

dlech, 2016-12-01:
It looks correct to me, but I have never used C++ with ev3dev. Ping @ddemidov.

ddemidov, 2016-12-01:
I don't have a pistorms, but the following code works for me on EV3:

https://gist.github.com/ddemidov/b5aedd94ba86648f63377c09257f0b38

The difference is that I am using `ev3dev::sensor` class instead of `ev3dev::i2c_sensor`.

The output is:
```
$ ./hello-imu
connected: 1
address: in1:i2c17
driver_name: ms-absolute-imu
tilt.x: -45
tilt.y: 63
tilt.z: -58
accel.x: 643
accel.y: -524
accel.z: 549
compass: 173
mag.x: 75
mag.y: 938
mag.z: -888
gyro.x: 30
gyro.y: 48
gyro.z: 4
```

ddemidov, 2016-12-01:
@dlech, `i2c_sensor` class [tries to find](https://github.com/ddemidov/ev3dev-lang-cpp/blob/53c69fdbc460d9aae1d5df51cc1433392153b803/ev3dev.cpp#L608) a device with [driver_name="nxt-i2c-sensor"](https://github.com/ddemidov/ev3dev-lang-cpp/blob/53c69fdbc460d9aae1d5df51cc1433392153b803/ev3dev.h#L151). Is that correct behavior?

dlech, 2016-12-01:
No. I think "nxt-i2c-sensor" is a generic name for an unknown sensor, but I don't know if any cases that actually use this name. Also, IIRC, there was a discussion a while back about dropping the I2C subclass from the ev3dev-lang specification. The only time you would need/want a subclass is if you want to simplify the use of the `direct` attribute. Also, I suppose you could simplify the constructor by having a separate address parameter that appends `:i2c%d` to the `address`.

bmegli, 2016-12-01:
@ddemidov, @dlech 

See also ev3dev/ev3dev-lang/issues/167, it seems to be about that.

You can see there what @WasabiFan did in his implementation for `i2c_sensor` and what I needed in my fork of ev3dev-lang-cpp.

In my fork of c++ bindings I have constructor that accepts anything (not only "nxt-i2c-sensor") and I use it to disable polling, set sensor mode, send command to sensor, and finally use direct attribute.

ddemidov, 2016-12-01:
https://github.com/ddemidov/ev3dev-lang-cpp/commit/0f13224bc0619cb100c228bd6e3ea965307e52ed

ddemidov, 2016-12-01:
@bmegli, would you like to make a pull request? I see you have a couple of other fixes there.

bmegli, 2016-12-01:
@ddemidov the other fix if for `device_index`, I am not sure if it is still part of official bindings specification.

I have this feeling that I am the only person using it.

The rest is for custom makefile I use with some of my programs (object file I link against instead of library) so it shouldn't be pulled.

evthree, 2016-12-01:
@bmegli  
> I have this feeling that I am the only person using it.

If you have any generic fixes please put them in. I am sure a lot of folks (like me) are spending a lot of time trying to figure all of this out.

bmegli, 2016-12-01:
@evthree 

Ok, I am sorry about that. I sometimes get depressed and loose self-confidence and my-code-confidence.

@ddemidov 

Since you have already made a commit for `i2c_sensor` I will make a pull only for `device_index` (and try to recall and explain why it was needed)

bmegli, 2016-12-01:
PR issued with explanation and fix for `device_index` bug. 

This PR is directly to (liquid generated?) c++ source, so maybe the real fix should go somewhere else.

evthree, 2016-12-01:
Thank you all for the quick responses and fixes. My code is working now. I am posting it here as a complete working C++ example for others who might encounter this in the future:

pistormsimu.cpp:
```
#include <stdio.h>
#include <unistd.h>
#include "../ev3dev/ev3dev.h"

using namespace ev3dev;

int main( int argc, char* argv[] )
{
  lego_port imu_port( "pistorms:BAS2" );
  imu_port.set_mode( "i2c-thru" );
  i2c_sensor imu( "pistorms:BAS2:i2c17", { "ms-absolute-imu" } );

  printf( "%s\n", (imu.connected() ? "connected" : "not connected") );

  while( true )
  {
    int imuval = imu.value();
    printf( "imu: %d       \r", imuval );
    usleep( 10000 );
  }

  return 0;
}
```
ev3dev.h and ev3dev.cpp must have two parameters in the constructor as in https://github.com/ddemidov/ev3dev-lang-cpp/commit/0f13224bc0619cb100c228bd6e3ea965307e52ed

Build using:
```g++ pistormsimu.cpp  ../ev3dev/ev3dev.cpp -std=c++11 -o pistormsimu -D EV3DEV_PLATFORM_BRICKPI```

closed 2017-07-25 by dlech