ev3dev/ev3dev #542: IR-SEEK mode inoperative

jfcloutier, 2016-02-07:
I installed the linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb package.

## Infrared sensor

While the IR-PROX mode works...

    robot@ev3dev:/sys/class/lego-sensor/sensor2$ echo IR-PROX > mode
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    76
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    13
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    2

The IR-SEEK mode does not. I had the beacon on channel 1 and the IR sensor could not see it no matter how close it was or how I positioned it relative to the sensor.

    robot@ev3dev:/sys/class/lego-sensor/sensor2$ echo IR-SEEK > mode
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
    -4
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1
    robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value1
    -1

dlech, 2016-02-07:
The `-4` value tells me that there are communication errors with the BrickPi. I have found that EV3/UART sensors are not the most reliable on BrickPi. See <https://github.com/DexterInd/BrickPi/issues/24>. If there were more of me, I would fix up the BrickPi firmware.

rhempel, 2016-02-07:
Ha Ha - you are Brent in the book "The Phoenix project". A must read if you're involved at all in delivering software.

[http://www.amazon.ca/The-Phoenix-Project-Helping-Business/dp/0988262592]

jfcloutier, 2016-02-07:
Any hint from the Dexter folks whether they are working on fixing their firmware?

dlech, 2016-02-07:
No, they haven't touched it in years.

jfcloutier, 2016-02-07:
That's sad. I also have an NXT kit. Does the BrickPi firmware have the same problems with NXT sensors?

dlech, 2016-02-07:
BrickPi was actually designed for NXT - it came out before EV3. So yes, NXT sensors work better with BrickPi. No support for LEGO NXT Ultrasonic sensor though.

rhempel, 2016-02-07:
@dlech - would the BeagleBone be a better fit for @jfcloutier? The hardware motor drivers are closer to  what's on the EV3, yes?

jfcloutier, 2016-02-07:
That's encouraging. I'll try with my NXT sensors/motors. BTW my EV3 ultrasonic sensor seems to behave well on the ev3dev/BrickPi. So there's hope.

dlech, 2016-02-07:
@rhempel, if you already have the BrickPi, then might as well get the most out of it.