ev3dev/ev3dev #543: Can't set or read the speed of tachomotors

jfcloutier, 2016-02-07:
I installed the linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb package.

I can not specify the speed and I can not read the current speed of a tachomotor

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat speed
    cat: speed: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 4 > speed_sp
    -bash: echo: write error: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat speed_sp
    0

dlech, 2016-02-07:
This is currently not implemented for BirckPi.

dlech, 2016-02-07:
Again, this would be best solved by modifying the BrickPi firmware.

dlech, 2016-02-07:
Somewhat related: http://www.dexterindustries.com/topic/accuracy-of-rotating-a-motor/

jfcloutier, 2016-02-07:
That's surprising. Setting/reading motor speed is kind of important. Let's hope Dexter fixes their firmware soonish.

dlech, 2016-03-25:
Found a great paper on calculating motor speed.

http://www.diegm.uniud.it/petrella/Azionamenti%20Elettrici%20II/Tesine/Petrella%20et%20al.%20-%20Speed%20Measurement%20Algorithms%20for%20Low-Resolution%20Incremental%20Encoder%20Equipped%20Drives_a%20Comparative%20Analysis.pdf

WasabiFan, 2016-03-26:
@dlech Are you saying that that paper helps you bypass the firmware limitations so that you can measure motor speed?

dlech, 2016-03-26:
I would say it explains all of the options so that we can make the best choice given the constraints we have to work with.

dlech, 2016-04-11:
This fix has been released in the `4.1.21-10-ev3dev-rpi` kernel. Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech