ev3dev/ev3dev #544: Can't read the current duty cycle of tachomotors

jfcloutier, 2016-02-07:
I installed the linux-image-4.1.15-10-rc2-ev3dev-rpi2_1_armhf.deb package.

I can specify the duty cycle but not read the current duty cycle.

    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat duty_cycle
    cat: duty_cycle: Operation not supported
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 100 > duty_cycle_sp
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ echo stop > command
    robot@ev3dev:/sys/class/tacho-motor/motor0$ cat duty_cycle
    cat: duty_cycle: Operation not supported

dlech, 2016-02-07:
This is currently not implemented, but should be trivial to implement.

jabrena, 2016-02-07:
Yes, I observed the same effect in PiStorm.
Anyway @jfcloutier, in the case of Large Motors, you can use the attribute speed_sp once the motor has enabled the regulated features. This evening, I was testing it.

``` bash
echo 500 > speed_sp
cat speed_sp
echo run-forever > command
sleep 5
echo stop > command
```

dlech, 2016-02-07:
PiStorms and BrickPi have totally different motor controllers. This issue is about BrickPi. `speed_sp` is not supported on BrickPi.

dlech, 2016-04-11:
This fix has been released in the `4.1.21-10-ev3dev-rpi` kernel. Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech