ev3dev/ev3dev #547: Out of memory when building Erlang on EV3 running release ev3dev-jessie-2015-12-30

jfcloutier, 2016-02-11:
While I could build Erlang from sources in the previous release (ev3-ev3dev-jessie-2015-09-13), I now run into an *out of memory* condition. 

Note that I MUST build Erlang from sources in order to install an 18.* version which I require.

    robot@ev3dev:~/otp_src_18.2.1$ make
     MAKE	depend
    make[1]: Entering directory '/home/robot/otp_src_18.2.1/erts/emulator'
     MAKE	generate
    [...]
    CC	/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_dfa_exec.o
    GEN	/home/robot/otp_src_18.2.1/erts/emulator/pcre/pcre_exec_loop_break_cases.inc
    CC	/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_exec.o

    cc1: out of memory allocating 8251148 bytes after a total of 17690624 bytes
    pcre/pcre.mk:64: recipe for target '/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_exec.o' failed
    make[3]: *** [/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_exec.o] Error 1

    robot@ev3dev:~/otp_src_18.2.1$ vmstat -s
        59436 K total memory
        41028 K used memory
         5696 K active memory
        21232 K inactive memory
        18408 K free memory
         2184 K buffer memory
        21344 K swap cache
        65532 K total swap
         8216 K used swap
        57316 K free swap

Any suggestion?

Thanks!

dlech, 2016-02-11:
Don't compile on your EV3. [Use brickstrap.](http://www.ev3dev.org/docs/tutorials/using-brickstrap-to-cross-compile/)

dlech, 2016-02-11:
Or mount more swap as in #520.

jfcloutier, 2016-02-11:
Trying both. Thanks!

jfcloutier, 2016-02-11:
All good. Thanks again.

closed 2016-02-11 by jfcloutier