ev3dev/ev3dev #548: Issue with instructions to Set Up An nfs FileShare

fantauzzi, 2016-02-11:
Following instructions provided at [Set Up An nfs FileShare](https://github.com/ev3dev/ev3dev/wiki/Set-Up-An-nfs-FileShare) , at the time to mount the share I got:

`robot@ev3dev:~$ mount /home/robot/nfs/ev3dev
mount.nfs: rpc.statd is not running but is required for remote locking.
mount.nfs: Either use '-o nolock' to keep locks local, or start statd.
mount.nfs: an incorrect mount option was specified
`

I fixed the problem by running, in the brick shell:

`sudo /etc/init.d/rpcbind start`

and then trying to mount the directory again. If others have the same issue too, consider to add the fix in the instructions page.

dlech, 2016-02-11:
rpcbind is disabled by default, so yet, the instructions need to be updated.

dlech, 2016-11-17:
It looks like this has been fixed (using `systemctl` instead of `init.d`)

http://www.ev3dev.org/docs/tutorials/setting-up-an-nfs-file-share/#how-to-do-it---ev3

closed 2016-11-17 by dlech

andrewstarks, 2017-11-28:
@dlech I don't think it has. I followed the directions that you linked to and had the same problem. It was tricky to track down but running rpcbind before mounting worked made it work. I don't know enough about linux to fix it in a more permanent way.