ev3dev/ev3dev #549: Set-up of ev3dev library on PC gives error (ValueError: Cannot find the version number!)

fantauzzi, 2016-02-11:
I have tried to install the ev3dev library on my Ubuntu PC, following instructions here:
[Setting Up a Python Development Environment with PyCharm](http://www.ev3dev.org/docs/tutorials/setting-up-python-pycharm/
)

After cloning the git repository (which went OK) I got this error running setup:

`> python3 ev3dev-lang-python/setup.py install`

`Traceback (most recent call last):
  File "ev3dev-lang-python/setup.py", line 7, in <module>
    version=git_version(),
  File "/home/fanta/workspace/3rd_party/ev3dev-lang-python/git_version.py", line 74, in git_version
    raise ValueError("Cannot find the version number!")
ValueError: Cannot find the version number!
`

(using python instead of python3 I got the same error).

dlech, 2016-02-11:
ev3dev-lang-python comes pre-installed (python 2 only for now). If you have further issues with ev3dev-lang-python, it has it's own GitHub site and issue tracker at <https://github.com/rhempel/ev3dev-lang-python>.

We should update the docs to not clone from git.

ddemidov, 2016-02-12:
The setup script should be run from the project root. So the following should work:
```
cd ev3dev-lang-python
python3 setup.py install
```

fantauzzi, 2016-02-12:
Thanks. Also, consider to update the tutorial to indicate to issue the command as superuser, otherwise it will fail like:

\> python3 setup.py install
running install
Checking .pth file support in /usr/local/lib/python3.4/dist-packages/
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/usr/local/lib/python3.4/dist-packages/test-easy-install-2325.pth'

So it could be something like
`sudo python3 setup.py install`

or using the options to specify the installation directory

dlech, 2016-03-14:
OK to close?

closed 2016-10-11 by dlech