ev3dev/ev3dev #550: ev3dev c++ file access

synmehdi, 2016-02-14:
I am developing a PID control system to create a balancing robot in ev3dev using c++.
The PID calculation is done in a computer and the motor output is sent via Wi-Fi to the Lego.
I am not using any C++ binding. I'm just opening duty_cycle_sp file and writing in it.
When I tried UDP protocol for sending the data, It works very well for a short time and then it has a lag during some time, periodically. 
TCP works better and I'm able to have only a 5ms latency which is acceptable for what i want to do.
My question are:
Why would the UDP protocol lag some times? because UDP is faster.
What is the best way to do it?
I have an infinite while loop that is constantly receiving data. Is it okay to flush the content in the file to write in duty_cycle_sp file without having to open it and close it in the loop?
I'm a beginner in this area.

rhempel, 2016-02-14:
Why not run the PID algorithm on the EV3?

synmehdi, 2016-02-14:
Thank you for your answer.
My goal is to use networking in this project. My sensor is a smartphone on the robot that sends processed IMU data to a computer with calculates the PID and sends back motor power values to the robot. It is also interesting because i cant track all that's happening from the computer and do 3D modeling...
I've already done all the computer part and most of the brick programming (which is only a TCP client).

dlech, 2016-02-14:
> Is it okay to flush the content in the file to write in duty_cycle_sp file without having to open it and close it in the loop?

Use the `seek` function so that you don't have to open and close the file in the loop. Pseudocode:

    file = open(name)
    while(true) {
        file.seek(0)
        file.write(data)
    }
    file.close()

dlech, 2016-02-14:
> Why would the UDP protocol lag some times?

The processor is busy with something else?

synmehdi, 2016-02-15:
With a TCP connexion I don't have the same problem. 
The only thing that my c++ program does is send a token and receive data from my computer then write that data on the motor files. Maybe it is the writing file part costly for the CPU. I will try with the seek method. 
Or maybe the processor is busy maintaining the ssh connection too.
Is there a method to know?
Is there any direct method to control the motors?

dlech, 2016-02-15:
Since the data is received slower via TCP, maybe the writes are able to keep up, but not with the faster UDP. The seek method is currently the most direct way to control the motor.

closed 2017-07-25 by dlech